openwrt/packages #16256: acme: Restart uhttpd after renewing certificate

K2IE, 2021-07-30:
I'm not certain if the issue is in acme proper or in luci-app-acme, but whenever my certificate is renewed I have an issue.  uhttpd is not restarted as part of the process and so at some point my certificate will expire, I must then ssh in and manually restart uhttpd.

This ought to be triggered at the end of the renewal process if "Use for uhttpd" is checked in luci-app-acme.

CodeFetch, 2021-07-31:
You could introduce a generic configuration parameter in uacme for triggering a shell-command after renewal. Hard-coding this to uhttpd is noch a good idea. Instead you could define a generic trigger script which like "uacme_restart_httpd" and then let uhttpd/nginx/apache whatever provide that file.

scde, 2021-08-21:
Hi,

I have the same issue and think `uhttpd` is not restarted but only reloaded. See https://github.com/openwrt/packages/blob/b03fe54e0e988f62c33503cb20cdb30b366870ad/net/acme/files/run.sh#L144
I guess for an easy fix you could just replace `reload` with `restart` in that line. Since `nginx` restarts every time as well you might want to do it like that.

But if you want to prevent unwanted `uhttpd` restarts you would need to look a bit closer into it and only trigger if there was a successful renewal. Maybe by passing a return value to the `post_checks()` function and using it to branch between reload and restart.

Best regards
scde

Neustradamus, 2022-08-15:
@scde, @CodeFetch, @scde: I think that this issue can be closed:
- https://github.com/openwrt/packages/pull/16482
- https://github.com/openwrt/packages/pull/16545
- https://github.com/openwrt/packages/pull/16546

scde, 2022-08-15:
Hi @Neustradamus,
yeah. From my point of view you can close it. I didn't observe the need for a manual uhttpd restart after a cert update since this change. The script seems to work correctly now.
Since I didn't open the issue I can't close it, sorry.
BR scde

Neustradamus, 2022-08-15:
@scde: Thanks for your reply!

Hope that an admin or @K2IE can close it...

closed 2022-08-16 by feckert