openwrt/packages #16261: strongswan: missing deps

stintel, 2021-07-31:
Maintainer: @pprindeville @Thermi 
Environment: OpenWrt master

Description:
There seem to be some missing deps.

```
Sun Aug  1 02:18:18 2021 daemon.info ipsec: 00[LIB] feature CUSTOM:libcharon-receiver in critical plugin 'charon' has unmet dependency: CUSTOM:socket
Sun Aug  1 02:18:18 2021 daemon.info ipsec: 00[LIB] feature CUSTOM:libcharon in critical plugin 'charon' has unmet dependency: CUSTOM:libcharon-receiver
```

Thermi, 2021-07-31:
Looks like you don't have the package installed that provides `libstrongswan-socket-default.so` or `libstrongswan-socket-dynamic.so`.

stintel, 2021-08-01:
Yeah fixed it already but swanctl should depend on the socket package.

Thermi, 2021-08-01:
No, not swanctl. The strongswan-charon package should.

commit 603f70e9 (2022-08-15): strongswan: add strongswan-mod-socket

This a virtual package that is satisfied by either
strongswan-mod-socket-default or strongswan-mod-socket-dynamic, and is
required by the charon daemon. When neither of these packages is
installed, charon will not function.

Closes #16261, #16263 and #16367.
closed 2022-08-15 by pprindeville