openwrt/packages #16268: miniupnpd: can not work with openwrt master

openips, 2021-08-01:
the miniupnpd commit 107f3376b5114cd17c115e25026b031bd439e9be
test not work with openwrt commit ac4e8aa2f8d98158ea7b749f877269f1f5fa9c

and miniupnpd commit b8f9e4354efd7f722a28eb45cec429a22a0d7a9d can work with openwrt master very well

check and fix bugs 
thks

openips, 2021-08-03:
bye the way
the miniupnpd  PKG_VERSION:=2.2.2 PKG_RELEASE:=3 not work with openwrt master
the  miniupnpd PKG_VERSION:=2.2.1 PKG_RELEASE:=2 work with openwrt master  very well

BKPepe, 2021-08-03:
Which test? Can you provide more details on what is happening with your environment?

openips, 2021-08-04:
thks your reply
my test img is TARGET_x86_64

with OpenWrt SNAPSHOT r16765-6f3a05ebb0 / LuCI Master git-21.124.24916-0faf9a4 and miniupnpd 2.2.1-2  
they work very well
[image]

with OpenWrt SNAPSHOT r17247-022d1fe80f / LuCI Master git-21.196.22411-7d9ab8d and miniupnpd 2.2.2-3
luci can not see any upnp clients 
[image]

all the configs is same

if you want something else ,tell me pls 
thks very much

BKPepe, 2021-08-04:
I am so sorry, but from the screenshots, it is not clear what is going on. I need to ask you for logs from **miniupnd**, which can be seen in **system log** or in `/var/log/messages`. Also, how did you configure miniupnpd can be helpful as well.

faust2199, 2021-08-05:
> thks your reply
> my test img is TARGET_x86_64
>
> with OpenWrt SNAPSHOT r16765-6f3a05ebb0 / LuCI Master git-21.124.24916-0faf9a4 and miniupnpd 2.2.1-2
> they work very well
> [image]
>
> with OpenWrt SNAPSHOT [...]

What's your WAN IP address? Is it a private address (10.0.0.0 – 10.255.255.255, 172.16.0.0 – 172.31.255.255, 192.168.0.0 – 192.168.255.255 or 100.64.0.0 to 100.127.255.255)?

openips, 2021-08-07:
I have 3 wan. one have public ipv4  ,the other 2 have private ipv4,and 1 public ipv6 
the upnp use the port with 1 privat ipv4 and 1 public ipv6 to work
[image]
[image]

openips, 2021-08-07:
my upnpd

config upnpd 'config'
        option download '1024'
        option upload '512'
        option internal_iface 'lan'
        option port '5000'
        option upnp_lease_file '/var/run/miniupnpd.leases'
        option ext_ip_reserved_ignore '1'
        option enabled '1'
        option uuid 'b12a26f3-e262-4cd3-978b-2f67e8b17c85'
        option igdv1 '1'

config perm_rule
        option action 'allow'
        option ext_ports '1024-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '1024-65535'
        option comment 'Allow high ports'

config perm_rule
        option action 'deny'
        option ext_ports '0-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '0-65535'
        option comment 'Default deny'

openips, 2021-08-07:
with OpenWrt SNAPSHOT r16765-6f3a05ebb0 / LuCI Master git-21.124.24916-0faf9a4 and miniupnpd 2.2.1-2 
they work very well 
miniupnpd log is here
[upnp_ok.log](https://github.com/openwrt/packages/files/6948730/upnp_ok.log)

openips, 2021-08-07:
with OpenWrt SNAPSHOT r17247-022d1fe80f / LuCI Master git-21.196.22411-7d9ab8d and miniupnpd 2.2.2-3
luci can not see any upnp clients
miniupnpd log is here
[upnp_false.log](https://github.com/openwrt/packages/files/6948743/upnp_false.log)

openips, 2021-08-07:
if you want something else ,tell me pls
thks again

faust2199, 2021-08-08:
> with OpenWrt SNAPSHOT r17247-022d1fe80f / LuCI Master git-21.196.22411-7d9ab8d and miniupnpd 2.2.2-3
> luci can not see any upnp clients
> miniupnpd log is here
> [upnp_false.log](https://github.com/openwrt/packages/files/6948743/upnp_false.log)

"option ext_ip_reserved_ignore '1'" no long works as of miniupnpd 2.2.2-3 due to this commit: https://github.com/openwrt/packages/commit/107f3376b5114cd17c115e25026b031bd439e9be. The author of that commit may be able to suggest a way to make it work again. Currently I have to use my own build, with the miniupnpd patch added back, to preserve the previous behavior.

openips, 2021-08-15:
> I am so sorry, but from the screenshots, it is not clear what is going on. I need to ask you for logs from **miniupnd**, which can be seen in **system log** or in `/var/log/messages`. Also, how did you configure miniupnpd can be helpful as well.

the log is here

Neustradamus, 2021-08-17:
To follow this bug

BKPepe, 2021-08-17:
@openips The necessary details are included once miniupnd starts and that's missing.  Feel free to include it and consult it with miniupnd developers. Also, it is missing the content of configuration file ``miniupnpd.conf``. Which should be generated dynamically before the start.

BKPepe, 2021-08-17:
@openips Also, you can include output from ``ip address show`` to see how do you have configured your network to be able to help you more.

openips, 2021-08-17:
> @openips Also, you can include output from ``ip address show`` to see how do you have configured your network to be able to help you more.

@BKPepe  I have provided it above, please check

openips, 2021-08-17:
> my upnpd config
> 
> config upnpd 'config'
>         option download '1024'
>         option upload '512'
>         option internal_iface 'lan'
>         option port '5000'
>         option upnp_lease_file '/var/run/miniupnpd.leases'
>         option ext_ip_reserved_ignore '1'
>         option enabled '1'
>         option uuid 'b12a26f3-e262-4cd3-978b-2f67e8b17c85'
>         option igdv1 '1'
> 
> config perm_rule
>         option action 'allow'
>         option ext_ports '1024-65535'
>         option int_addr '0.0.0.0/0'
>         option int_ports '1024-65535'
>         option comment 'Allow high ports'
> 
> config perm_rule
>         option action 'deny'
>         option ext_ports '0-65535'
>         option int_addr '0.0.0.0/0'
>         option int_ports '0-65535'
>         option comment 'Default deny'
> 

@BKPepe

openips, 2021-08-17:
> I have 3 wan. one have public ipv4  ,the other 2 have private ipv4,and 1 public ipv6 
> the upnp use the port with 1 privat ipv4 and 1 public ipv6 to work
> [image]
> [image]
>  

@BKPepe

BKPepe, 2021-08-17:
Well, I am still missing the `miniupnd` log from start to know something more. Otherwise, it would help if you reached miniupnd developers regarding it like @pali. OpenWrt does not have any maintainer for this package and the patch there was a hack, which is not accepted here as it was rejected in upstream. Because of that, I'm closing it here.

Feel free to continue it here:
https://github.com/miniupnp/miniupnp/issues/556

Next time, include outputs and format it as code instead of screenshots. It makes it hard to find there anything as you can not search in it, etc.

closed 2021-08-17 by BKPepe

Neustradamus, 2021-08-18:
Thomas BERNARD, the author of @miniupnp project must read it.

Neustradamus, 2021-08-27:
@ all: @pali has done some comments on the miniupnp ticket from @openips.
- https://github.com/miniupnp/miniupnp/issues/556

The problem is not in miniupnp.

@openips has created a new ticket here:
- https://github.com/openwrt/packages/issues/16480

Can you look?

openips, 2021-08-27:
> @ all: @pali has done some comments on the miniupnp ticket from @openips.
> 
> * [The latest updated version does not work properly.There are no active redirects. miniupnp/miniupnp#556](https://github.com/miniupnp/miniupnp/issues/556)
> 
> The problem is not in miniupnp.
> 
> @openips has created a new ticket here:
> 
> * [miniupnpd:  The latest updated version does not work properly beasue of detected wrong wan port #16480](https://github.com/openwrt/packages/issues/16480)
> 
> Can you look?

thks

openips, 2021-08-27:
> > with OpenWrt SNAPSHOT r17247-022d1fe80f / LuCI Master git-21.196.22411-7d9ab8d and miniupnpd 2.2.2-3
> > luci can not see any upnp clients
> > miniupnpd log is here
> > [upnp_false.log](https://github.com/openwrt/packages/files/6948743/upnp_false.log)
> 
> "option ext_ip_reserved_ignore '1'" no longer works as of miniupnpd 2.2.2-3 due to this commit: [107f337](https://github.com/openwrt/packages/commit/107f3376b5114cd17c115e25026b031bd439e9be). The author of that commit may be able to suggest a way to make it work again. Currently I have to use my own build, with the miniupnpd patch added back, to preserve the previous behavior.

could you share your own patch？ thks

Neustradamus, 2021-08-31:
@BKPepe, @jow-, @nbd168, @neheb, @hnyman, @hauke, @antonlacon, @champtar, @ldir-EDB0, @ja-pa, @dengqf6, @ehrmann, @axet, @ptpt52, @MarcoMartins86, @dianlujitao, @wongsyrone, @stintel, @lipnitsk, @stangri, @dbermond, @sbyx, @moonchen, @mattsm, @faust2199: Can you look this problem?

It will be solved before 21.x official release?

The problem is confirmed in OpenWrt:
- https://github.com/miniupnp/miniupnp/issues/556
- https://github.com/openwrt/packages/issues/16480

Thanks in advance.

wongsyrone, 2021-08-31:
The root cause: https://github.com/miniupnp/miniupnp/issues/333
IMO, only Chinese ISPs distribute private IP as WAN IP through PPPoE.

wongsyrone, 2021-08-31:
I think you should request public IP from the carrier or enable UPNP on the ONU (Optical Network Unit). 

I find someone mentions Port Control Protocol (PCP): https://www.reddit.com/r/HomeNetworking/comments/cx1rl3/i_am_on_carriergrade_nat_cgn_and_port_forwarding/

> 2.1.  Deployment Scenarios
> 
>    PCP can be used in various deployment scenarios, including:
> 
>    o  Basic NAT [RFC3022]
> 
>    o  Network Address and Port Translation [RFC3022], such as commonly
>       deployed in residential NAT devices
> 
>    o  Carrier-Grade NAT [RFC6888]
> 
>    o  Dual-Stack Lite (DS-Lite) [RFC6333]
> 
>    o  NAT that is Layer-2 Aware [L2NAT]
> 
>    o  Dual-Stack Extra Lite [RFC6619]
> 
>    o  NAT64, both Stateless [RFC6145] and Stateful [RFC6146]
> 
>    o  IPv4 and IPv6 simple firewall control [RFC6092]
> 
>    o  IPv6-to-IPv6 Network Prefix Translation (NPTv6) [RFC6296]
> 

You can find more here: https://datatracker.ietf.org/doc/html/rfc6887#section-2.1

openips, 2021-09-02:
> The root cause: [miniupnp/miniupnp#333](https://github.com/miniupnp/miniupnp/issues/333)
> IMO, only Chinese ISPs distribute private IP as WAN IP through PPPoE.
my bug is miniupnpd detected wrong wan port 

I have 3 wan. one have public ipv4 ,the other 2 have private ipv4,and 1 public ipv6
the upnp use the port with 1 privat ipv4 and 1 public ipv6 to work

Neustradamus, 2021-09-11:
Dear all @openwrt guys, have you found the problem?
- https://github.com/openwrt/packages/issues/16480
- It is not a miniupnp bug: https://github.com/miniupnp/miniupnp/issues/556

Neustradamus, 2021-09-16:
There is a new patch from @ptpt52:
- https://github.com/openwrt/packages/pull/16011#issuecomment-920538698

It replaces the old 301-ext_ip_reserved_ignore.patch.

Tested by @openips!