openwrt/packages #16271: openblas: failed to build with ccache

1715173329, 2021-08-01:
Maintainer: @commodo
Environment: rockchip

Description:
An error occurred with ccache enabled:
```
touch /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.prepared_383546e81b293a74c89507494dbcce09_6664517399ebbbc92a37c5bb081b5c53_check
. /mnt/openwrt-snapshot/include/shell.sh; gzip -dc /mnt/openwrt-snapshot/dl/OpenBLAS-0.3.15.tar.gz | tar -C /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15
touch /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.prepared_383546e81b293a74c89507494dbcce09_6664517399ebbbc92a37c5bb081b5c53
rm -f /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.configured_*
rm -f /mnt/openwrt-snapshot/staging_dir/target-aarch64_generic_musl/stamp/.openblas_installed
(cd /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/./; if [ -x ./configure ]; then find /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/ -name config.guess | xargs -r chmod u+w; find /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/ -name config.guess | xargs -r -n1 cp --remove-destination /mnt/openwrt-snapshot/scripts/config.guess; find /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/ -name config.sub | xargs -r chmod u+w; find /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/ -name config.sub | xargs -r -n1 cp --remove-destination /mnt/openwrt-snapshot/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include " LDFLAGS="-L/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.built
touch /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/.built_check
CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include " LDFLAGS="-L/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/mnt/openwrt-snapshot/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro " make  -C /mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15=OpenBLAS-0.3.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" CROSS=1 HOSTCC=ccache gcc CROSS_SUFFIX=aarch64-openwrt-linux-musl- BINARY=64 NUM_THREADS=2 PREFIX=/usr COMMON_OPT="" TARGET=ARMV8 NOFORTRAN=1 ;
make[3]: Entering directory '/mnt/openwrt-snapshot/build_dir/target-aarch64_generic_musl/OpenBLAS-0.3.15'
ccache: invalid option -- 'D'
Usage:
    ccache [options]
    ccache compiler [compiler options]
    compiler [compiler options]          (via symbolic link)

Common options:
    -c, --cleanup              delete old files and recalculate size counters
                               (normally not needed as this is done
                               automatically)
    -C, --clear                clear the cache completely (except configuration)
        --config-path PATH     operate on configuration file PATH instead of the
                               default
    -d, --directory PATH       operate on cache directory PATH instead of the
                               default
        --evict-older-than AGE remove files older than AGE (unsigned integer
                               with a d (days) or s (seconds) suffix)
    -F, --max-files NUM        set maximum number of files in cache to NUM (use
                               0 for no limit)
    -M, --max-size SIZE        set maximum size of cache to SIZE (use 0 for no
                               limit); available suffixes: k, M, G, T (decimal)
                               and Ki, Mi, Gi, Ti (binary); default suffix: G
    -X, --recompress LEVEL     recompress the cache to level LEVEL (integer or
                               "uncompressed") using the Zstandard algorithm;
                               see "Cache compression" in the manual for details
    -o, --set-config KEY=VAL   set configuration item KEY to value VAL
    -x, --show-compression     show compression statistics
    -p, --show-config          show current configuration options in
                               human-readable format
    -s, --show-stats           show summary of configuration and statistics
[... cut ...]
make: *** [/mnt/openwrt-snapshot/include/toplevel.mk:230: package/feeds/packages/openblas/compile] Error 2

```

My workaround is disabling ccache for HOSTCC:
```patch
diff --git a/libs/openblas/Makefile b/libs/openblas/Makefile
index 4fc8eb2a0..8cb2f8ab6 100644
--- a/libs/openblas/Makefile
+++ b/libs/openblas/Makefile
@@ -78,7 +78,7 @@ endif
 
 MAKE_FLAGS += \
 	CROSS=1 \
-	HOSTCC=$(HOSTCC) \
+	HOSTCC=$(HOSTCC_NOCACHE) \
 	CROSS_SUFFIX=$(TARGET_CROSS) \
 	BINARY=$(OPENBLAS_BINARY) \
 	NUM_THREADS=2 \
```

commodo, 2021-08-02:
> My workaround is disabling ccache for HOSTCC:

That may actually be the fix :p
Thank you for confirming it works with ccache.

I need to updated OpenBLAS anyway this week.
Will include this in the update.

1715173329, 2021-08-02:
> I need to updated OpenBLAS anyway this week.
Will include this in the update.

Okay, thank you so much!

commodo, 2021-08-02:
opened https://github.com/openwrt/packages/pull/16276

closed 2021-08-02 by BKPepe