openwrt/packages #16297: FeatureRequest: Drop Open Uplinks

Massimo-B, 2021-08-06:
Hi,

please add an option to drop automatically added Open Uplinks after a configuration timeout.

When using the option "Add Open Uplinks" the list will be extended a lot when driving by car. This is quite useful but the list should drop unused uplinks when not being seen for a while and only if those have been added automatically.

feckert, 2021-08-17:
I think you have to be more specific about what you want.
Which LuCI application are you talking about?

Massimo-B, 2021-08-19:
Sorry, this feature request was meant for package travelmate. I just used the link on https://openwrt.org/packages/pkgdata/travelmate

dibdot, 2021-08-22:
What are "unused uplinks"? A time based fifo logic to drop open uplinks is not possible due to the absent of a reliable clock (rtc). Anyway, since travelmate 2.0.5 you can limit the max. amount of autoadded open uplinks.

closed 2021-08-22 by dibdot

Massimo-B, 2021-08-24:
OpenWrt has no reliable RTC? My Netgear R7800 has a reliable $(date) when running. Or you you mean when restarting?

travelmate should mark which uplinks are auto-added. Those uplinks could be just all dropped (as workaround) after a restart with no valid clock until NTP has synced. When up and running, it will re--auto-add those uplinks anyway when visible. But while driving around with a running travelmate, it is going to add a huge list of open uplinks on the way. This is good and expected, but should also clean the list after configurable timeout when uplinks are not seen anymore.

That workaround dropping all uplinks after restart is only required if there is no working uplink on the way to get a synced clock via NTP. What do you think?

Massimo-B, 2021-09-01:
Could you please describe why dropping unused uplinks isn't possible because of an unreliable clock and why the clock is not reliable?

dibdot, 2021-09-01:
RTC stands for (battery buffered) Real Time Clock - and that's not available ... at least on travel router that I've seen so far. Without that date/time is not a reliable source to set a starting point - ntp requires a working internet connection and is of no help here.

Massimo-B, 2021-09-02:
I that case, the design could be like this: Every reboot, automatically added uplinks are removed. While running, the uplinks could use the temporary clock and being removed after a "not used" timeout.
This is what I actually do every day, deleting hundreds of uplinks. This of course only happens when moving, not when staying at one location.

Of course this feature should just be an option to be enabled. This would be very useful when moving the location a lot (called travelling) and just matching the overall use case of travelmate.

dibdot, 2021-09-02:
I don't plan to implement such "feature". As wrote before in the current travelmate version you can limit the auto-added uplinks (default 5). Travelmate is not made for "war driving" etc. and I refuse to delete dozens of wireless & travelmate config entries during boot - "not used" is too vague a criterion.

Massimo-B, 2021-09-06:
About the limit feature, does that mean after auto-adding 5 (default) open uplinks, travelmate stops adding new ones, even after restart? So it stores which uplinks have been auto-added?

I know that travelmate needs to adapt wireless config entries in the back which may be expensive. But it's the way it already works. The bad thing are increasing configurations with old obsolete uplinks that may affect OpenWrt performance at some point.

I don't call using uplinks "war driving" as I don't create any maps with location information or publish any stuff. I'm just using open uplinks just like I can configure any Android phone to just use open uplinks. Is that already "war driving"?
To be honest, travelmate already does so, it adds open uplinks automatically, that is what it is meant for and which is an important feature for me, when traveling. That's the root of the name travelmate.
Because I'm not interested in logging or archiving any open uplinks of the past weeks (which may lead to "war driving"), there is only one option missing: Dropping auto-added open uplinks again and letting travelmate continue adding new ones.
Adding such an option doesn't do any harm from my point of view, nothing more serious than it already does by adding open uplinks.

ToeiRei, 2023-03-17:
> I don't plan to implement such "feature". As wrote before in the current travelmate version you can limit the auto-added uplinks (default 5). Travelmate is not made for "war driving" etc. and I [...]

I do travel a lot and I have a bunch of hotel networks saved that are basically 'once used' - same with Restaurants, Railway stations, etc. - Every now and then I have to spend hours as well to get rid of those networks to keep my list 'manageable'.