openwrt/packages #16298: mwan3: httping method fails to return correct status but manual run works properly

jike212, 2021-08-07:
Maintainer: @feckert 
Environment:  TP-Link C2600, OpenWrt 19.07, mwan3 2.8.16-1

**Description**:
When enabling `httping` tracking on my primary interface (due to ICMP not working on my network) all attempts that mwan3tracking makes fail (and are reflected as failed in the system log)

However, when I run `httping` manually to the same tracking IP from the router itself, the attempts succeed and indicate that it is indeed working.

This in turn creates a state where the interface is incorrectly marked as offline. I've tried using the `nping` package as well however all attempts with that package appear to fail which appears to be a unique issue to my ISP but prevents me from using it instead of `httping`.

**Possible location of the issue:**
Looking through the `mwan3track` source code, I see the line that populates the command (my tracking IP does not use SSL):
https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L291

When I replace the variables with the actual values from my config and run it manually from within the router, the command appears to work (I remove the `-q` flag so I can observe the output in the terminal)

When mwan3 runs the command however, the syslog indicates that the command was not successful.

Below is my mwan3 config for the interface (I'm only testing with one tracking IP for the time being):
```
config interface 'wan'
        option enabled '1'
        option family 'ipv4'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '2'
        option up '2'
        option failure_interval '3'
        option recovery_interval '3'
        option keep_failure_interval '1'
        option reliability '1'
        option initial_state 'online'
        list track_ip 'xxx.xxx.xxx.xxx'
        option track_method 'httping'
        list flush_conntrack 'ifup'
        list flush_conntrack 'ifdown'
```

feckert, 2021-08-12:
This is an issue in openwrt-19.07 and does not work as expected. Could you please check if your setup works with the upcoming openwrt-21.02 stable release?

jike212, 2021-08-13:
Hi @feckert, thank you for responding. I'll need to wait until 21.02 final is released to deploy it on our router as I need to minimize as much downtime as possible. Hopefully it will be released soon!

feckert, 2023-05-03:
We are now on 22.03 and missing respond from @jike212 so I assume this issue is fixed

closed 2023-05-03 by feckert