openwrt/packages #16303: tor doesn't start since commit ca6528f002d74445e3d0a336aeb9074fc337307a

e9hack, 2021-08-08:
Maintainer:
Hauke Mehrtens <hauke@hauke-m.de>
Peter Wagner <tripolar@gmx.at>

Environment:
TP-Link Archer C7, OpenWrt master

Description:
tor doesn't start since commit ca6528f002d74445e3d0a336aeb9074fc337307a:

tor: update init script

Changes:
-add uci config
-create pid file by procd
-add reload function

If logging is activated I got the following message:

Aug 08 09:36:39.000 [warn] Unable to open "/var/run/tor.pid" for writing: Permission denied
Aug 08 09:36:39.000 [err] Unable to write PIDFile "/var/run/tor.pid"
Aug 08 09:36:39.000 [err] set_options(): Bug: Acting on config options left us in a broken state. Dying. (on Tor 0.4.6.6 )
Aug 08 09:36:39.000 [err] Reading config failed--see warnings above.

Tor isn't able to create the pid file and terminates. The reload function does also not work. This does fix the issues:

```
diff --git a/net/tor/files/tor.init b/net/tor/files/tor.init
index 142e2512d..33794daa4 100644
--- a/net/tor/files/tor.init
+++ b/net/tor/files/tor.init
@@ -32,10 +32,13 @@ generate_conf() {
 }

 reload_service() {
-       procd_send_signal /usr/sbin/tor
+       procd_send_signal tor
 }

 start_service() {
+       touch /var/run/tor.pid
+       chown tor:tor /var/run/tor.pid
+
        mkdir -m 0700 -p /var/lib/tor
        chown -R tor:tor /var/lib/tor
```

Wagnerp, 2021-08-08:
@e9hack I think you've included the wrong person. I've never heard of this.

e9hack, 2021-08-08:
@Wagnerp Yes you are right. I mean another Peter Wagner.

Wagnerp, 2021-08-08:
@e9hack That's ok

neheb, 2021-08-12:
ping @ja-pa