openwrt/packages #16313: yubico-pam: pam_yubico.so path (e.g. for sshd)

l33tl4bs, 2021-08-10:
Environment: (mips, 19.07)

Description:

In case anyone has the same problem trying to get yubico OTP based login to work for SSH login:

- assuming you already installed openssh-server-pam and yubico-pam packages and configured PAM, SSH according to a guide like this: https://docs.termius.com/termius-handbook/yubikey-support
- the missing piece to get it to work on openwrt if you are getting this error message `Permission denied (keyboard-interactive)`  when trying to connect over ssh because of pam_yubico.so not being loaded is:
   - in /etc/pam.d/sshd, add the absolute path to pam_yubico.so
      - original (based on guide): `auth sufficient pam_yubico.so id=xxxxx debug authfile=/etc/yubico_pass` 
      - modified (openwrt support): `auth sufficient /lib/security/pam_yubico.so id=xxxxx debug authfile=/etc/yubico_pass`

dangowrt, 2021-08-29:
I wasn't aware that `libpam-yubico` is available in source. Where did you find the sourcecode to compile `pam_yubico.so` for MIPS target?

l33tl4bs, 2021-08-29:
See Sources link in: https://openwrt.org/packages/pkgdata/yubico-pam

It links to here:
https://developers.yubico.com/yubico-pam/Releases/

* I did not have to compile it myself as it’s already available in opkg