openwrt/packages #16344: Unclear&Unnecessary message

proukornew, 2021-08-12:
https://github.com/openwrt/packages/blob/4608a73bf0e9ff33abb2761aa23af3de7649e523/net/unbound/files/unbound.sh#L659
causes such config:
...
auth-zone:
  name: ip6.arpa.
  master: lax.xfr.dns.icann.org
  master: iad.xfr.dns.icann.org
  fallback-enabled: yes
  for-downstream: no
  for-upstream: yes
  zonefile: ip6.arpa.zone

 # Special zone  (noname) [in skip] was not enabled or had UCI conflicts.

 # Special zone  (noname) [in skip] was not enabled or had UCI conflicts.
...
skip == no skip

EricLuehrsen, 2021-08-17:
The message means what it says. It is about as clean as I could make it including ESL considerations. If you set UCI `enable 0` for the zone, or the zone definition cannot be unraveled (UCI-->Unbound), then this message appears. In particular, UCI form has compressed the expressions for what can be extremely verbose in `unbound.conf` form. Unbound will crash for bad `unbound.conf` file, and it doesn't have any graceful defaults. This may create a "soft brick" like experience with a router. So instead, these extra zones are just skipped and Unbound is allowed to start. Of course doing that creates a silent error. These messages ensure the condition is flagged in the system log.

EricLuehrsen, 2021-08-17:
The message appears in `unbound.conf` for two conditions
- you set UCI `enable 0` for the zone
- zone definition in UCI form will not unravel propperly into `unbound.conf` form

This is done because Unbound will crash for bad `unbound.conf` files and does not have some graceful default operation. This may be a "soft brick" experience for some router users. The zones are instead skipped. This would be a silent error. The comments are placed in `unbound.conf` as overt markers of where the zone configuration should be.

brianjmurrell, 2022-08-07:
How to know the cause of a _zone definition in UCI form will not unravel propperly into unbound.conf form_ though?  It would be useful if there were more detail about what failed in the unravelling.

brianjmurrell, 2022-08-07:
And it's worth noting that the `$zonename` in the comments is always empty.

brianjmurrell, 2022-12-22:
Anything on the above two comments?  I have in my `/var/lib/unbound/unbound.conf`:

```
# /var/lib/unbound/zone.conf.tmp generated by UCI 2022-12-22T09:06:50-05:00
 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.
```

which is a bit unhelpful as I don't even know which zones it's complaining about.  My unbound config is:

```
config unbound 'ub_main'
	option add_extra_dns '0'
	option add_local_fqdn '1'
	option add_wan_fqdn '0'
	option dhcp_link 'none'
	option dns64 '0'
	option domain 'lan'
	option domain_type 'static'
	option edns_size '1232'
	option extended_stats '0'
	option hide_binddata '1'
	option interface_auto '1'
	option listen_port '53'
	option localservice '1'
	option manual_conf '0'
	option num_threads '1'
	option protocol 'default'
	option rate_limit '0'
	option rebind_localhost '0'
	option rebind_protection '1'
	option recursion 'default'
	option resource 'default'
	option root_age '9'
	option ttl_min '120'
	option ttl_neg_max '1000'
	option validator '0'
	option verbosity '1'
	list iface_wan 'wan'
	option enabled '1'
	option unbound_control '1'
	option iface_trig 'lan wan'

config zone 'auth_icann'
	option enabled '0'
	option fallback '1'
	option url_dir 'https://www.internic.net/domain/'
	option zone_type 'auth_zone'
	list server 'lax.xfr.dns.icann.org'
	list server 'iad.xfr.dns.icann.org'
	list zone_name '.'
	list zone_name 'arpa.'
	list zone_name 'in-addr.arpa.'
	list zone_name 'ip6.arpa.'

config zone 'fwd_isp'
	option enabled '0'
	option fallback '1'
	option resolv_conf '1'
	option zone_type 'forward_zone'
	list zone_name 'isp-bill.example.com.'
	list zone_name 'isp-mail.example.net.'

config zone 'fwd_google'
	option enabled '0'
	option fallback '1'
	option tls_index 'dns.google'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '8.8.4.4'
	list server '8.8.8.8'
	list server '2001:4860:4860::8844'
	list server '2001:4860:4860::8888'
	list zone_name '.'

config zone 'fwd_cloudflare'
	option enabled '0'
	option fallback '1'
	option tls_index 'cloudflare-dns.com'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '1.1.1.1'
	list server '1.0.0.1'
	list server '2606:4700:4700::1111'
	list server '2606:4700:4700::1001'
	list zone_name '.'

config zone 'fwd_example_com'
	option fallback '0'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name 'example.com'
	option server '10.75.22.247'

config zone 'fwd_10_75_22'
	option fallback '1'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name '22.75.10.in-addr.arpa'
	option server '10.75.22.247'
```

Given the suggestion in the [above comment](https://github.com/openwrt/packages/issues/16344#issuecomment-900061760) I suppose my 4 `# Special zone  was not enabled or had UCI conflicts.` are for the four above zones set to `enabled '0'`.  But I cannot know for sure due to the zone name being missing.

~~But then also what about the two additional zones I have defined?  They are neither in my `/var/lib/unbound/unbound.conf` nor are there presumably any warnings about why they are missing.~~
Changing the `server` and `zone_name` settings above from `option` to `list` seems to have resolved why they were missing.  But I was always under the impression that a `list` of one and `option` were equivalent.

And then the question becomes how do I make those `list` using `uci`?  I'm using the following `uci batch` commands to add these zones:

```uci
set unbound.fwd_example_com=zone
set unbound.fwd_example_com.fallback='0'
set unbound.fwd_example_com.enabled='1'
set unbound.fwd_example_com.zone_type='forward_zone'
set unbound.fwd_example_com.zone_name='example.com'
set unbound.fwd_example_com.server='10.75.22.247'
set unbound.fwd_10_75_22=zone
set unbound.fwd_10_75_22.fallback='1'
set unbound.fwd_10_75_22.enabled='1'
set unbound.fwd_10_75_22.zone_type='forward_zone'
set unbound.fwd_10_75_22.zone_name='22.75.10.in-addr.arpa'
set unbound.fwd_10_75_22.server='10.75.22.247'
```

and that is what is setting `server` and `zone_name` to `option`s and not `list`s.