openwrt/packages #16370: modemmanager errors on 21.02-RC4

erdoukki, 2021-08-15:
Maintainer: @nickberry17
Nicholas Smith <nicholas.smith@telcoantennas.com.au>
Environment: 
Model | Globalscale Marvell ESPRESSOBin Ultra Board
-- | --
Architecture | ARMv8 Processor rev 4
Firmware Version | OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6 / LuCI openwrt-21.02 branch git-21.188.55099-e52e1de
Kernel Version | 5.4.137

I get recently issues with modemmanager...
Reinstalling do not help:
```
root@ULTRA-5G:~# opkg info modemmanager
Package: modemmanager
Version: 1.16.2-1
Depends: libc, glib2, dbus, ppp, libmbim, libqmi
Status: install user installed
Architecture: aarch64_cortex-a53
Installed-Time: 1629006663
```

service restart loop in buggy state !
Luci Network do not show device !

may be like I have an environment bug, a ps shows strange state of modemanager inti.d !

Description:
```
 9207 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p ;                                          mm_clean
10025 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p ;                                          mm_clean
10809 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p ;                                          mm_clean
```
sometimes it loads correctly but show strange errors in logread, which fall in timeout : 
```
Sun Aug 15 08:29:19 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Aug 15 08:29:19 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=mlan0, subsystem=net, sysfspath=/sys/devices/platform/soc/d0070000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/mlan0
Sun Aug 15 08:29:19 2021 user.notice ModemManager: hotplug: event reported: action=add, name=mlan0, subsystem=net
Sun Aug 15 08:29:19 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Aug 15 08:29:19 2021 daemon.warn [15675]: <warn>  [mlan0] invalid sysfs path read for net/mlan0
Sun Aug 15 08:29:19 2021 daemon.info procd: Instance modemmanager::instance1 s in a crash loop 6 crashes, 3 seconds since last crash
Sun Aug 15 08:29:19 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:21 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:21 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:22 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:23 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:24 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:24 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:26 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:26 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:27 2021 user.notice ModemManager: hotplug: error: modem not detected at sysfs path
Sun Aug 15 08:29:28 2021 user.notice ModemManager: hotplug: error: timed out waiting for the modem to get exported at /sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3
```

I have installed watchcat, and use a script for wanmonitor.
This issue appear recently...
I have issue for detecting the modem (LUCI error) and have add it in UCI/shell with `mmcli -L` and `mmcli -m 0`
```
config interface 'wwan'
	option lowpower '1'
	option proto 'modemmanager'
	option device '/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3'
	option apn 'free'
	option auth 'none'
	option pdptype 'ipv4'
```

```
root@ULTRA-5G:~# ls /var/run/modemmanager/ -hal
drwxr-xr-x    2 root     root         120 Aug 15 08:29 .
drwxr-xr-x    8 root     root         380 Aug 15 08:22 ..
-rw-r--r--    1 root     root          15 Aug 15 08:21 cdcwdm.cache
-rw-r--r--    1 root     root        2.0K Aug 15 08:21 events.cache
-rw-r--r--    1 root     root           6 Aug 15 08:29 modemmanager.pid
-rw-r--r--    1 root     root          91 Aug 15 08:29 sysfs.cache
```

```
root@ULTRA-5G:~# lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux 5.4.137 xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 2c7c:0306 Quectel EP06-E
Bus 001 Device 002: ID 1a40:0101  USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.137 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0003 Linux 5.4.137 xhci-hcd xHCI Host Controller
```

erdoukki, 2021-08-15:
> I have installed watchcat, and use a script for wanmonitor.

removed from startup and disabled for testing...
Same issue !

multiple init started in ps:
```
26987 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p /var/run/modemmanager;                             
27915 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p /var/run/modemmanager;                             
28797 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p /var/run/modemmanager;                             
29721 root      1428 S    sh -c . /usr/share/ModemManager/modemmanager.common;                                    mkdir -m 0755 -p /var/run/modemmanager;                             
```

erdoukki, 2021-08-15:
```
root@ULTRA-5G:~# ls /var/run/modemmanager/
cdcwdm.cache      events.cache      modemmanager.pid  sysfs.cache
```

```
root@ULTRA-5G:~# cat /var/run/modemmanager/cdcwdm.cache 
wwan0 cdc-wdm0
```

```
root@ULTRA-5G:~# cat /var/run/modemmanager/events.cache 
add,ttyS15,tty,/sys/devices/platform/serial8250/tty/ttyS15
add,ttyS6,tty,/sys/devices/platform/serial8250/tty/ttyS6
add,ttyS13,tty,/sys/devices/platform/serial8250/tty/ttyS13
add,console,tty,/sys/devices/virtual/tty/console
add,ttyS4,tty,/sys/devices/platform/serial8250/tty/ttyS4
add,ttyS11,tty,/sys/devices/platform/serial8250/tty/ttyS11
add,ttyS2,tty,/sys/devices/platform/serial8250/tty/ttyS2
add,ttyS0,tty,/sys/devices/platform/serial8250/tty/ttyS0
add,ttyS9,tty,/sys/devices/platform/serial8250/tty/ttyS9
add,ttyS7,tty,/sys/devices/platform/serial8250/tty/ttyS7
add,ttyS14,tty,/sys/devices/platform/serial8250/tty/ttyS14
add,ttyS5,tty,/sys/devices/platform/serial8250/tty/ttyS5
add,ttyMV0,tty,/sys/devices/platform/soc/soc:internal-regs@d0000000/d0012000.serial/tty/ttyMV0
add,ttyS12,tty,/sys/devices/platform/serial8250/tty/ttyS12
add,ttyS3,tty,/sys/devices/platform/serial8250/tty/ttyS3
add,ttyS10,tty,/sys/devices/platform/serial8250/tty/ttyS10
add,ttyS1,tty,/sys/devices/platform/serial8250/tty/ttyS1
add,ttyS8,tty,/sys/devices/platform/serial8250/tty/ttyS8
add,br-lan,net,/sys/devices/virtual/net/br-lan
add,wlan0,net,/sys/devices/platform/soc/d0070000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/wlan0
add,mlan0,net,/sys/devices/platform/soc/d0070000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/mlan0
add,ttyUSB0,tty,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.0/ttyUSB0/tty/ttyUSB0
add,ttyUSB1,tty,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.1/ttyUSB1/tty/ttyUSB1
add,ttyUSB2,tty,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.2/ttyUSB2/tty/ttyUSB2
add,ttyUSB3,tty,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.3/ttyUSB3/tty/ttyUSB3
add,wwan0,net,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.4/net/wwan0
add,cdc-wdm0,usbmisc,/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3/1-1.3:1.4/net/wwan0
```

```
root@ULTRA-5G:~# cat /var/run/modemmanager/sysfs.cache 
/sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3 processed
```

erdoukki, 2021-08-15:
works (partially, but better) in multiple command-lines:
In first terminal
```
root@ULTRA-5G:~# mmcli -M
No modems were found
(+) /org/freedesktop/ModemManager1/Modem/0 [Quectel] EP06-E
(-) /org/freedesktop/ModemManager1/Modem/0 [Quectel] EP06-E
(+) /org/freedesktop/ModemManager1/Modem/0 [Quectel] EP06-E
(-) /org/freedesktop/ModemManager1/Modem/0 [Quectel] EP06-E
(+) /org/freedesktop/ModemManager1/Modem/1 [Quectel] EP06-E
```
In second terminal
```
root@ULTRA-5G:~# mmcli --debug
or
root@ULTRA-5G:~# mmcli --debug
```
sending AT command to change from QMI (QMI do not works !?) to MBMI (partially works) and IPV4V6 to IPV4 (IPV4 do not work with IPV4V6 !?)

Lot of issues here... sorry...hope I can help to fix them !

erdoukki, 2021-08-15:
```
[11031]: <warn>  [modem0] couldn't query SIM slots: QMI protocol error (94): 'NotSupported'
[11031]: <warn>  [modem0/sim0] couldn't load list of emergency numbers: Failed to parse CRSM query result '+CRSM: 148,8,""'
```

erdoukki, 2021-08-15:
```
root@ULTRA-5G:~# mmcli -m 0 --simple-connect="apn=Free"
error: couldn't connect the modem: 'Timeout was reached'
```

```
[16429]: <info>  [modem0] simple connect started...
[16429]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
[16429]: <info>  [modem0] simple connect state (5/8): register
[16429]: <info>  [modem0] 3GPP registration state changed (idle -> denied)
[16429]: <info>  [modem0] 3GPP registration state changed (denied -> idle)
```

erdoukki, 2021-08-15:
in QMI mode:
```
Sun Aug 15 10:43:26 2021 daemon.notice netifd: wwan (24148): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.NetworkTimeout: Network timeout'
```

erdoukki, 2021-08-15:
```
[11827]: <info>  [modem0] simple connect started...
[11827]: <info>  [modem0] simple connect state (3/8): enable
[11827]: <info>  [modem0] state changed (disabled -> enabling)
[11827]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
[11827]: <info>  [modem0] state changed (enabling -> enabled)
[11827]: <info>  [modem0] simple connect state (5/8): register
[11827]: <info>  [modem0] 3GPP registration state changed (unknown -> registering)
[11827]: <info>  [modem0] 3GPP registration state changed (registering -> home)
[11827]: <info>  [modem0] state changed (enabled -> registered)
[11827]: <info>  [modem0] simple connect state (6/8): bearer
[11827]: <info>  [modem0] simple connect state (7/8): connect
[11827]: <info>  [modem0] state changed (registered -> connecting)
[11827]: <info>  [modem0/bearer3] couldn't start network: QMI protocol error (14): 'CallFailed'
[11827]: <info>  [modem0/bearer3] call end reason (1): generic-unspecified
[11827]: <info>  [modem0/bearer3] verbose call end reason (6,55): [3gpp] multiple-connection-to-same-pdn-not-allowed
[11827]: <warn>  [modem0/bearer3] connection attempt #1 failed: QMI protocol error (14): 'CallFailed'
[11827]: <info>  [modem0] state changed (connecting -> registered)
[11827]: <info>  [modem0/bearer3] connection #1 finished: duration 0s, tx: 0 bytes, rx :0 bytes
[11827]: <info>  [modem0] state changed (registered -> disabling)
[11827]: <info>  [modem0] 3GPP registration state changed (home -> unknown)
[11827]: <info>  [modem0] state changed (disabling -> disabled)
```

erdoukki, 2021-08-15:
MBIM Mode: 
```
[11827]: <info>  [modem1] simple connect started...
[11827]: <info>  [modem1] simple connect state (3/8): enable
[11827]: <info>  [modem1] state changed (disabled -> enabling)
[11827]: <warn>  [modem1] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem1] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem1] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem1] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <warn>  [modem1] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11827]: <info>  [modem1] simple connect state (4/8): wait to get fully enabled
[11827]: <info>  [modem1] state changed (enabling -> enabled)
[11827]: <info>  [modem1] simple connect state (5/8): register
[11827]: <info>  [modem1] 3GPP registration state changed (unknown -> registering)
[11827]: <info>  [modem1] 3GPP registration state changed (registering -> home)
[11827]: <info>  [modem1] state changed (enabled -> registered)
[11827]: <info>  [modem1] simple connect state (6/8): bearer
[11827]: <info>  [modem1] simple connect state (7/8): connect
[11827]: <info>  [modem1] state changed (registered -> connecting)
[11827]: <warn>  [modem1/bearer9] connection attempt #1 failed: Unsupported error (115): unknown
[11827]: <info>  [modem1] state changed (connecting -> registered)
[11827]: <info>  [modem1/bearer9] connection #1 finished: duration 0s, tx: 0 bytes, rx :0 bytes
[11827]: <info>  [modem1] state changed (registered -> disabling)
[11827]: <info>  [modem1] 3GPP registration state changed (home -> unknown)
[11827]: <info>  [modem1] state changed (disabling -> disabled)
```

erdoukki, 2021-08-15:
One success... in MBMI mode, with a reboot, starting ModemManager manually and sending a AT for REBOOT (to force detect the modem)...
Then:
```
(+) /org/freedesktop/ModemManager1/Modem/0 [Quectel] EP06-E
[11753]: <info>  [modem0] simple connect started...
[11753]: <info>  [modem0] simple connect state (3/8): enable
[11753]: <info>  [modem0] state changed (disabled -> enabling)
[11753]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11753]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11753]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11753]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11753]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
[11753]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
[11753]: <info>  [modem0] state changed (enabling -> enabled)
[11753]: <info>  [modem0] simple connect state (5/8): register
[11753]: <info>  [modem0] 3GPP registration state changed (unknown -> registering)
[11753]: <info>  [modem0] 3GPP registration state changed (registering -> home)
[11753]: <info>  [modem0] state changed (enabled -> registered)
[11753]: <info>  [modem0] 3GPP registration state changed (home -> idle)
[11753]: <info>  [modem0] state changed (registered -> enabled)
[11753]: <warn>  [modem0] couldn't load operator code: Current operator MCC/MNC is still unknown
[11753]: <warn>  [modem0] couldn't load operator name: Current operator name is still unknown
[11753]: <info>  [modem0] 3GPP registration state changed (idle -> searching)
[11753]: <info>  [modem0] state changed (enabled -> searching)
[11753]: <info>  [modem0] 3GPP registration state changed (searching -> registering)
[11753]: <info>  [modem0] 3GPP registration state changed (registering -> home)
[11753]: <info>  [modem0] state changed (searching -> registered)
[11753]: <info>  [modem0] simple connect state (6/8): bearer
[11753]: <info>  [modem0] simple connect state (7/8): connect
[11753]: <info>  [modem0] state changed (registered -> connecting)
[11753]: <info>  [modem0] state changed (connecting -> connected)
[11753]: <info>  [modem0] simple connect state (8/8): all done
```

erdoukki, 2021-08-15:
I have remove watchcat and luci-app-watchcat, and will reboot to see (test) !
my watchcat config was:
```
root@ULTRA-5G:~# cat /etc/config/watchcat

config watchcat
	option pinghosts '8.8.8.8'
	option mode 'restart_iface'
	option period '10m'
	option pingperiod '30s'
	option pingsize 'standard'
	option interface 'wwan0'
	option mmifacename 'wwan'

config watchcat
	option mode 'restart_iface'
	option period '1h'
	option pinghosts '8.8.8.8'
	option pingperiod '30s'
	option pingsize 'standard'
	option interface 'wan'
```

erdoukki, 2021-08-15:
I change the 21.02 init.d using the 19.07 script: 
https://github.com/openwrt/packages/blob/openwrt-19.07/net/modemmanager/files/modemmanager.init
Done a service modemanager restart => OK
Done a reboot... => no modem detected but service = OK
Done an AT to reboot card (and only that) => modem detected and network WWAN = (still) KO
```
Sun Aug 15 11:39:15 2021 daemon.notice netifd: wwan (9705): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.GprsUnknown: Unsupported error (55): unknown'
```

erdoukki, 2021-08-15:
```
root@ULTRA-5G:~# mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: c5daa1f9c488c79a76c0cfd75a1f0bba26717ed0
  ----------------------------------
  Hardware |           manufacturer: Quectel
           |                  model: EP06-E
           |      firmware revision: EP06ELAR04A04M4G
           |           h/w revision: EP06-E
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 868186040847129
  ----------------------------------
  System   |                 device: /sys/devices/platform/soc/soc:internal-regs@d0000000/d005e000.usb/usb1/1-1/1-1.3
           |                drivers: option1, cdc_mbim
           |                 plugin: quectel
           |           primary port: cdc-wdm0
           |                  ports: cdc-wdm0 (mbim), ttyUSB0 (qcdm), ttyUSB1 (gps), 
           |                         ttyUSB2 (at), ttyUSB3 (at), wwan0 (net)
  ----------------------------------
  Numbers  |                    own: +33749749712
  ----------------------------------
  Status   |         unlock retries: sim-pin2 (3)
           |                  state: disabled
           |            power state: on
           |         signal quality: 19% (cached)
  ----------------------------------
  Modes    |              supported: allowed: 3g, 4g; preferred: none
           |                current: allowed: 3g, 4g; preferred: none
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 868186040847129
           |          enabled locks: fixed-dialing
  ----------------------------------
  3GPP EPS |   ue mode of operation: ps-2
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
  ----------------------------------
  Bearer   |                  paths: /org/freedesktop/ModemManager1/Bearer/5

```

erdoukki, 2021-08-15:
```
root@ULTRA-5G:~# mmcli -m 0 --bearer 7
  ------------------------
  General    |       path: /org/freedesktop/ModemManager1/Bearer/7
             |       type: default
  ------------------------
  Status     |  connected: no
             |  suspended: no
             | ip timeout: 20
  ------------------------
  Properties |        apn: Free
             |    roaming: allowed
             |    ip type: ipv4
  ------------------------
  Statistics |   attempts: 1
             |   attempts: 1
```

erdoukki, 2021-08-15:
restart of the wwan interface from LUCI 
```
Sun Aug 15 11:50:57 2021 daemon.notice netifd: Interface 'wwan' is setting up now
Sun Aug 15 11:50:57 2021 daemon.notice netifd: wwan (10207): modem available at /org/freedesktop/ModemManager1/Modem/0
Sun Aug 15 11:50:57 2021 daemon.notice netifd: wwan (10207): starting connection with apn 'free'...
Sun Aug 15 11:50:57 2021 daemon.info [5902]: <info>  [modem0] simple connect started...
Sun Aug 15 11:50:57 2021 daemon.info [5902]: <info>  [modem0] simple connect state (3/8): enable
Sun Aug 15 11:50:57 2021 daemon.info [5902]: <info>  [modem0] state changed (disabled -> enabling)
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0] couldn't convert SMS part contents from UTF-16BE to UTF-8: not decoding any text
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] state changed (enabling -> enabled)
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] simple connect state (5/8): register
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] 3GPP registration state changed (unknown -> registering)
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] 3GPP registration state changed (registering -> home)
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] state changed (enabled -> registered)
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] simple connect state (6/8): bearer
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] simple connect state (7/8): connect
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] state changed (registered -> connecting)
Sun Aug 15 11:50:58 2021 daemon.warn [5902]: <warn>  [modem0/bearer9] connection attempt #1 failed: Unsupported error (55): unknown
Sun Aug 15 11:50:58 2021 daemon.info [5902]: <info>  [modem0] state changed (connecting -> registered)
Sun Aug 15 11:50:59 2021 daemon.info [5902]: <info>  [modem0/bearer9] connection #1 finished: duration 0s, tx: 0 bytes, rx :0 bytes
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10207): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.GprsUnknown: Unsupported error (55): unknown'
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10251): stopping network
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10251): running disconnection (common)
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10251): Command failed: Permission denied
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10251): successfully disconnected all bearers in the modem
Sun Aug 15 11:50:59 2021 daemon.info [5902]: <info>  [modem0] state changed (registered -> disabling)
Sun Aug 15 11:50:59 2021 daemon.info [5902]: <info>  [modem0] 3GPP registration state changed (home -> unknown)
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10251): successfully disabled the modem
Sun Aug 15 11:50:59 2021 daemon.notice netifd: Interface 'wwan' is now down
Sun Aug 15 11:50:59 2021 daemon.info [5902]: <info>  [modem0] state changed (disabling -> disabled)

```

erdoukki, 2021-08-15:
QMI with error:`
```
Sun Aug 15 11:50:59 2021 daemon.notice netifd: wwan (10207): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.GprsUnknown: Unsupported error (55): unknown'
```
```
 at+cpin?
+CPIN: READY
OK
```
```
 at+cops?
+COPS: 0,0,"Free Free",7
OK
```
```
 at+creg?
+CREG: 0,0
OK

```
```
 at+cgdcont?
+CGDCONT: 1,"IPV4V6","","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
+CGDCONT: 4,"IP","Free","0.0.0.0",0,0,0,0
OK
```
```
at+cgpaddr=1
+CGPADDR: 1,"42.13.228.135.1.79.246.237.0.0.0.0.0.235.137.160"
OK
```

erdoukki, 2021-08-15:
> `Sun Aug 15 11:39:15 2021 daemon.notice netifd: wwan (9705): error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.GprsUnknown: Unsupported error (55): unknown'`

```
at+cgpaddr=1
+ESM ERROR: 5,55
+CGPADDR: 1,"42.13.228.135.1.47.255.37.0.0.0.0.1.121.239.28"
OK
```

```
 at+cgdcont?
+CGDCONT: 1,"IPV4V6","","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
+CGDCONT: 4,"IP","Free","0.0.0.0",0,0,0,0
+CGDCONT: 5,"IP","free","0.0.0.0",0,0,0,0
OK
```

erdoukki, 2021-08-15:
>> http://howltestuffworks.blogspot.com/2012/02/pdn-connectivity-reject.html
```
 #55:  multiple PDN connections for a given APN not allowed 
```

erdoukki, 2021-08-16:
```
[6279]: <debug> [1629096899.482588] [mlan0] preloading contents and properties...
[6279]: <warn>  [1629096899.482759] [mlan0] invalid sysfs path read for net/mlan0
**
ERROR:kerneldevice/mm-kernel-device-generic.c:234:ptr_array_add_sysfs_attribute_link_basename: assertion failed: (array && sysfs_path && attribute)
Bail out! ERROR:kerneldevice/mm-kernel-device-generic.c:234:ptr_array_add_sysfs_attribute_link_basename: assertion failed: (array && sysfs_path && attribute)
Aborted
```

feckert, 2023-09-13:
Should be fixed with the libqmi changes made in openwrt-21.02

closed 2023-09-13 by feckert