openwrt/packages #16393: [19.07] https-dns-proxy: recompile with -fPIC

BKPepe, 2021-08-18:
Maintainer: @stangri 
Environment: aarch64_cortex-a53 (Turris MOX), OpenWrt 19.07.8

Config:
```
CONFIG_PKG_FORTIFY_SOURCE_2=y
CONFIG_PKG_ASLR_PIE_ALL=y
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
CONFIG_PKG_RELRO_FULL=y
```

Description:

```
[ 37%] Building C object CMakeFiles/https_dns_proxy.dir/src/https_client.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c: In function 'write_buffer':
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:53:30: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
     ELOG_REQ("Out of memory!");
                              ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c: In function 'https_fetch_ctx_process_response':
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:297:49: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
                  "but no response was sent back");
                                                 ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:307:82: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
         WLOG_REQ("No response (probably connection has been closed or timed out)");
                                                                                  ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:383:48: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
           DLOG_REQ("CURLINFO_HTTP_VERSION: 1.0");
                                                ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:386:48: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
           DLOG_REQ("CURLINFO_HTTP_VERSION: 1.1");
                                                ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:389:46: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
           DLOG_REQ("CURLINFO_HTTP_VERSION: 2");
                                              ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:420:38: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
       ELOG_REQ("Error getting timing");
                                      ^
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c: In function 'https_fetch_ctx_cleanup':
/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29/src/https_client.c:442:60: warning: ISO C99 requires at least one argument for the "..." in a variadic macro
         ILOG_REQ("Response was faulty, skipping DNS reply.");
                                                            ^
[ 50%] Building C object CMakeFiles/https_dns_proxy.dir/src/logging.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 62%] Building C object CMakeFiles/https_dns_proxy.dir/src/main.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 75%] Building C object CMakeFiles/https_dns_proxy.dir/src/options.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 87%] Building C object CMakeFiles/https_dns_proxy.dir/src/stat.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[100%] Linking C executable https_dns_proxy
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: CMakeFiles/https_dns_proxy.dir/src/options.c.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `optarg' which may bind externally can not be used when making a shared object; recompile with -fPIC
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: CMakeFiles/https_dns_proxy.dir/src/options.c.o(.text+0x120): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `optarg'
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
```

BKPepe, 2021-08-18:
This also affects OpenWrt 21.02 and OpenWrt daily snapshots.

21.02 log:
```
-- The C compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /mox/build/staging_dir/host/bin/git (found version "2.20.1") 
-- Version: 2021.08.18-da3da22e1f
-- clang-tidy not found.
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29
make[3]: Entering directory '/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
make[4]: Entering directory '/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
make[5]: Entering directory '/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
Scanning dependencies of target https_dns_proxy
make[5]: Leaving directory '/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
make[5]: Entering directory '/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
[ 12%] Building C object CMakeFiles/https_dns_proxy.dir/src/dns_poller.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 25%] Building C object CMakeFiles/https_dns_proxy.dir/src/dns_server.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 37%] Building C object CMakeFiles/https_dns_proxy.dir/src/https_client.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 50%] Building C object CMakeFiles/https_dns_proxy.dir/src/logging.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 62%] Building C object CMakeFiles/https_dns_proxy.dir/src/main.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 75%] Building C object CMakeFiles/https_dns_proxy.dir/src/options.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[ 87%] Building C object CMakeFiles/https_dns_proxy.dir/src/stat.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[100%] Linking C executable https_dns_proxy
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: CMakeFiles/https_dns_proxy.dir/src/options.c.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `optarg' which may bind externally can not be used when making a shared object; recompile with -fPIC
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: CMakeFiles/https_dns_proxy.dir/src/options.c.o(.text+0x120): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `optarg'
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/https_dns_proxy.dir/build.make:193: https_dns_proxy] Error 1
```

daily snapshots (master branch):

```
-- The C compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /mox/build/staging_dir/host/bin/git (found version "2.20.1") 
[... 20 lines cut ...]
-- Build files have been written to: /mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29
ninja: Entering directory `/mox/build/build_dir/target-aarch64_cortex-a53_musl/https-dns-proxy-2021-07-29'
[1/8] Building C object CMakeFiles/https_dns_proxy.dir/src/dns_poller.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[2/8] Building C object CMakeFiles/https_dns_proxy.dir/src/dns_server.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[3/8] Building C object CMakeFiles/https_dns_proxy.dir/src/https_client.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[4/8] Building C object CMakeFiles/https_dns_proxy.dir/src/logging.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[5/8] Building C object CMakeFiles/https_dns_proxy.dir/src/main.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[6/8] Building C object CMakeFiles/https_dns_proxy.dir/src/options.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[7/8] Building C object CMakeFiles/https_dns_proxy.dir/src/stat.c.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
[8/8] Linking C executable https_dns_proxy
FAILED: https_dns_proxy 
[... cut ...]
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: CMakeFiles/https_dns_proxy.dir/src/options.c.o(.text+0x120): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `optarg'
/mox/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

stangri, 2021-08-18:
I've just grabbed the 21.02.0-rc4 SDK for x86_64 and I can't reproduce:

```sh

$ rm -f .config && make defconfig
$ ./scripts/feeds update -f
$ ./scripts/feeds install libcares libcurl libev libgsasl ca-bundle luci-base golang https-dns-proxy
$ umask 022
$ make -j$(($(nproc)+1)) 
Collecting package info: done
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'eip197-mini-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'r8169-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'e100-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'bnx2-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'bnx2x-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'ar3k-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'mwifiex-sdio-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb2', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'edgeport-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-phy-bcm-ns-usb3', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'amdgpu-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'radeon-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'prism54-firmware', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'rtl8192su-firmware', which does not exist
tmp/.config-package.in:7:warning: ignoring type redefinition of 'PACKAGE_ca-bundle' from 'bool' to 'tristate'
tmp/.config-package.in:94:warning: ignoring type redefinition of 'PACKAGE_libc' from 'bool' to 'tristate'
tmp/.config-package.in:122:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'bool' to 'tristate'
tmp/.config-package.in:243:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'bool' to 'tristate'
tmp/.config-package.in:271:warning: ignoring type redefinition of 'PACKAGE_librt' from 'bool' to 'tristate'
tmp/.config-package.in:427:warning: ignoring type redefinition of 'PACKAGE_rpcd' from 'bool' to 'tristate'
tmp/.config-package.in:444:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-file' from 'bool' to 'tristate'
tmp/.config-package.in:458:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-iwinfo' from 'bool' to 'tristate'
tmp/.config-package.in:487:warning: ignoring type redefinition of 'PACKAGE_ubus' from 'bool' to 'tristate'
tmp/.config-package.in:501:warning: ignoring type redefinition of 'PACKAGE_ubusd' from 'bool' to 'tristate'
tmp/.config-package.in:514:warning: ignoring type redefinition of 'PACKAGE_uci' from 'bool' to 'tristate'
tmp/.config-package.in:622:warning: ignoring type redefinition of 'PACKAGE_libiwinfo-data' from 'bool' to 'tristate'
tmp/.config-package.in:783:warning: ignoring type redefinition of 'PACKAGE_libiwinfo-lua' from 'bool' to 'tristate'
tmp/.config-package.in:797:warning: ignoring type redefinition of 'PACKAGE_lua' from 'bool' to 'tristate'
tmp/.config-package.in:2180:warning: ignoring type redefinition of 'PACKAGE_libwolfssl' from 'bool' to 'tristate'
tmp/.config-package.in:2275:warning: ignoring type redefinition of 'PACKAGE_libblkid' from 'bool' to 'tristate'
tmp/.config-package.in:2291:warning: ignoring type redefinition of 'PACKAGE_libblobmsg-json' from 'bool' to 'tristate'
tmp/.config-package.in:2704:warning: ignoring type redefinition of 'PACKAGE_libiwinfo' from 'bool' to 'tristate'
tmp/.config-package.in:2720:warning: ignoring type redefinition of 'PACKAGE_libjson-c' from 'bool' to 'tristate'
tmp/.config-package.in:2744:warning: ignoring type redefinition of 'PACKAGE_liblua' from 'bool' to 'tristate'
tmp/.config-package.in:2759:warning: ignoring type redefinition of 'PACKAGE_liblucihttp' from 'bool' to 'tristate'
tmp/.config-package.in:2770:warning: ignoring type redefinition of 'PACKAGE_liblucihttp-lua' from 'bool' to 'tristate'
tmp/.config-package.in:2869:warning: ignoring type redefinition of 'PACKAGE_libnl-tiny' from 'bool' to 'tristate'
tmp/.config-package.in:2926:warning: ignoring type redefinition of 'PACKAGE_libsmartcols' from 'bool' to 'tristate'
tmp/.config-package.in:3008:warning: ignoring type redefinition of 'PACKAGE_libubox' from 'bool' to 'tristate'
tmp/.config-package.in:3032:warning: ignoring type redefinition of 'PACKAGE_libubus' from 'bool' to 'tristate'
tmp/.config-package.in:3044:warning: ignoring type redefinition of 'PACKAGE_libubus-lua' from 'bool' to 'tristate'
tmp/.config-package.in:3057:warning: ignoring type redefinition of 'PACKAGE_libuci' from 'bool' to 'tristate'
tmp/.config-package.in:3094:warning: ignoring type redefinition of 'PACKAGE_libuuid' from 'bool' to 'tristate'
tmp/.config-package.in:3185:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-luci' from 'bool' to 'tristate'
tmp/.config-package.in:3245:warning: ignoring type redefinition of 'PACKAGE_luci-base' from 'bool' to 'tristate'
tmp/.config-package.in:3283:warning: ignoring type redefinition of 'LUCI_LANG_ar' from 'bool' to 'tristate'
[... cut ...]
 make[3] -C feeds/packages/libs/gnutls clean-build
 make[3] -C feeds/packages/libs/gnutls compile
 make[3] -C feeds/packages/lang/python/python3 clean-build
 make[3] -C feeds/packages/lang/python/python3 compile
 make[3] -C feeds/packages/libs/openldap clean-build
 make[3] -C feeds/packages/libs/openldap compile
 make[3] -C feeds/packages/devel/meson clean-build
 make[3] -C feeds/packages/devel/meson compile
 make[2] package/index

 ```

The `https-dns-proxy` builds fine here. Are the buildbots failing on it?

BKPepe, 2021-08-19:
Dear @stangri,

Everything that you need is included in OP. If you check it carefully, you will see that I was using [hardening.mk](https://github.com/openwrt/openwrt/blob/c5616a8ae1649ef55d362462a39aa133b65dc1e3/include/hardening.mk), thus the kernel options listed there and that I could reproduce it on ARM64. On the other hand, you could take a look if someone does not report it for other packages.

If the issue does not exist on buildbots, you don't care about it since this is your mess in the stable branch, which was not there? With `https-dns-proxy` from **03/06/2021**, I was able to compile it.

Let me show you the steps on how you can reproduce it on your end.

**We can not use SDK**
See https://github.com/openwrt/openwrt/blob/c5616a8ae1649ef55d362462a39aa133b65dc1e3/config/Config-build.in#L247

1. Clone OpenWrt repository and checkout the branch which you want (I used 21.02 branch)
2. Choose the platform to compile it with (in my case `Marvell EBU Armada`; ``3700LP``; any target profile, I choose: ``3720``)
3. **In Global build settings set**:
a) `all` - **User space ASLR PIE**
b) `strong` - **User space Stack-Smashing protection** (for both of them)
c) `aggresive` - **Enable buffer-overflows detection**
d) `full` - **enable RELRO protection**
4. Compile tools, toolchain
5. Update feeds and install them
6. Compile `https-dns-proxy`

Then I'm pretty sure that you will encounter my reported issue.

stangri, 2021-08-19:
You obviously know more about it than I do, so if you believe that [this](https://github.com/stangri/source.openwrt.melmac.net/commit/63c916bcf6c66fa24717b3ab0a5ec635125cad3e) will solve the issue, let me know so I can send a PR. I have no other ideas on what to do after checking some other packages besides it.

stangri, 2021-08-20:
Dear @BKPepe 

I have no intention of looking into this beyond what I've already done (and will probably revert the commit linked above as it seems unneeded). The build passed CI and builds fine for vanilla OpenWrt with default settings for all branch repos.

If you're interested in having this package in the repository for what is essentially a paid OpenWrt derivative, you're welcome to contribute a fix (which you seem to be very well capable of, without creating needless drama), do fixes in your tree or downgrade the package in your tree.

This is a WON'T FIX from me.

BKPepe, 2021-08-20:
:clap: :clap: 

You broke the only stable branch we have (I mean OpenWrt 19.07), and now, you don't want to fix. 

There are two solutions:
1. revert your commit, which broke it
2. fix your mess, which you created

I provided you necessary steps how you can reproduce it and you refused it to fix it.

This is broken in all branches due to your recent changes and should it be fixed downstream? Wow. Who creates a drama now? This is what are you doing with your packages, etc., right? I know you don't like me because you don't follow [CONTRIBUTING.md](https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md) at all as I pointed it in almost every pull request which you created.

It's interesting that hardening is available in OpenWrt and based on your comment, it should not be used as you do not support it and this is the only package that has been broken off otherwise, all of them were fixed or reported.

If you are not willing to fix this, I will revert your commit.

champtar, 2021-08-20:
@BKPepe @stangri no need for drama

@stangri you can call Turris OS a 'paid OpenWrt derivative' but they are contributing back to OpenWrt a lot. Also they are just trying to enable OpenWrt hardening option so we need to fix this build failure at some point.

@BKPepe buildbot does build the package just fine (https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/https-dns-proxy_2021-07-29-1_aarch64_cortex-a53.ipk) and https-dns-proxy 2021-07-29 does bring what seems important fixes (#16252) so instead of threatening to revert an improvement for vanilla OpenWrt user, let's take a bit of time to just fix the issue for people compiling with non default options, and then work to have that added to the CI

BKPepe, 2021-08-20:
Well, it is a question, if those important fixes were needed in the stable branch and the only one stable release, which is there. Thus making the package unavailable or even won't fixing this stuff when the initial idea was introduced and then throwing it away. I am not sure if that is the correct behavior.

So, yeah, reverting it is and will be the last choice when it is marked as no-fix from the maintainer side.

champtar, 2021-08-20:
Again, default OpenWrt is not broken. Please give me a bit of time to look at it.

PR #16413 (2021-08-21): https-dns-proxy: patch CMakeList.txt to use OpenWrt CFLAGS
Maintainer: @stangri 
Compile tested: CONFIG_TARGET_mvebu_cortexa9_DEVICE_cznic_turris-omnia / master
Run tested: no

Description:
This fixes compilation issues with ASLR PIE enabled (will fix #16393 once merged in 19.07)

We were compiling with '-g -DDEBUG'

https-dns-proxy_2021-07-29-*_arm_cortex-a9_vfpv3-d16.ipk
shrink from 19514 to 19095
closed 2021-08-21 by stangri