openwrt/packages #16402: luci-app-bcp38: fails CAIDA bcp38 tests

tiagofreire-pt, 2021-08-19:
Maintainer: @tohojo 
Environment: `ARMv7`, `Linksys EA8300`, `OpenWrt 19.07.8 r11364-ef56c85848 / LuCI openwrt-19.07 branch git-21.189.23240-7b931da`

### Description:

I did some tests using CAIDA software at different moments of `luci-app-bcp38`:

Before installing it: https://spoofer.caida.org/report.php?sessionid=1198440
After: https://spoofer.caida.org/report.php?sessionid=1198445

### Configuration

[image]

I can't see any difference between both tests.

My ISP uses a public addressable IP address on the DHCP provided gateway:

```
# ip route
default via 84.90.192.1 dev eth1  src 84.90.195.xxx
84.90.192.0/22 dev eth1 scope link  src 84.90.195.xxx
```

[image]

[image]

Disabling `Auto-detect upstream IP`: https://spoofer.caida.org/report.php?sessionid=1198480

No changes either.

For what is worth, another test with another portuguese ISP using the same "lan-style-first-hop" RFC1918:

```
Path 4 (to: 130.206.158.142)
192.168.10.1	0
10.240.192.1	0
* * *	0
195.8.30.241	8657
195.8.0.198	8657
195.8.0.198	8657
195.66.226.161	0
* * *	0
* * *	0
* * *	0
62.40.98.72	20965
83.97.88.130	21320
130.206.245.93	766
130.206.195.2	766
130.206.158.2	766
130.206.158.50	766
130.206.158.142	766 
```

Using `10.240.192.1` as first-hop (**but not as gateway!**), passed 100% at these tests. The router is not running Openwrt, instead Opnsense. So, It seems the issue is with Openwrt+luci-app-bcp38 and not with this type of ISP first hops.

In: https://spoofer.caida.org/report.php?sessionid=1198489

tohojo, 2021-08-19:
The bcp38 package drops packets with *destination* IP in the local (rfc1918) subnets. Filtering on source-address doesn't really make sense in a default openwrt setup, because there's a MASQUERADE NAT rule on outbound traffic that will rewrite the source address anyway, so filtering them on outbound would have no effect...

tiagofreire-pt, 2021-08-19:
> The bcp38 package drops packets with _destination_ IP in the local (rfc1918) subnets. Filtering on source-address doesn't really make sense in a default openwrt setup, because there's a MASQUERADE [...]

So, do you suggest filtering it not at the "wan" interface but rather at the "bridge-lan" one? (presuming a default OpenWRT configuration)

tohojo, 2021-08-19:
tiagofreire-pt ***@***.***> writes:

> So, do you suggest filtering it not at the "wan" interface but rather
> at the "bridge-lan" one? (presuming a default OpenWRT configuration)

Yeah, if you want to filter on source address, one way to go about it
would be to drop everything with a source address other than the
internal LAN subnet when coming from the LAN interface...