openwrt/packages #16431: auc

aparcar, 2021-08-22:
Maintainer: @dangowrt 
Environment: SNAPSHOTS

Description:
If there are issues post image build, e.g. a missing opkg key or any other message from `sysupgrade`, that message should be shown rather than the `stdout` and `stderr` from the server, which is currently the case. If the server responds a build error `stdout` and `stderr` are helpful.

dangowrt, 2021-09-10:
Both are shown right now, local and remote output. I thought that even in case of local errors it may be useful to see the remote `stdout` and `stderr` as that may contain a hint towards why the image failed locally.

aparcar, 2021-09-11:
I wonder if we should show the local error after the remote build log since it tends to be very long.