openwrt/packages #16485: rsyncd: failing when called from client with --perms and group permissions

ulboet, 2021-08-28:
Maintainer: Maxim Storchak <m.storchak@gmail.com>
Environment: kirkwood, zyxel nsa310s, 21.02.0-rc4

Description:
when trying to rsync from a remote computer (raspian, ubuntu, synology) when the remote rsync is called with --perms and the to be transferred file has group or all permissions set (the user permissions alone work in any combination) throws an error of 

* rsync: failed to set permissions on ".test.iFkGGp" (in Root): No such file or directory (2), if the file does not exist in the openwrt-target called "Root", but transfers the file (time stamp is correctly transferred by --times)  and
* 
* rsync: failed to set permissions on "test" (in Root): No such file or directory (2), if the file exists in the target.

openwrt was compiled options ACL and xattrs active, tried with rsync 3.2.3, 3.2.2 and 3.1.3:
Capabilities:
    64-bit files, 64-bit inums, 32-bit timestamps, 64-bit long ints,
    no socketpairs, hardlinks, symlinks, IPv6, batchfiles, inplace,
    append, ACLs, xattrs, no iconv, symtimes, prealloc

mstorchak, 2021-08-29:
Could you please describe the steps for reproducing the issue?
Are you using rsyncd? Does it run as root or a (dedicated?) unprivileged user?

ulboet, 2021-08-29:
I'm running syncd as service, but same result from command-line as rsync --daemon --no-detach by ssh-login as root. chown and chmod are working in the target directory. The files are arriving, they get the permission 600 root:root, executable permission / directories are transferred as well.

That's the used rsyncd.conf

pid file = /var/run/rsyncd.pid
log file = /var/log/rsyncd.log

reverse lookup = false
max verbosity = 4

[Root]
  path = /mnt/sda1/./test
  read only = no
  write only = no
  uid = root
  gid = root
  list = yes
  use chroot = yes

mstorchak, 2021-08-31:
Got it, thanks for the details, I'll have a closer look.

mstorchak, 2021-09-02:
@ulboet, I apologize for the delay.
Thank you for the detailed explanation, I was able to reproduce the issue with rsync --daemon.

This happens because you enable chroot. For chmod to work you need to mount /proc at /mnt/sda1/proc:
```
trace:12163 fstatat64(AT_FDCWD, "/proc/self/fd/5", 0x7fdc8568, 0) = -1 ENOENT (No such file or directory)

trace_proc:12208 fstatat64(AT_FDCWD, "/proc/self/fd/5", {st_mode=S_IFREG|0700, st_size=1858, ...}, 0) = 0
trace_proc:12208 fchmodat(AT_FDCWD, "/proc/self/fd/5", 0755) = 0
```

mstorchak, 2021-09-02:
For the sake of completeness, here's my setup:
```

root@vortex-box:/mnt/dump/rsync_root# ls -la
drwxr-xr-x    4 root     root          4096 Sep  2 11:17 .
drwxr-xr-x   16 root     root          4096 Sep  2 11:00 ..
drwxr-xr-x    2 root     root          4096 Sep  2 11:14 Root
dr-xr-xr-x   83 root     root             0 Jan  1  1970 proc
-rw-r--r--    1 root     root           252 Sep  2 11:01 rsyncd.conf
-rw-r--r--    1 root     root          2858 Sep  2 11:17 rsyncd.log
-rw-r--r--    1 root     root         35750 Sep  2 11:07 trace
-rw-r--r--    1 root     root         38162 Sep  2 11:17 trace_proc

root@vortex-box:/mnt/dump/rsync_root# cat rsyncd.conf
pid file = /mnt/dump/rsync_root/rsyncd.pid
log file = /mnt/dump/rsync_root/rsyncd.log

reverse lookup = false
max verbosity = 4

[Root]
path = /mnt/dump/rsync_root/./Root
read only = no
write only = no
uid = root
gid = root
list = yes
use chroot = yes

root@vortex-box:/mnt/dump/rsync_root# mount -t proc
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
proc on /mnt/dump/rsync_root/proc type proc (rw,nosuid,nodev,noexec,noatime)
```
running rsync as
```
strace  -f -o trace_proc rsync --daemon --config=/mnt/dump/rsync_root/rsyncd.conf --no-detach -v
```

The results:
```
root@vortex-box:/mnt/dump/rsync_root# ls -la Root/pre-commit*
-rwx------    1 210390   21176         1858 Aug  6 17:47 Root/pre-commit
-rwxr-xr-x    1 210390   21176         1858 Aug  6 17:47 Root/pre-commit_proc
```