openwrt/packages #16486: https-dns-proxy: freezes over less than 24h

idarek, 2021-08-29:
Environment: LuCI openwrt-19.07 branch (git-21.238.28320-53f59d3) / OpenWrt 19.07.8 r11364-ef56c85848

Description: Package freezes over time, typically in less than 24h in the default configuration. Need to stop or restart to resume DNS to work in the network. When freshly installed and started returning following error (but working) in `logread`

```
Sun Aug 29 08:12:51 2021 daemon.info https-dns-proxy[497]: [E] 1630221171.753180 https_client.c:229 E6D7: CURLOPT_HTTP_VERSION error 1: Error
Sun Aug 29 08:12:51 2021 daemon.info https-dns-proxy[497]: [E] 1630221171.753256 https_client.c:231 Try to run application with -x argument! Forcing HTTP/1.1 version.
Sun Aug 29 08:12:51 2021 daemon.info https-dns-proxy[496]: [E] 1630221171.754282 https_client.c:229 E6D7: CURLOPT_HTTP_VERSION error 1: Error
Sun Aug 29 08:12:51 2021 daemon.info https-dns-proxy[496]: [E] 1630221171.754346 https_client.c:231 Try to run application with -x argument! Forcing HTTP/1.1 version.
```

As suggested [here](https://forum.openwrt.org/t/https-dns-proxy-vs-stubby/52446/10) adding `option use_http1 '1'` to `/etc/config/https-dns-proxy` remove error above but still need to investigate if not freezing later on (will update on that).

stangri, 2021-08-29:
I suspect it happens when the WAN interface is updated but `https-dns-proxy` does not restart. Some time in the near future, I will be pushing an update which restarts https-dns-proxy in such an event instead of reloading it (which doesn't seem to solve the problem).

idarek, 2021-08-29:
Are there been any update in recent days that start causing this issue? Just updated to 19.07.8 and got this issue since last 3 days, where on .7 been working for 90 days constantly without issue…

stangri, 2021-08-29:
There was an update to 19.07 branch 9 days ago.

You can try editing the `/etc/hotplug.d/iface/90-https-dns-proxy` file on the router and using `restart` instead of `reload` on [line 5](https://github.com/openwrt/packages/blob/openwrt-19.07/net/https-dns-proxy/files/https-dns-proxy.hotplug.iface#L5).

idarek, 2021-08-29:
Thanks, will try that, but don’t think it’s relate to my issue directly where there is no issue (down) on my WAN interface, it simply stop working overnight and during a day notice very slow dns responses when it’s on comparing to when it’s off. Haven’t experienced this before hence start looking in logs. 

Will let you know if that change anything.

idarek, 2021-08-29:
Yep, just to confirm.
In the current version, the package is hardly useable. When turned on, the DNS resolve in the local network is very slow making it impossible to use.

main error before users in local network start reporting issues is:
```
Sun Aug 29 22:05:04 2021 daemon.info https-dns-proxy[26650]: [E] 1630271104.111734 https_client.c:229 FBA4: CURLOPT_HTTP_VERSION error 1: Error
```

stangri, 2021-08-30:
> Other references [#111](https://github.com/aarond10/https_dns_proxy/issues/111)

If you want to use HTTP/2-only resolvers you will need to update to 21.02 or snapshot where libcurl is now built with HTTP/2 support.

Without additional details there's no actionable information in this.

idarek, 2021-08-30:
There is still plenty packages missing on 21 version and snapshot is not suitable for production.

is there any way to enable detailed login to measure what happening with DNS queries?

idarek, 2021-08-30:
Yes, it looks like this issue relates to some (not all devices) and on 21.02 is working well.

closed 2021-08-30 by idarek

stangri, 2021-08-30:
> Update: Nothing found in logs, think its something to do with overall performance. Will try to update to 21.x and have a try again.

Could also be picky resolvers, some resolvers only allow HTTP/2 nowdays, so resolution fails on non-HTTP/2 compliant systems. I know I've added about half a dozen resolvers to the luci app once the HTTP/2 support was introduced to curl in 21.02.

idarek, 2021-08-30:
Thanks for your hard work.