openwrt/packages #16487: libstdcpp6 missing

dirk99, 2021-08-29:
Hi folks,

i wan't to install php7-cli, but libstdcpp6 is missing. 

Thanks

feckert, 2021-08-31:
There is only the package libstdcpp.
I think it's the ABI version. https://github.com/openwrt/openwrt/blob/ec6293febc244d187e71a6e54f44920be679cde4/package/libs/toolchain/Makefile#L90
What exactly does your error message look like?

vlna, 2021-09-19:
Similar here, OpenWrt 21.02.0, r16279-5cc0535800 and I want to install _zerotier_ package

[image]

vlna, 2021-09-19:
Solution/workaroud:
```
cd /tmp
wget https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/libstdcpp6_8.4.0-3_mips_24kc.ipk
opkg install libstdcpp6_8.4.0-3_mips_24kc.ipk
```
After this I was able to install _zerotier_ package

Rygle, 2021-09-29:
I did a similar workaround to @vlna, but I believe there are different versions of the package, so be careful!
Another version is here - https://downloads.openwrt.org/releases/21.02.0/targets/ipq806x/generic/packages/libstdcpp6_8.4.0-3_arm_cortex-a15_neon-vfpv4.ipk

hnyman, 2023-01-14:
Related to old version.
The package seems to be in the 21.02.5 (and 21.02.0) download directory.

closed 2023-01-14 by hnyman