openwrt/packages #16495: MWAN3 .user trouble

reinerotto, 2021-08-30:
mwan3.user not working as expected. For details, pls check
https://forum.openwrt.org/t/mwan3-user-trouble/104547

I suspect, that there might be an issue regarding udhpc acquiring an IP, to trigger second "connected".
And then a deadlock between inetd and mwan3 trying to lock/run the mwan3.user script

Willing to help testing; actually I use polling of tmp/run/mwan3/iface_state/*

feckert, 2021-09-02:
Sorry but I cannot reproduce your problem.

My mwan3.user script:
```
if [ "${ACTION}" == "connected" ]; then
        if [ "${INTERFACE}" == "wan" ] || [ "${INTERFACE}" == "wwan" ] || [ "${INTERFACE}" == "wifiwan" ]; then
                logger "msgq ${INTERFACE} connected"
                logger "msgq trigger post_old_msg_q"
        fi
fi
```

Offline detection:
```
Thu Sep  2 15:42:22 2021 user.info mwan3track[18619]: Check (ping) failed for target "8.8.8.8" on interface wan (eth0). Current score: 6
Thu Sep  2 15:42:24 2021 user.info mwan3track[18619]: Check (ping) failed for target "1.1.1.1" on interface wan (eth0). Current score: 6
Thu Sep  2 15:42:26 2021 user.info mwan3track[18619]: Check (ping) failed for target "9.9.9.9" on interface wan (eth0). Current score: 6
Thu Sep  2 15:42:31 2021 user.info mwan3track[18619]: Check (ping) failed for target "update.test.de" on interface wan (eth0). Current score: 6
Thu Sep  2 15:42:31 2021 user.notice mwan3track[18619]: Interface wan (eth0) is disconnecting
Thu Sep  2 15:42:31 2021 user.notice mwan3test/iface[19958]: ====== INTERFACE=wan ACTION=disconnecting ======
Thu Sep  2 15:42:31 2021 user.notice mwan3test/iface[19958]: ====== USER=root ACTION=disconnecting SHLVL=1 HOTPLUG_TYPE=iface LOGNAME=root DEVICENAME= PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan PWD=/ DEVICE=eth0 ======
Thu Sep  2 15:42:35 2021 user.info mwan3track[18619]: Check (ping) failed for target "8.8.8.8" on interface wan (eth0). Current score: 5
Thu Sep  2 15:42:37 2021 user.info mwan3track[18619]: Check (ping) failed for target "1.1.1.1" on interface wan (eth0). Current score: 5
Thu Sep  2 15:42:39 2021 user.info mwan3track[18619]: Check (ping) failed for target "9.9.9.9" on interface wan (eth0). Current score: 5
Thu Sep  2 15:42:44 2021 user.info mwan3track[18619]: Check (ping) failed for target "update.test.de" on interface wan (eth0). Current score: 5
Thu Sep  2 15:42:47 2021 user.info mwan3track[18619]: Check (ping) failed for target "8.8.8.8" on interface wan (eth0). Current score: 4
Thu Sep  2 15:42:49 2021 user.info mwan3track[18619]: Check (ping) failed for target "1.1.1.1" on interface wan (eth0). Current score: 4
Thu Sep  2 15:42:51 2021 user.info mwan3track[18619]: Check (ping) failed for target "9.9.9.9" on interface wan (eth0). Current score: 4
Thu Sep  2 15:42:53 2021 local0.err EffiLinkControl.sh: Status: UNKNOWN Net: 192.168.0.0/24 NatNetwork: NONE
Thu Sep  2 15:42:53 2021 local0.warn EffiLinkControl.sh: Couldn't get a status information from the EffiLink server
Thu Sep  2 15:42:56 2021 user.info mwan3track[18619]: Check (ping) failed for target "update.tdt.de" on interface wan (eth0). Current score: 4
Thu Sep  2 15:42:56 2021 user.notice mwan3track[18619]: Interface wan (eth0) is offline
Thu Sep  2 15:42:56 2021 user.notice mwan3test/iface[20285]: ====== INTERFACE=wan ACTION=disconnected ======
Thu Sep  2 15:42:56 2021 user.notice mwan3test/iface[20285]: ====== USER=root ACTION=disconnected SHLVL=1 HOTPLUG_TYPE=iface LOGNAME=root DEVICENAME= PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan PWD=/ DEVICE=eth0 ======
Thu Sep  2 15:42:56 2021 user.notice mwan3-hotplug[20285]: Execute disconnected event on interface wan (eth0)
```

Online Detection:
```
Thu Sep  2 15:43:14 2021 user.info mwan3track[18619]: Check (ping) success for target "9.9.9.9" on interface wan (eth0). Current score: 0
Thu Sep  2 15:43:14 2021 user.notice mwan3track[18619]: Interface wan (eth0) is connecting
Thu Sep  2 15:43:14 2021 user.notice mwan3test/iface[20621]: ====== INTERFACE=wan ACTION=connecting ======
Thu Sep  2 15:43:14 2021 user.notice mwan3test/iface[20621]: ====== USER=root ACTION=connecting SHLVL=1 HOTPLUG_TYPE=iface LOGNAME=root DEVICENAME= PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan PWD=/ DEVICE=eth0 ======
Thu Sep  2 15:43:14 2021 user.info mwan3track[18619]: Lost 18 ping(s) on interface wan (eth0). Current score: 0
Thu Sep  2 15:43:19 2021 user.info mwan3track[18619]: Check (ping) success for target "8.8.8.8" on interface wan (eth0). Current score: 1
Thu Sep  2 15:43:24 2021 user.info mwan3track[18619]: Check (ping) success for target "8.8.8.8" on interface wan (eth0). Current score: 2
Thu Sep  2 15:43:24 2021 user.notice mwan3track[18619]: Interface wan (eth0) is online
Thu Sep  2 15:43:24 2021 user.notice mwan3test/iface[20888]: ====== INTERFACE=wan ACTION=connected ======
Thu Sep  2 15:43:24 2021 user.notice mwan3test/iface[20888]: ====== USER=root ACTION=connected SHLVL=1 FIRSTCONNECT= HOTPLUG_TYPE=iface LOGNAME=root DEVICENAME= PATH=/usr/sbin:/usr/bin:/sbin:/bin INTERFACE=wan PWD=/ DEVICE=eth0 ======
Thu Sep  2 15:43:24 2021 user.notice mwan3-hotplug[20888]: Execute connected event on interface wan (eth0)
Thu Sep  2 15:43:25 2021 user.notice root: msgq wan connected
Thu Sep  2 15:43:25 2021 user.notice root: msgq trigger post_old_msg_q
Thu Sep  2 15:43:30 2021 user.info mwan3track[18619]: Check (ping) success for target "8.8.8.8" on interface wan (eth0). Current score: 3
```

reinerotto, 2021-09-02:
One more try on my system, with some comments:
Thu Sep  2 18:44:37 2021 daemon.notice netifd: Interface 'wan' is enabled
Thu Sep  2 18:44:40 2021 daemon.notice netifd: Network device 'eth1' link is up
Thu Sep  2 18:44:40 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Thu Sep  2 18:44:40 2021 daemon.notice netifd: Interface 'wan' is setting up now
Thu Sep  2 18:44:40 2021 kern.info kernel: [14364.026354] eth1: link up (100Mbps/Full duplex)
Thu Sep  2 18:44:40 2021 daemon.notice netifd: wan (15161): udhcpc: started, v1.33.1
Thu Sep  2 18:44:40 2021 daemon.notice netifd: wan (15161): udhcpc: sending discover
Thu Sep  2 18:44:43 2021 daemon.notice netifd: wan (15161): udhcpc: sending discover
Thu Sep  2 18:44:44 2021 daemon.notice netifd: wan (15161): udhcpc: sending select for 192.168.20.158
Thu Sep  2 18:44:44 2021 daemon.notice netifd: wan (15161): udhcpc: lease of 192.168.20.158 obtained, lease time 43200
Thu Sep  2 18:44:44 2021 daemon.notice netifd: Interface 'wan' is now up
Thu Sep  2 18:44:46 2021 user.notice mwan3-hotplug[15401]: Execute ifup event on interface wan (eth1)
#May be, next is because I do NOT use the openwrt-firewall, but plain and simple iptables instead ? 
Thu Sep  2 18:44:46 2021 user.warn mwan3-hotplug[15401]: failed to add 192.168.20.0/24 dev eth1 proto static scope link metric 10 to table 1
Thu Sep  2 18:44:46 2021 user.warn mwan3-hotplug[15401]: failed to add default via 192.168.20.1 dev eth1 proto static src 192.168.20.158 metric 10 to table 1
Thu Sep  2 18:44:47 2021 user.info mwan3track[1591]: Detect ifup event on interface wan (eth1)
Thu Sep  2 18:44:47 2021 user.info mwan3track[1591]: Check (ping) success for target "1.1.1.1" on interface wan (eth1). Current score: 0
Thu Sep  2 18:44:47 2021 user.notice mwan3track[1591]: Interface wan (eth1) is connecting
Thu Sep  2 18:44:48 2021 user.notice mwan3-hotplug: ACTION=ifup SHLVL=1 INTERFACE=wan PWD=/ DEVICE=eth1
Thu Sep  2 18:44:49 2021 user.notice mwan3-hotplug: ACTION=connecting SHLVL=1 INTERFACE=wan PWD=/ DEVICE=eth1
Thu Sep  2 18:44:49 2021 user.notice mwan3track[1591]: Interface wan (eth1) is online
#Next is a msg, sent via mqtt (mosquitto), after detection (via polling) of an IP on eth1
Thu Sep  2 18:44:49 2021 user.notice TX: {   "v": 3,   "ts": 1630608287,   "d": {      "d": "wired",      "os": {        "ip": null      },      "ns": {        "ip": "192.168.20.158"      }   } }
Thu Sep  2 18:44:50 2021 user.notice mwan3-hotplug: ACTION=connected SHLVL=1 INTERFACE=wan PWD=/ DEVICE=eth1
Thu Sep  2 18:44:50 2021 user.notice root: msgq wan connected #From my mwan3.user
Thu Sep  2 18:44:50 2021 user.notice root: msgq trigger post_old_msg_q #From my mwan3.user
Thu Sep  2 18:44:50 2021 user.info mwan3track[1591]: Check (ping) success for target "1.1.1.1" on interface wan (eth1). Current score: 1
Thu Sep  2 18:44:50 2021 user.notice mwan3track[1591]: Interface wan (eth1) is connecting
Thu Sep  2 18:44:51 2021 user.notice mwan3-hotplug: ACTION=connecting SHLVL=1 INTERFACE=wan PWD=/ DEVICE=eth1
Thu Sep  2 18:44:57 2021 user.info mwan3track[1591]: Check (ping) success for target "1.1.1.1" on interface wan (eth1). Current score: 2
Thu Sep  2 18:44:57 2021 user.notice mwan3track[1591]: Interface wan (eth1) is online
Thu Sep  2 18:44:57 2021 user.notice mwan3-hotplug[16348]: Execute connected event on interface wan (eth1)
Thu Sep  2 18:44:58 2021 user.notice mwan3-hotplug: ACTION=connected SHLVL=1 INTERFACE=wan PWD=/ DEVICE=eth1
Thu Sep  2 18:44:58 2021 user.notice root: msgq wan connected

Thus, there are 2 mayor differences to your scenario:
1) No usage of openwrt-firewall on my system, but plain iptables instead.
(No other issues noticed, besides the one we are discussing here.)
2) There is a tcp-connection initiated, before first "connected".

reinerotto, 2021-09-03:
cat mwan3

config globals 'globals'
        option mmx_mask '0x3F00'

config interface 'wan'
        option enabled '1'
        list track_ip '1.1.1.1'
        list track_ip '8.8.8.8'
        option family 'ipv4'
        option reliability '1'
        option count '1'
        option timeout '2'
        option failure_latency '1000'
        option recovery_latency '500'
        option failure_loss '20'
        option recovery_loss '5'
        option interval '5'
        option down '3'
        option up '3'

config interface 'wwan'
        option enabled '1'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'
        option family 'ipv4'
        option reliability '1'
        option count '1'
        option timeout '2'
        option failure_latency '1000'
        option recovery_latency '500'
        option failure_loss '20'
        option recovery_loss '5'
        option interval '5'
        option down '3'
        option up '3'

config interface 'wifiwan'
        option enabled '1'
        list track_ip '8.8.4.4'
        list track_ip '1.1.1.1'
        option family 'ipv4'
        option reliability '1'
        option count '1'
        option timeout '2'
        option failure_latency '1000'
        option recovery_latency '500'
        option failure_loss '20'
        option recovery_loss '5'
        option interval '5'
        option down '3'
        option up '3'

config member 'wifiwan_m1'
        option interface 'wifiwan'
        option metric '2'
        option weight '2'

config member 'wwan_m2'
        option interface 'wwan'
        option metric '3'
        option weight '3'

config member 'wan_m3'
        option interface 'wan'
        option metric '1'
        option weight '1'

config policy 'standard'
        list use_member 'wifiwan_m1'
        list use_member 'wwan_m2'
        list use_member 'wan_m3'

config rule 'default_rule'
        option dest_ip '0.0.0.0/0'
        option use_policy 'standard'

feckert, 2023-09-20:
Issue is already over two years old and there have been some changes in newer mwan3 versions that may have solved the problem.
https://github.com/openwrt/packages/commit/5f0461be8b149aeb5732481d38c9d69650996f8c
https://github.com/openwrt/packages/commit/997c0842ca8b05de83a1ecd3160ecd18fa3d8aee