openwrt/packages #16511: v2raya: 0.0.0.0 fails to start management interface on 2017

wulfy23, 2021-09-01:
Maintainer: @1715173329
Environment: r1745x

Description:

```
lsof -i -nP | grep v2ray | grep -v 443 | wc -l
0
```

as stated in the issue title... possibly modify initscript to select br-lan ip if 2017 mgmt ip is 0.0.0.0

cubercsl, 2021-09-01:
https://github.com/openwrt/packages/blob/90ec599f9ef7aa011febe5528ba8b0b2016bfe02/net/v2raya/files/v2raya.init#L40

The param_arg `--pluginlistenport`  has been deprecated in 1.5.2. Just commet it out and try again.

1715173329, 2021-09-01:
I apologize for this problem, will fix this in next release.
Please see the reply above for solution.

wulfy23, 2021-09-01:
sweet. thankyou for the work on this package and the fast response.

predators46, 2021-09-06:
Hy 1715173329

does v2raya not have service v2raya restart command i get error

`root@OpenWrt:/tmp# /etc/init.d/v2raya restart
Command failed: Not found`

predators46, 2021-09-07:
v2raya is great but I gave up using it. there is no information to access the web gui v2raya I compared with adguardhome very complete information when it finished installing in the system-log. to access the v2raya web gui I have to enter the v2raya command in the terminal and wait for the information to access v2raya example 192.168.1.10:2017 but if the terminal is closed or the router is rebooted I can no longer access the v2raya web gui and I enter the v2raya command in the terminal and get an error message `2021/09/07 03:40:41 nutsdb.go:23: db.buildIndexes error: invalid argument `

cubercsl, 2021-09-07:
> to access the v2raya web gui I have to enter the v2raya command in the terminal and wait for the information to access v2raya example 192.168.1.10:2017 but if the terminal is closed or the router is [...]

https://github.com/openwrt/packages/issues/16511#issuecomment-909992875

```
uci set v2raya.config.enabled='1'
uci commit v2raya
/etc/init.d/v2raya enable
/etc/init.d/v2raya start
```

predators46, 2021-09-07:
> > to access the v2raya web gui I have to enter the v2raya command in the terminal and wait for the information to access v2raya example 192.168.1.10:2017 but if the terminal is closed or the router is rebooted I can no longer access the v2raya web gui and I enter the v2raya command in the terminal and get an error message `2021/09/07 03:40:41 nutsdb.go:23: db.buildIndexes error: invalid argument`
> 
> @predators46
> You can set autostart v2raya.
> 
> [#16511 (comment)](https://github.com/openwrt/packages/issues/16511#issuecomment-909992875)
> 
> ```
> uci set v2raya.config.enabled='1'
> uci commit v2raya
> /etc/init.d/v2raya enable
> /etc/init.d/v2raya start
> ```

I've followed everything and it doesn't work when I reboot my router the v2raya web gui can't be accessed anymore

1715173329, 2021-09-07:
@predators46 you need modify the init script...
See the second comment.

predators46, 2021-09-07:
> @predators46 you need modify the init script...
> See the second comment.

I've changed the init script when the router reboots I can't access the v2raya web gui anymore

1715173329, 2021-09-07:
> I've changed the init script when the router reboots I can't access the v2raya web gui anymore

Please add the following commands to `start_service()` to see the real errors:
```
procd_set_param stdout 1
procd_set_param stderr 1
```

See https://openwrt.org/docs/guide-developer/procd-init-scripts#defining_service_instances

predators46, 2021-09-07:
Thanks all solved

malikshi, 2021-09-07:
i tried to install via opkg install v2rayA, it's seem not available in 21.02.0

1715173329, 2021-09-15:
Please try with #16634

PR #16634 (2021-09-18): v2raya: Update to 1.5.3
Maintainer: me
Compile tested: ipq807x, rockchip
Run tested: rk3328 nanopi-r2s

Description:
- Added conffiles
- Refreshed init srcipt to adapt the new arguments
- Renamed package name to lowercase (suggestion from upstream)
- Updated dependencies and license

This closes #16511.
closed 2021-09-16 by wulfy23