openwrt/packages #16525: ieee8021xclient: wpad only dependency limitation

wulfy23, 2021-09-03:
Maintainer: @yangfl
Environment: recent master all

Description: ieee8021xclient has a singular dependency on wpad ... this creates a conflict when other fully featured wpad bundles/hostapd are selected which afaik would also support the featureset...

```
define Package/ieee8021xclient
  SECTION:=net
  CATEGORY:=Network
  MAINTAINER:=David Yang <mmyangfl@gmail.com>
  TITLE:=Wired 802.1x client config support
  DEPENDS:=+wpad
  PKGARCH:=all
endef
```

dangowrt, 2021-09-03:
Imho `ieee8021xclient` should depend on `wpa_supplicant` which is provided by all variants of `wpa_supplicant` and `wpad`.

closed 2021-09-05 by wulfy23

ptpt52, 2021-09-06:
@wulfy23 @neheb @yangfl 
```
WARNING: Makefile 'package/feeds/packages/ieee8021xclient/Makefile' has a dependency on 'wpa_supplicant', which does not exist
```

azgh, 2021-09-16:
> @wulfy23 @neheb @yangfl
> 
> ```
> WARNING: Makefile 'package/feeds/packages/ieee8021xclient/Makefile' has a dependency on 'wpa_supplicant', which does not exist
> ```

I had the same problem.