openwrt/packages #16526: adblock: puts youtube into restricted mode

rayrapetyan, 2021-09-03:
Maintainer: dev@brenken.org
Environment: OpenWrt 19.07.8 r11364-ef56c85848 on Linksys WRT1200AC

Description:

Once adblock is enabled with a different blocklist sources, youtube is being put into the "restricted mode" (the switch is ON and you can't turn it OFF) on all devices. 

Tried to whitelist following domains:
```
googleapis.com
youtube.com
doubleclick.net
google.com
```
with no luck.

Observation: tcpdump shows no NXDOMAIN entries when refreshing page with a restricted video on youtube.

Only a full disabling of adblock brings youtube into normal mode.

dibdot, 2021-09-04:
Not reproducible, please provide your config and the output of `/etc/init.d/adblock status`.

acme-im, 2021-09-04:
Sorry for a poor description.

Steps to reproduce:

1. Configure adblock as shown below:
```
config adblock 'global'
	option adb_debug '0'
	option adb_forcedns '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_backup '1'
	option adb_maxqueue '4'
	option adb_dns 'dnsmasq'
	option adb_fetchutil 'uclient-fetch'
	option adb_backupdir '/etc/adblock'
	option adb_safesearch '1'
	option adb_report '1'
	option adb_repiface 'br-lan'
	list adb_sources 'adaway'
	list adb_sources 'adguard'
	list adb_sources 'android_tracking'
	list adb_sources 'andryou'
	list adb_sources 'anti_ad'
	list adb_sources 'anudeep'
	list adb_sources 'bitcoin'
	list adb_sources 'disconnect'
	list adb_sources 'energized_porn'
	list adb_sources 'firetv_tracking'
	list adb_sources 'games_tracking'
	list adb_sources 'openphish'
	list adb_sources 'phishing_army'
	list adb_sources 'reg_ru'
	list adb_sources 'shallalist'
	list adb_sources 'smarttv_tracking'
	list adb_sources 'spam404'
	list adb_sources 'stevenblack'
	list adb_sources 'stopforumspam'
	list adb_sources 'wally3k'
	list adb_sources 'whocares'
	list adb_sources 'winhelp'
	list adb_sources 'winspy'
	list adb_sources 'yoyo'
	option adb_enabled '1'
```
2. Restart adblock
3. Open a new incognito tab in Chrome and try to open: https://www.youtube.com/watch?v=vkAauL2g6nc
4. 
Expected: video should start playing
Actual:
```
Video unavailable
This video is restricted. Please check the Google Workspace administrator and/or the network administrator restrictions.
```
message shown. Restricted mode set to ON in user profile and can't be disabled.

```
/etc/init.d/adblock status
::: adblock runtime information
  + adblock_status  : enabled
  + adblock_version : 4.0.7
  + blocked_domains : 397678
  + active_sources  : adaway adguard android_tracking andryou anti_ad anudeep bitcoin disconnect energized_porn firetv_tracking games_tracking openphish phishing_army reg_ru shallalist smarttv_tracking spam404 stevenblack stopforumspam wally3k whocares winhelp winspy yoyo 
  + dns_backend     : dnsmasq, /tmp/dnsmasq.d
  + run_utils       : /bin/uclient-fetch, /usr/bin/awk
  + run_ifaces      : trigger: -, report: br-lan
  + run_directories : base: /tmp, backup: /etc/adblock, report: /tmp, jail: /tmp
  + run_flags       : backup: 1, reset: 0, flush: 0, force: 0, search: 1, report: 1, mail: 0, jail: 0
  + last_run        : reload, 1m 44s, 511/371/355, 04.09.2021 08:36:50
  + system          : Linksys WRT1200AC, OpenWrt 19.07.8 r11364-ef56c85848
```

rayrapetyan, 2021-09-04:
Sorry for a poor description.

Steps to reproduce:

1. Configure adblock as shown below:
```
config adblock 'global'
	option adb_debug '0'
	option adb_forcedns '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_backup '1'
	option adb_maxqueue '4'
	option adb_dns 'dnsmasq'
	option adb_fetchutil 'uclient-fetch'
	option adb_backupdir '/etc/adblock'
	option adb_safesearch '1'
	option adb_report '1'
	option adb_repiface 'br-lan'
	list adb_sources 'adaway'
	list adb_sources 'adguard'
	list adb_sources 'android_tracking'
	list adb_sources 'andryou'
	list adb_sources 'anti_ad'
	list adb_sources 'anudeep'
	list adb_sources 'bitcoin'
	list adb_sources 'disconnect'
	list adb_sources 'energized_porn'
	list adb_sources 'firetv_tracking'
	list adb_sources 'games_tracking'
	list adb_sources 'openphish'
	list adb_sources 'phishing_army'
	list adb_sources 'reg_ru'
	list adb_sources 'shallalist'
	list adb_sources 'smarttv_tracking'
	list adb_sources 'spam404'
	list adb_sources 'stevenblack'
	list adb_sources 'stopforumspam'
	list adb_sources 'wally3k'
	list adb_sources 'whocares'
	list adb_sources 'winhelp'
	list adb_sources 'winspy'
	list adb_sources 'yoyo'
	option adb_enabled '1'
```
2. Restart adblock
3. Open a new incognito tab in Chrome and try to open: https://www.youtube.com/watch?v=vkAauL2g6nc

Expected: video should start playing
Actual:
```
Video unavailable
This video is restricted. Please check the Google Workspace administrator and/or the network administrator restrictions.
```
message shown. Restricted mode is set to ON in a user profile and can't be disabled.

```
/etc/init.d/adblock status
::: adblock runtime information
  + adblock_status  : enabled
  + adblock_version : 4.0.7
  + blocked_domains : 397678
  + active_sources  : adaway adguard android_tracking andryou anti_ad anudeep bitcoin disconnect energized_porn firetv_tracking games_tracking openphish phishing_army reg_ru shallalist smarttv_tracking spam404 stevenblack stopforumspam wally3k whocares winhelp winspy yoyo 
  + dns_backend     : dnsmasq, /tmp/dnsmasq.d
  + run_utils       : /bin/uclient-fetch, /usr/bin/awk
  + run_ifaces      : trigger: -, report: br-lan
  + run_directories : base: /tmp, backup: /etc/adblock, report: /tmp, jail: /tmp
  + run_flags       : backup: 1, reset: 0, flush: 0, force: 0, search: 1, report: 1, mail: 0, jail: 0
  + last_run        : reload, 1m 44s, 511/371/355, 04.09.2021 08:36:50
  + system          : Linksys WRT1200AC, OpenWrt 19.07.8 r11364-ef56c85848
```

dibdot, 2021-09-04:
```
[...]
option adb_safesearch '1'
[...]
```
Well, works as configured ... :smile: 
You've enabled 'safesearch', e.g.
[image]

Options for you:
* disable safesearch at all
* remove youtube from safesearch
* relax youtube filter (see separate option)

closed 2021-09-04 by dibdot