openwrt/packages #16528: samba4: Files installed unstripped

daniel-santos, 2021-09-04:
Maintainer: @Andy2244 (Andy Walsh <andy.walsh44+github@gmail.com>)
Environment: mipsel_24kc_musl (mt7620a) v21.02

Samba's executables are making it to squashfs unstripped.  I'm running with -g3, so it was more noticeable:

```
debug_size               file
772181      (0x000bc855) build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/pdbedit
868903      (0x000d4227) build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/smbpasswd
994409      (0x000f2c69) build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/smbd
1037067     (0x000fd30b) build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/nmblookup
1366895     (0x0014db6f) build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/nmbd
```

daniel-santos, 2021-09-04:
Hrm, was using binutils 2.35.1 with `strip --strip-unneeded --remove-section=.comment --remove-section=.note`.  When I run this manually, I get no change, so this is likely a binutils bug.

daniel-santos, 2021-09-04:
OK, so the problem still exists with binutils 2.32, so re-opening.  Everything in my build_dir/target*/root-ramips directory is stripped except for samba4 and libperl.so.

daniel-santos, 2021-09-04:
Here's my ugly fix for you:
```patch
diff --git a/net/samba4/Makefile b/net/samba4/Makefile
index f3108fc58..af2c9296c 100644
--- a/net/samba4/Makefile
+++ b/net/samba4/Makefile
@@ -130,7 +130,7 @@ endef
 TARGET_CFLAGS += $(FPIC) -ffunction-sections -fdata-sections
 TARGET_LDFLAGS += -Wl,--gc-sections,--as-needed
 # dont mess with sambas private rpath!
-RSTRIP:=:
+RSTRIP:=do_strip() { find $$$$* -type f -executable | xargs -d '\n' $(TARGET_CROSS)strip --strip-unneeded || true; }; do_strip
 
 CONFIGURE_VARS += \
        CPP="$(TARGET_CROSS)cpp" \
```

I think I should really clean up the whole stripping mechanism and present it upstream.  You can't just run around and snip out crap you don't like at random.  I argued last year about the --strip-all being wrong, but now I see why that isn't breaking things -- if the rstrip.sh script detects it's relocatable, it treats it like a kernel module -- ugly.

Andy2244, 2021-09-08:
Its related to this bug: #10783

I had to disable the default rstrip script, since it breaks sambas private rpath.