openwrt/packages #16535: bandwithd linking errors

neheb, 2021-09-06:
Maintainer: @padre-lacroix 
Environment: ath79

I can't explain these and other linking failures that I've been getting locally:

```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.o: in function `sqliteCheckTables':
sqlite.c:(.text+0x4a): undefined reference to `sqlite3_get_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x4e): undefined reference to `sqlite3_get_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x50): undefined reference to `sqlite3_get_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x54): undefined reference to `sqlite3_get_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x86): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x8a): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x8c): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x90): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x9a): undefined reference to `sqlite3_free'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x9e): undefined reference to `sqlite3_free'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xa0): undefined reference to `sqlite3_free'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xa4): undefined reference to `sqlite3_free'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xae): undefined reference to `sqlite3_close'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xb2): undefined reference to `sqlite3_close'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xb8): undefined reference to `sqlite3_close'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xbc): undefined reference to `sqlite3_close'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xca): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xce): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xd0): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xd4): undefined reference to `sqlite3_free_table'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xfa): undefined reference to `sqlite3_exec'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0xfe): undefined reference to `sqlite3_exec'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x100): undefined reference to `sqlite3_exec'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x104): undefined reference to `sqlite3_exec'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x132): undefined reference to `sqlite3_exec'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.o:sqlite.c:(.text+0x136): more undefined references to `sqlite3_exec' follow
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.o: in function `sqliteInit':
sqlite.c:(.text+0x23a): undefined reference to `sqlite3_open'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x262): undefined reference to `sqlite3_open'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x264): undefined reference to `sqlite3_open'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x268): undefined reference to `sqlite3_open'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.o: in function `sqliteDetermineSensorID':
sqlite.c:(.text+0x38e): undefined reference to `sqlite3_prepare_v2'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x392): undefined reference to `sqlite3_prepare_v2'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x394): undefined reference to `sqlite3_prepare_v2'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x398): undefined reference to `sqlite3_prepare_v2'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-11.2.0_musl/lib/gcc/powerpc-openwrt-linux-musl/11.2.0/../../../../powerpc-openwrt-linux-musl/bin/ld: sqlite.c:(.text+0x3c6): undefined reference to `s [... cut ...]
make[2]: Leaving directory '/home/mangix/devstuff/openwrt/feeds/packages/utils/bandwidthd'
time: package/feeds/packages/bandwidthd/no-db/compile#2.44#0.72#4.09
    ERROR: package/feeds/packages/bandwidthd failed to build (build variant: no-db).
make[1]: *** [package/Makefile:116: package/feeds/packages/bandwidthd/compile] Error 1
make[1]: Leaving directory '/home/mangix/devstuff/openwrt'
make: *** [/home/mangix/devstuff/openwrt/include/toplevel.mk:230: package/bandwidthd/compile] Error 2
```

They seem legit though. It seems -lsqlite3 is missing somewhere. Not sure where.

padre-lacroix, 2021-09-07:
Yes: it seems that links to sqlite.  I have not checked further in the problem: I may have time to do it this week, otherwise it mayhave to go to the next weekend, as I am quite busy at this moment. I also need to build a virtual machine for the build environment, as the one that I have is quite old.
I have a few questions:
- Does it hapens only on the powwer-pc architecture, or all the architecture?
- Do you know if sqlite changed lately?  I looked, and it does not seemed to have change lately.

neheb, 2021-09-07:
This ended up being a ccache bug. Closing.

closed 2021-09-07 by neheb