openwrt/packages #16544: python3-light: bluetooth disabled by default

divanikus, 2021-09-06:
Hi. 

We are working on `imx6` firmware and stumbled upon base `python3-light` isn't having bluetooth support enabled by default.  The platform, by itself, contains onboard bluetooth, and as far as it is more of IoT device, it would be really cool to have support of it through `python3`. 

What are proposed steps in that case? We don't really wish to fork `python3-light` with different name. Having a separate private feed for it is also doesn't seem to be good, because it might mess up things for users. So what can we do? Any suggestions are welcome.

jefferyto, 2021-09-23:
If you compile the Python package locally, you can enable Bluetooth support in `make menuconfig` (the python3-light package should have a "Configuration" option). I believe Bluetooth isn't enabled by default to save disk space.

@commodo Anything you would like to add?

divanikus, 2021-09-23:
@jefferyto We are aware of it, but how to distribute it? No one is able to decipher if that `python` is compiled with bluetooth support or not just by observing package name. I doubt how `opkg` would handle it if we add it into our custom feed - would it download our "proper" version or official one? Also lots of dependency hell lies here I guess.

According to our test difference between `python` with bluetooth enabled and disabled is something around mere 2-5 KB.

jefferyto, 2021-09-23:
>  I doubt how `opkg` would handle it if we add it into our custom feed - would it download our "proper" version or official one?

If your python3-light package has a "newer" version string, then opkg should favour it over the one provided in the packages feed.

> According to our test difference between `python` with bluetooth enabled and disabled is something around mere 2-5 KB.

Does this include the library dependency for Bluetooth support (bluez-libs)?

divanikus, 2021-09-23:
> If your python3-light package has a "newer" version string, then opkg should favour it over the one provided in the packages feed.

Well, I don't believe it is a right thing to do, because it is not a trully "newer" version, unless there is a specific semantic for that, maybe something like Debian's epochs.

> Does this include the library dependency for Bluetooth support (bluez-libs)?

Unfortunately not, I'm talking just about `python3-light` package itself.

jefferyto, 2021-09-23:
> Well, I don't believe it is a right thing to do, because it is not a trully "newer" version, unless there is a specific semantic for that, maybe something like Debian's epochs.

That's why I put "newer" in quotation marks. I believe opkg handles version numbers in a similar, if not the same, way as apt/dpkg does, including support for epochs and things like `+really`.

divanikus, 2021-09-23:
I'll try to sum up the problem. It's not about how to trick `opkg` for exact one package, but rather about how to properly handle this situation for end user, which can install programs from both our feed and official feeds. And we need to distribute not just our software, but a variant of an official package.

jefferyto, 2021-09-23:
I use Ubuntu which has a [libreoffice](https://packages.ubuntu.com/hirsute-updates/libreoffice) package. I also have the [LibreOffice Fresh PPA](https://launchpad.net/~libreoffice/+archive/ubuntu/ppa) in my sources. Both official and PPA sources have a "libreoffice" package. When I install libreoffice, I get the one from the PPA because it has a higher version number.

If you have both the packages feed and a custom feed in your opkg sources and you install a package available on both feeds, I would assume opkg would behave in the same way (pick the one with the higher version number).

If it doesn't, then I would suggest asking on the [mailing list](https://lists.openwrt.org/mailman/listinfo/openwrt-devel) or on the [forum](https://forum.openwrt.org/), or filing a bug in the [bug tracker](https://bugs.openwrt.org/), because this would be an issue with opkg, not the Python package.

divanikus, 2021-09-23:
The problem is, your PPA is meant for having a fresher version of LibreOffice, while we need something with exact same as in core, but with different compile flags. So your suggestion is just to make it's version "higher" artificially?

jefferyto, 2021-09-24:
> So your suggestion is just to make it's version "higher" artificially?

Yes - perhaps `3.9.7-1custom1`.

Or you can name your package something else and have it `PROVIDES:=python3`.

Or you can clone/compile the whole packages feed and provide it instead of the official packages feed. (The other advantage of this is you don't get unexpected updates/changes from the official packages feed.)

commodo, 2021-09-25:
apologies for the late reply;
these days i'm a bit more busy with life

this situation with having the same version of a package but with different CFLAGS is a bit new to me;
there is the VARIANT mechanism that was meant to handle this situation, but we're still using it for some old Python 2/3 variant stuff;
even so, VARIANTs for Python3 [with different CFLAGS] is a bit much;

the idea of supplying `3.9.7-1custom1` is a good one;

but stepping back a bit, i'm wondering if we could re-visit this bluetooth support;
we could try to split a `python3-bluetooth` package which pulls in `bluez-libs` on installation;
but we would need to make sure that the `socketmodule` can build with just `bluetooth.h`
the situation may have changed since I initially added this bluetooth support;
bluez-libs may not be required;
looking through cpython source (in socketmodule.h & .c) i can't tell where bluez is actually required anymore;

i will take a look and come back;

commodo, 2021-09-25:
good news: bluez-libs seems required only for build-time;

i don't remember why it was packaged, or whether i did not pay attention to how bluez was needed;
moving it from DEPENDS to PKG_BUILD_DEPENDS seems sufficient;

@divanikus   can you test this change on your setup to see that BT works?   https://github.com/commodo/packages/commit/3c27c138de5cc85f965ce4a5a63778197de63f22
i only tested this briefly with a snippet from `Lib/test/test_socket.py`

this does increase the build-time of Python3 (unfortunately), but it does allow BT support on by default with minimal size increase;

jefferyto, 2021-09-26:
@commodo that's good news, thanks for looking into this

commodo, 2021-09-27:
created PR https://github.com/openwrt/packages/pull/16748

divanikus, 2021-09-27:
As a drop-in replacement it surely works, we have checked. But we had bluez installed already by other software. I'll try to spin up a fresh instance.

closed 2021-09-27 by neheb

commodo, 2021-09-27:
> @commodo As far as I remember, the reason behind disabling bluetooth was runtime crashes (because of missing libs) in socket module on systems without bluetooth libs installed. Seems like it's not the case anymore, 'cause I can't see any dependency to bluez libs via `ldd`.

maybe that was the case with Python2;
who knows?
there were some differences between them;

but i don't feel like chasing the past now :)