openwrt/packages #16591: <wifidog>: v1.3.0-6 new realese morethan 1 session cant work

henok400, 2021-09-09:
OpenWrt 21.02 package wifidog1.3.0-6 cant work more than 1 wifi user login the auth server logout all session users. this issue not have the previous packages. im from cloudwifizone wifidog base service provider.

henok400, 2021-09-10:
This is system log show wifidog after give internet access the 2nd device login automatically limit internet 
```
Fri Sep 10 15:32:24 2021 daemon.info wifidog[1864]: Captured 10.0.1.165 requesting [http%3A%2F%2Fwww.msftncsi.com%2Fncsi.txt] and re-directing them to login page
Fri Sep 10 15:32:25 2021 daemon.info wifidog[1864]: Auth server returned authentication code 1
Fri Sep 10 15:32:25 2021 daemon.info wifidog[1864]: Got ALLOWED from central server authenticating token e6sf19c5i0cwzfrbx0bi21v4jubhjo2d from 10.0.1.165 at 1c:bf:ce:22:3e:56 - adding to firewall and redirecting them to portal

Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Received connection from 10.0.1.165, spawning worker thread
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Got client MAC address for ip 10.0.1.165: 1c:bf:ce:22:3e:56
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Check host 149.154.167.51:80 is in whitelist or not
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: rule mask cloudwifizone.com
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: host 149.154.167.51:80 is not in cloudwifizone.com, continue
Fri Sep 10 15:35:50 2021 daemon.info wifidog[1864]: Captured 10.0.1.165 requesting [http%3A%2F%2F149.154.167.51%3A80%2Fapi] and re-directing them to login page

```

henok400, 2021-09-13:
I have comparing the previous version and replace it some .ipk files problem show openwrt v21.02
This is wifidog install openwrt v21.02
```
Installing wifidog (1.3.0-6) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/packages/mips_24kc/packages/wifidog_1.3.0-6_mips_24kc.ipk
Installing kmod-ipt-extra (5.4.143-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/kmod-ipt-extra_5.4.143-1_mips_24kc.ipk
Installing iptables-mod-extra (1.8.7-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/iptables-mod-extra_1.8.7-1_mips_24kc.ipk
Installing kmod-ipt-ipopt (5.4.143-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/kmod-ipt-ipopt_5.4.143-1_mips_24kc.ipk
Installing iptables-mod-ipopt (1.8.7-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/iptables-mod-ipopt_1.8.7-1_mips_24kc.ipk
Installing kmod-ipt-nat-extra (5.4.143-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/kmod-ipt-nat-extra_5.4.143-1_mips_24kc.ipk
Installing iptables-mod-nat-extra (1.8.7-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/iptables-mod-nat-extra_1.8.7-1_mips_24kc.ipk
```
And wifidog v 1.3.0-4 in openwrt v21.02.0
```
root@OpenWrt:~# opkg install wifidog
Installing wifidog (1.3.0-4) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mips_24kc/packages/wifidog_1.3.0-4_mips_24kc.ipk
Installing kmod-ipt-extra (4.14.241-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/kmods/4.14.241-1-342af9e4f67b3447c53216ab8e3b12a1/kmod-ipt-extra_4.14.241-1_mips_24kc.ipk
Installing iptables-mod-extra (1.8.3-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/packages/iptables-mod-extra_1.8.3-1_mips_24kc.ipk
Installing kmod-ipt-ipopt (4.14.241-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/kmods/4.14.241-1-342af9e4f67b3447c53216ab8e3b12a1/kmod-ipt-ipopt_4.14.241-1_mips_24kc.ipk
Installing iptables-mod-ipopt (1.8.3-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/packages/iptables-mod-ipopt_1.8.3-1_mips_24kc.ipk
Installing kmod-ipt-nat-extra (4.14.241-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/kmods/4.14.241-1-342af9e4f67b3447c53216ab8e3b12a1/kmod-ipt-nat-extra_4.14.241-1_mips_24kc.ipk
Installing iptables-mod-nat-extra (1.8.3-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ar71xx/generic/packages/iptables-mod-nat-extra_1.8.3-1_mips_24kc.ipk
```
I have replace old version wifidog v1.3.0-4 into openwrt v21.02 the following .ipk 
-  wifidog (1.3.0-4)
-  iptables-mod-extra (1.8.3-1)
-  iptables-mod-ipopt (1.8.3-1) 
-  iptables-mod-nat-extra (1.8.3-1)
But still problem continued. It cant work more than 1 device. if 2nd device connect to internet then all device disconnected. the captive portal start authentication and success but no internet.    
Im focusing openwrt v21.02 is the issue ip table cant manage it.

deivisonmoraes, 2021-12-09:
I have same problem... Wifidog firewall crachs apparently. 

There is a problem in this version of iptables as there is no iptables aliases (needed for the old syntax of wifidog commands). One of the errors this causes is in the command: iptables -t filter -A WiFiDog_br-lan_Internet -m state --state INVALID -j DROP
iptables v1.8.7 (legacy): Couldn't load match `state':No such file or directory

I don't know if this problem can be related. I tested it with the same settings on a debian gateway and this problem doesn't happen.

However the problem of connecting only 1 user occurs with me too.

tks