openwrt/packages #16594: adguardhome: panic: qtls.ClientHelloInfo doesn't match

foxcris, 2021-09-09:
Maintainer: @dobo90
Environment: 
Openwrt Version: OpenWrt 21.02.0, r16279-5cc0535800
Hardware: fritzbox 4040

Description:
adguardhome is listed as started but no adguardhome proess is running

Executing the AdGuardHome binary manually gives the following error
```
root@owrt1:/etc/openvpn# /usr/bin/AdGuardHome --help
panic: qtls.ClientHelloInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-15.init.0()
        github.com/marten-seemann/qtls-go1-15@v0.1.1/unsafe.go:20 +0x168
```

derterz, 2021-09-12:
Same here after upgrading from an old nightly 21.02 build to 21.02.0 r16279-5cc0535800 on my Linksys MR8300 (Dallas) [ARMv7 Processor rev 5 (v7l)].

Manually downloading and running the latest version works fine: https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.106.3/AdGuardHome_linux_armv7.tar.gz

BKPepe, 2021-09-14:
I cherry-picked all the updates related to adguardhome from the ``master`` branch into ``openwrt-21.02``. Hopefully, it will solve your issues. If yes, then please close it. If not, feel free to discuss it with the maintainer.

predators46, 2021-09-14:
hmm. i tried on lede 17.01.x it runs fine

[image]

BKPepe, 2021-09-14:
LEDE 17.01.x is not supported. Supported versions are OpenWrt 19.07 and OpenWrt 21.02.

predators46, 2021-09-14:
> LEDE 17.01.x tidak didukung. Versi yang didukung adalah OpenWrt 19.07 dan OpenWrt 21.02.

> LEDE 17.01.x is not supported. Supported versions are OpenWrt 19.07 and OpenWrt 21.02.

Yes, I know

HinataKato, 2021-09-14:
Same, I was able to use it at 21.02.0-rc4, after upgrading to 21.02.0 it doesn't work

foxcris, 2021-09-14:
@BKPepe how am i able to perform a test? Till now there is no updated package available via opkg. Do just have to wait a bit?

foxcris, 2021-09-15:
The new package is now available. Unfortunately it does not resolve the issue:
````
root@owrt1:~# /usr/bin/AdGuardHome --help
panic: qtls.CertificateRequestInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-16.init.0()
        github.com/marten-seemann/qtls-go1-16@v0.1.3/unsafe.go:17 +0x1e0
````

EdiTurn, 2021-09-17:
Also got this issue in master. 
Currently I change the version to 0.107.0-b.11 in Makefile, and it works well.

Nadias-jp, 2021-10-04:
I overwrote AdguardHome and it worked.
https://github.com/AdguardTeam/AdGuardHome/releases/tag/v0.106.3

```
panic: qtls.CertificateRequestInfo doesn't match
 
goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-16.init.0()
        github.com/marten-seemann/qtls-go1-16@v0.1.3/unsafe.go:17 +0x248
```

closed 2021-12-25 by BKPepe