openwrt/packages #16595: Strongswan does not work after upgrading from OpenWrt 19.07 to 21.02 (stable)

joker32, 2021-09-09:
After upgrading to OpenWrt 21.02, Strongswan no longer works.
The logs say the following after trying a VPN connection from my iPhone.

Thu Sep  9 12:06:53 2021 daemon.info : 08[KNL] received netlink error: No such file or directory (2)
Thu Sep  9 12:06:53 2021 daemon.info : 08[KNL] unable to add SAD entry with SPI cb9d49fb (FAILED)
Thu Sep  9 12:06:53 2021 daemon.info : 08[KNL] received netlink error: No such file or directory (2)
Thu Sep  9 12:06:53 2021 daemon.info : 08[KNL] unable to add SAD entry with SPI 0b270962 (FAILED)
Thu Sep  9 12:06:53 2021 daemon.info : 08[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Thu Sep  9 12:06:53 2021 daemon.info : 08[IKE] failed to establish CHILD_SA, keeping IKE_SA

Knogle, 2021-09-24:
I think it is a similar issue that i have.
CONFIG_TCP_MD5SIG is missing in the kernel options since 21.02, it breaks some packages. Yours might be affected as well.

joker32, 2021-10-31:
Do I have a chance to fix the error? Is there a workaround? With a reinstallation to 21.02.0 it worked somehow. After the sysupgrade to 21.02.1 I now have the same problem again.

joker32, 2021-11-01:
opkg install kmod-crypto-gcm - solves the problem

https://github.com/openwrt/packages/issues/14360#issuecomment-917567474

closed 2021-11-01 by joker32