openwrt/packages #16599: mwan3: Wireguard client prevents mwan3 policy based routing for wireguard connection

ingenium13, 2021-09-09:
Maintainer: @feckert 
Environment: 19.07.6

Description:

My issue seems to be the same as #12800. I have two WANs setup in failover, one LTE and one DSL. My mwan3 policies set the LTE one to be preferred, with DSL being failover. Wireguard is configured as a client connecting to a remote server for a site to site VPN (ie, 0.0.0.0/0 isn't set as an allowed IP, only a few subnets are routed over the VPN). Wireguard adds a static route to the VPN server over the DSL interface, preventing the connection from being marked by mwan3. So no matter what it always goes over the DSL interface.

I tried setting route_allowed_ips to 0, and that correctly removes the routes for allowed IPs, however it still adds a route for the VPN server. I can manually remove the route once the tunnel is up, but by that point it's too late and the connection remains unmarked. I tried adding the iptables rules in https://github.com/openwrt/packages/issues/12800#issuecomment-657763513 but they didn't have any effect and the connection remained unmarked (or rather it continued to be marked with the mask). I tried adding a static route for the VPN server IP with a metric of 0, and that works for the VPN, but then it prevents any other access between the server IP and the DSL interface, which is the only interface allowing inbound connections (LTE has CGNAT). So that's not a viable solution either. And I assume that this would also break failover when LTE goes down.

champtar, 2021-09-09:
Have you tried `nohostroute` setting ?

ingenium13, 2021-09-09:
> Have you tried `nohostroute` setting ?

Thank you! Yes that did it. Luci would overwrite it in /etc/config/network and remove it from my config every time I started the tunnel, so I had to edit /usr/lib/wireguard/startvpn.sh to force add it to the final config, but after that it works and no longer adds the route. mwan3 then correctly marks the traffic.

closed 2021-09-09 by ingenium13

champtar, 2021-09-09:
> Thank you! Yes that did it. Luci would overwrite it in /etc/config/network and remove it from my config every time I started the tunnel, so I had to edit /usr/lib/wireguard/startvpn.sh to force add [...]

Can you open a ticket to have luci-wireguard fixed ?

feckert, 2021-09-10:
This is option is already applied into LuCI
https://github.com/openwrt/luci/commit/9684901dc27e37900d4c4cf09b1de3b836649e0e