openwrt/packages #16601: fail2ban: cannot be disabled

wulfy23, 2021-09-10:
Maintainer: @erdoukki 
Environment: master all

Description: service does not honour disabled.

```
/etc/init.d/fail2ban disable
/etc/init.d/fail2ban stop
################################### [root@dca632 /usbstick 54°] uci show firewall | tail -n4
firewall.@include[1]=include
firewall.@include[1].path='/etc/firewall.fail2ban'
firewall.@include[1].enabled='1'
firewall.@include[1].reload='1'
################################### [root@dca632 /usbstick 53°] head /etc/firewall.fail2ban 
#!/bin/sh
/etc/init.d/fail2ban restart
exit 0
################################### [root@dca632 /usbstick 55°] ifup wan; ps w | grep fail
18000 root     21420 S    {fail2ban-server} /usr/bin/python3 /usr/bin/fail2ban-server -xf -p /var/run/fail2ban/fail2ban.pid -s /var

```

erdoukki, 2021-09-10:
Héllo, thanks for the report...

For me all services works like this !
disable a service only stop it for autostart at boot...
You may still restart it and then, disabled or not, it will startup !

I just tried with syslog-ng to confirm this :

```
root@LPM:~# service syslog-ng stop
root@LPM:~# ps | grep syslog
 1763 root     16056 S    /usr/sbin/rsyslogd -n
 3429 root       872 S    /usr/sbin/addrwatch_syslog
28726 root      1348 R    grep syslog
root@LPM:~# service syslog-ng disable
root@LPM:~# service syslog-ng restart
Command failed: Not found
root@LPM:~# ps | grep syslog
 1763 root     16056 S    /usr/sbin/rsyslogd -n
 3429 root       872 S    /usr/sbin/addrwatch_syslog
28753 root     13316 S    /usr/sbin/syslog-ng --foreground
28765 root      1348 R    grep syslog
root@LPM:~# service syslog-ng stop
```

erdoukki, 2021-09-10:
If the problem is about firewall script, then I may look at adding in service disabling/enabling command a remove/add of the firewall script ?

erdoukki, 2021-09-10:
Because of the OpenWrt global aspect of this issue, I open a topic on the forum : https://forum.openwrt.org/t/service-still-restart-start-while-disabled/105996

erdoukki, 2021-09-10:
https://bugs.openwrt.org/index.php?do=details&task_id=4026

erdoukki, 2021-09-10:
I propose as a quikfix workaround : to modify /etc/firewall.fail2ban as follow
```
#!/bin/sh
/etc/init.d/fail2ban enabled && /etc/init.d/fail2ban restart
exit 0
```
If okay, I will propose a PR with this modification.

closed 2021-09-10 by wulfy23