openwrt/packages #16604: autoconf: /usr/bin/autoreconf: /builder/shared-workdir/build/sdk/staging_dir/host/bin/perl: bad interpreter: No such file or directory

acooler15, 2021-09-10:
Maintainer: @neheb
Environment:
 OpenWrt 21.02-SNAPSHOT, r16282-601864c09e
Description:
Autoreconf's Shebang is wrong
```
root@OpenWrt:~# opkg install autoconf
Installing autoconf (2.70-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/aarch64_cortex-a72/packages/autoconf_2.70-1_aarch64_cortex-a72.ipk
Configuring autoconf.
root@OpenWrt:~# autoreconf 
-ash: autoreconf: not found
root@OpenWrt:~# bash
root@OpenWrt:~# autoreconf 
bash: /usr/bin/autoreconf: /builder/shared-workdir/build/sdk/staging_dir/host/bin/perl: bad interpreter: No such file or directory
root@OpenWrt:~# less /usr/bin/autoreconf
```
[image]

acooler15, 2021-09-10:
Add `CONFIGURE_VARS += PERL=/bin/perl` to Makefile  might solve 。。。

neheb, 2021-09-12:
Patch welcome

closed 2021-09-12 by neheb