openwrt/packages #16623: 'nginx-ssl': /usr/lib/libxml2.so: undefined reference to `libiconv'

alex9434, 2021-09-13:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment:  OpenWrt 21.02.0 for bcm2711 (Raspberry Pi 4) cross-compiled on Debian Bullseye on amd64

Description:
```
objs/ngx_modules.o \
-L/media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib -L/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/usr/lib -L/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib -znow -zrelro -Wl,--gc-sections -Wl,-E -L/media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib -llua -lm -ldl -lxml2 -lubus -lubox -lblobmsg_json -ljson-c -lpthread -lm -lpcre -lssl -lcrypto -lz \
-Wl,-E
/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: warning: libiconv.so.2, needed by /media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/libxml2.so, not found (try using -rpath or -rpath-link)
/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/libxml2.so: undefined reference to `libiconv'
/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/libxml2.so: undefined reference to `libiconv_close'
/media/alex/data/share/internal/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /media/alex/data/share/internal/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/libxml2.so: undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
make[5]: *** [objs/Makefile:433: objs/nginx] Error 1
make[5]: Leaving directory '/media/alex/data/share/internal/openwrt/build_dir/target-aarch64_cortex-a72_musl/nginx-ssl/nginx-1.19.6'
make[4]: *** [Makefile:8: build] Error 2
make[4]: Leaving directory '/media/alex/data/share/internal/openwrt/build_dir/target-aarch64_cortex-a72_musl/nginx-ssl/nginx-1.19.6'
make[3]: *** [Makefile:549: /media/alex/data/share/internal/openwrt/build_dir/target-aarch64_cortex-a72_musl/nginx-ssl/nginx-1.19.6/.built] Error 2
make[3]: Leaving directory '/media/alex/data/share/internal/openwrt/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/ssl/compile#114.94#14.96#132.08
    ERROR: package/feeds/packages/nginx failed to build (build variant: ssl).
make[2]: *** [package/Makefile:114: package/feeds/packages/nginx/compile] Error 1

```

mhei, 2021-09-17:
Could you please provide your build configuration, e.g. output of `scripts/diffconfig.sh`. Thanks.

alex9434, 2021-09-21:
Apologies for the delay in getting back. I found the issue. I have activated the option " Compile with full language support" under "Global build settings". De-activating that option solved the problem. Do you want me to post the updated build-config?

mhei, 2021-09-22:
Thanks for your feedback. But if I understand correctly, then when you switch "on" this option, then the builds fails reproducible? That should not be the case... so yes, I'd be interested in that build-config ... reason is, that I already tried this and it worked for me, but maybe I overlooked something. So having your exact config would maybe help to see differences between my setup.

alex9434, 2021-10-07:
Hi @mhei, the issue was caused because I did the compilation on an arm64 computer. I did not have the same issue with amd64.