openwrt/packages #16630: mwan3: Disables tracking with error (16) on no response to dhcp discover

compliment, 2021-09-15:
Maintainer: @feckert 
Environment:  mips_24kc, 21.02

Description:

mwan3 disables tracking with error (16) when DHCP server doesn't respond after interface link up. It should keep tracking and execute disconnected hotplug after timeout.

```
Wed Sep 15 00:04:07 2021 user.notice mwan3-hotplug[4910]: Execute disconnected event on interface wwan (wwan0)
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain test
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain onion
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain localhost
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain local
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain invalid
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain bind
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using only locally-known addresses for domain arpa
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using nameserver 203.153.41.28#53
Wed Sep 15 00:04:11 2021 daemon.info dnsmasq[23354]: using nameserver 203.153.44.44#53
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan (5073): Stopping network wwan
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan_4 (27505): udhcpc: received SIGTERM
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan_4 (27505): udhcpc: unicasting a release of 100.82.205.139 to 100.82.205.137
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan_4 (27505): udhcpc: sending release
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan_4 (27505): udhcpc: entering released state
Wed Sep 15 00:04:12 2021 daemon.notice netifd: wwan_4 (27505): Command failed: Permission denied
Wed Sep 15 00:04:12 2021 daemon.notice netifd: Interface 'wwan_4' is now down
Wed Sep 15 00:04:12 2021 daemon.notice netifd: Network alias '' link is down
Wed Sep 15 00:04:12 2021 daemon.notice netifd: Interface 'wwan_4' has link connectivity loss
Wed Sep 15 00:04:12 2021 daemon.notice netifd: Interface 'wwan_4' is disabled
Wed Sep 15 00:04:13 2021 daemon.notice netifd: wwan (5073): sending -> AT^NDISDUP=1,0
Wed Sep 15 00:04:13 2021 daemon.notice netifd: wwan (5073): Command failed: Permission denied
Wed Sep 15 00:04:13 2021 daemon.notice netifd: Interface 'wwan' is now down
Wed Sep 15 00:04:13 2021 daemon.notice netifd: Interface 'wwan' is setting up now
Wed Sep 15 00:04:55 2021 daemon.notice netifd: wwan (5166): sending -> AT
Wed Sep 15 00:04:56 2021 daemon.notice netifd: wwan (5166): sending -> ATZ
Wed Sep 15 00:04:57 2021 daemon.notice netifd: wwan (5166): sending -> ATQ0
Wed Sep 15 00:04:57 2021 daemon.notice netifd: wwan (5166): sending -> ATV1
Wed Sep 15 00:04:58 2021 daemon.notice netifd: wwan (5166): sending -> ATE1
Wed Sep 15 00:04:59 2021 daemon.notice netifd: wwan (5166): sending -> ATS0=0
Wed Sep 15 00:04:59 2021 daemon.notice netifd: wwan (5166): sending -> AT+CGDCONT=1,"IP","airtelgrps.com"
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): SIM ready
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): PIN set successfully
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): Configuring modem
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): Starting network wwan
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): Connecting modem
Wed Sep 15 00:05:02 2021 daemon.notice netifd: wwan (5166): sending -> AT^NDISDUP=1,1,"airtelgrps.com"
Wed Sep 15 00:05:03 2021 daemon.notice netifd: wwan (5166): Setting up wwan0
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Interface 'wwan' is now up
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Network device 'wwan0' link is up
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Network alias 'wwan0' link is up
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Interface 'wwan_4' is enabled
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Interface 'wwan_4' has link connectivity
Wed Sep 15 00:05:03 2021 daemon.notice netifd: Interface 'wwan_4' is setting up now
Wed Sep 15 00:05:03 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: started, v1.33.0
Wed Sep 15 00:05:03 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: sending discover
Wed Sep 15 00:05:06 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: sending discover
Wed Sep 15 00:05:09 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: sending discover
Wed Sep 15 00:06:12 2021 user.notice mwan3-hotplug[5071]: Execute ifdown event on interface wwan (unknown)
Wed Sep 15 00:06:12 2021 user.info mwan3track[25798]: Detect ifdown event on interface wwan (wwan0)
Wed Sep 15 00:06:12 2021 user.notice mwan3track[25798]: Interface wwan (wwan0) is offline
Wed Sep 15 00:06:14 2021 user.notice mwan3-hotplug[5580]: Execute disconnected event on interface wwan (wwan0)
Wed Sep 15 00:06:19 2021 daemon.debug dnsmasq[23354]: stopped listening on ifb4wwan0(#8475): fe80::28c1:bdff:fe35:dcb9%ifb4wwan0 port 53
Wed Sep 15 00:06:21 2021 daemon.debug dnsmasq[23354]: listening on ifb4wwan0(#8479): fe80::2015:51ff:fe04:4c38%ifb4wwan0 port 53
Wed Sep 15 00:06:40 2021 daemon.notice netifd: wwan (6268): Stopping network wwan
Wed Sep 15 00:06:40 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: received SIGTERM
Wed Sep 15 00:06:40 2021 daemon.notice netifd: wwan_4 (5349): udhcpc: entering released state
Wed Sep 15 00:06:41 2021 daemon.notice netifd: wwan_4 (5349): Command failed: Permission denied
Wed Sep 15 00:06:41 2021 daemon.notice netifd: Interface 'wwan_4' is now down
Wed Sep 15 00:06:41 2021 daemon.notice netifd: Network alias '' link is down
Wed Sep 15 00:06:41 2021 daemon.notice netifd: Interface 'wwan_4' has link connectivity loss
[... 46 lines cut ...]
Wed Sep 15 00:08:47 2021 user.notice nat6: Using temporary addresses for outgoing connections on interface wwan0
Wed Sep 15 00:08:47 2021 user.notice nat6: Done setting up nat6 for zone="wan" on devices: pppoe-pppoe wwan0
Wed Sep 15 00:08:48 2021 daemon.debug dnsmasq[23354]: stopped listening on ifb4wwan0(#8479): fe80::2015:51ff:fe04:4c38%ifb4wwan0 port 53
Wed Sep 15 00:08:50 2021 daemon.debug dnsmasq[23354]: listening on ifb4wwan0(#8483): fe80::e845:b4ff:fe92:6db0%ifb4wwan0 port 53
Wed Sep 15 00:08:51 2021 user.notice mwan3-hotplug[6886]: Execute ifup event on interface wwan (wwan0)
Wed Sep 15 00:08:51 2021 user.warn mwan3-hotplug[6886]: failed to add 172.30.1.0/24 dev br-lan proto kernel scope link src 172.30.1.1 to table 1
Wed Sep 15 00:08:52 2021 user.info mwan3track[25798]: Detect ifup event on interface wwan ()
Wed Sep 15 00:08:56 2021 user.notice firewall: Reloading firewall due to ifup of wwan (wwan0)
Wed Sep 15 00:08:57 2021 user.notice nat6: Firewall config="cfg02dc81" zone="lan" zone_masq6="0".
Wed Sep 15 00:08:57 2021 user.notice nat6: Firewall config="cfg03dc81" zone="wan" zone_masq6="1".
Wed Sep 15 00:08:57 2021 user.notice nat6: Found firewall zone_name="wan" with zone_masq6="1" zone_masq6_privacy="1".
Wed Sep 15 00:08:57 2021 user.notice nat6: Setting up masquerading nat6 for zone_name="wan" with zone_masq6_privacy="1"
Wed Sep 15 00:08:57 2021 user.notice nat6: Ensuring ip6tables chain="zone_wan_postrouting" contains our MASQUERADE.
Wed Sep 15 00:08:57 2021 user.notice nat6: Ensuring ip6tables chain="zone_wan_input" contains our permissive DNAT rule.
Wed Sep 15 00:08:57 2021 user.notice nat6: Ensuring ip6tables chain="zone_wan_forward" contains our permissive DNAT rule.
Wed Sep 15 00:08:57 2021 user.notice nat6: Found device="pppoe-pppoe" for network_name="pppoe".
Wed Sep 15 00:08:57 2021 user.notice nat6: Enabling IPv6 temporary addresses for device="pppoe-pppoe".
Wed Sep 15 00:08:57 2021 user.notice nat6: Accepting router advertisements on pppoe-pppoe even if forwarding is enabled (required for temporary addresses)
Wed Sep 15 00:08:57 2021 user.notice nat6: Using temporary addresses for outgoing connections on interface pppoe-pppoe
Wed Sep 15 00:08:57 2021 user.notice nat6: Found device="wwan0" for network_name="wwan".
Wed Sep 15 00:08:57 2021 user.notice nat6: Enabling IPv6 temporary addresses for device="wwan0".
Wed Sep 15 00:08:57 2021 user.notice nat6: Accepting router advertisements on wwan0 even if forwarding is enabled (required for temporary addresses)
Wed Sep 15 00:08:57 2021 user.notice nat6: Using temporary addresses for outgoing connections on interface wwan0
Wed Sep 15 00:08:57 2021 user.notice nat6: Done setting up nat6 for zone="wan" on devices: pppoe-pppoe wwan0
```

feckert, 2021-09-22:
From my point of view, there is nothing the mwan3 can do.
Since the interface `wwan` does not entered the `online` state in `mwan3`.

From my point of view, the `mwan3` may only send a `disconnected` event if the interface was `online` beforehand.

You would then have to write some kind of self-written watcher script that catches this error?

compliment, 2021-09-22:
Thank you for replying. It's taken care of. I'll close this issue.

closed 2021-09-22 by compliment