openwrt/packages #16645: use "command -v" instead of "which" in build script

wbsdty331, 2021-09-18:
OS: Debian 11 sid (`Linux debian 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux`)

when i use `make` command, it shows:
```
/usr/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
WARNING: Makefile 'package/feeds/packages/ieee8021xclient/Makefile' has a dependency on 'wpa_supplicant', which does not exist
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
WARNING: Makefile 'package/feeds/packages/ieee8021xclient/Makefile' has a dependency on 'wpa_supplicant', which does not exist
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
/home/wb/openwrt/staging_dir/host/bin/which: this version of `which' is deprecated; use `command -v' in scripts instead.
```

for this question, i asked in a Debian group, a Debian Developer said the maintainer don't want to support this `which` script, please use `command -v` in scripts instead.

I don't know any other Linux distro has this situation.

FranklinYu, 2021-09-18:
FWIW, `which` has never been in the POSIX standard (AFAIK): https://pubs.opengroup.org/onlinepubs/9699919799/idx/utilities.html

Full story: [*Why not use "which"? What to use then?* - Unix & Linux Stack Exchange](https://unix.stackexchange.com/questions/85249/why-not-use-which-what-to-use-then)

wbsdty331, 2022-02-07:
Looks like 21.02.1 fixed

closed 2022-02-07 by wbsdty331