openwrt/packages #16649: openblas failures

neheb, 2021-09-19:
Maintainer: @commodo 

https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/openblas/compile.txt
https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/openblas/compile.txt

commodo, 2021-09-22:
@bhack @brada4 

can you provide some input in this?
my initial thought be to disable these builds until some is interested in these architectures;

bhack, 2021-09-23:
I see the first log the same platform as https://github.com/xianyi/OpenBLAS/issues/2564 ?

brada4, 2021-09-23:
It used to build 2 months ago when last change was made?
What else has changed in between:(maximum dependencies shown)
Build - `perl` `python` `make` `(g)cc`
Runtime `-kc -lm -lpthread`

neheb, 2021-09-23:
It has never been built with these platforms.

brada4, 2021-09-23:
More generic package works on your targeted platforms. It was done so to keep it small and maintainable.
MIPS24K and GENERIC targets were fixed "upstream" to get to current platform list

In openwrt package makefile you see ARCH aliasing - what is where your optimisation trip begins (it's counterpart in OpenBLAS is in Makefile.system). Probably any -march=mips64v99 benefit will be clamped by mandatory -Os in same command line. But your mileage may vary.

commodo, 2021-09-25:
> It used to build 2 months ago when last change was made?

so, there may be a slight confusion with OpenWrt's CI;
the one in Github is a minimal build-test with a few archs so that things don't slip in too blatantly;
a more complete arch-list is https://downloads.openwrt.org/snapshots/targets/  and there are some sub-archs (or sub-builds) in those folders;
well, it's not quite an arch list, but more of a platform-list;
in the faillogs is where the archs are shown a bit more clear: https://downloads.openwrt.org/snapshots/faillogs/
( but in the faillogs things show up only if they fail )

e.g. `bcm27xx` is the SoC family for all Raspberry Pis, but under that folder, there are some ARM32, one ARM64, depending on more specifics of each RPi 1,2,3,4 SoC

what i want to say here is: i knew (when getting OpenBLAS merged in) that we were not going to be able to cover all archs in a simple manner, without actually merging the package in, and letting this full CI kick in;
i could (obviously) reproduce a similar CI on my end, but ¯\_(ツ)_/¯   (i don't have the time unfortunately yet)

> MIPS24K and GENERIC targets were fixed "upstream" to get to current platform list

there's a MIPS 24K and a MIPSEL 24K (i assume the second is little endian); and it's that one that it's failing;
i don't have an idea if that makes any difference for OpenBLAS;

moving forward: 
* it looks like for MIPSEL I am using `OPENBLAS_TARGET:=MIPS24K` which may not be correct; would GENERIC be sufficient?
* for mips64_octeon it fails; but for malta (mips64) i can't seem to see a failure; maybe also GENERIC for MIPS64?

commodo, 2021-09-25:
> Is the first log the same platform as [xianyi/OpenBLAS#2564](https://github.com/xianyi/OpenBLAS/issues/2564) ?

not sure; it's MIPSEL 24KC ; may differ from MIPS 24KC

neheb, 2021-09-25:
Honestly it sounds like mipsel platforms should be using the mips makefile.

bhack, 2021-09-25:
> > Is the first log the same platform as [xianyi/OpenBLAS#2564](https://github.com/xianyi/OpenBLAS/issues/2564) ?
> 
> not sure; it's MIPSEL 24KC ; may differ from MIPS 24KC

I think is the same just little endian with the  `-EL` flag?

https://gcc.gnu.org/onlinedocs/gcc/MIPS-Options.html

brada4, 2021-09-26:
Would be nice to get invoked commands that cannot be derived from Makefile-s used to produce named output and change title to reflect essence of porting effort in progress.

commodo, 2021-09-26:
> Honestly it sounds like mipsel platforms should be using the mips makefile.

ah ; makes sense;
i've opened PR https://github.com/xianyi/OpenBLAS/pull/3387 
this fixes `mips{64}el` builds

> I think is the same just little endian with the `-EL` flag?
> 
> https://gcc.gnu.org/onlinedocs/gcc/MIPS-Options.html

seems so; i was a bit confused about the OpenBLAS Makefile system

> Would be nice to get invoked commands that cannot be derived from Makefile-s used to produce named output and change title to reflect essence of porting effort in progress.

i think i don't fully understand what you want to say here;
i'm suspecting you may want some detailed info/commands about how Makefiles are getting invoked?

if that's the case, for mips64, the make command is:
```
CFLAGS="-Os -pipe -mno-branch-likely -march=octeon+ -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/aardelean/work/openwrt/openwrt/build_dir/target-mips64_octeonplus_64_musl/OpenBLAS-0.3.17=OpenBLAS-0.3.17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/usr/include -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include/fortify -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -march=octeon+ -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/aardelean/work/openwrt/openwrt/build_dir/target-mips64_octeonplus_64_musl/OpenBLAS-0.3.17=OpenBLAS-0.3.17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/usr/include -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include/fortify -I/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include " LDFLAGS="-L/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/usr/lib -L/home/aardelean/work/openwrt/openwrt/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/lib -znow -zrelro " make  -C /home/aardelean/work/openwrt/openwrt/build_dir/target-mips64_octeonplus_64_musl/OpenBLAS-0.3.17/. AR="mips64-openwrt-linux-musl-gcc-ar" AS="mips64-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -march=octeon+ -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/aardelean/work/openwrt/openwrt/build_dir/target-mips64_octeonplus_64_musl/OpenBLAS-0.3.17=OpenBLAS-0.3.17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips64-openwrt-linux-musl-ld NM="mips64-openwrt-linux-musl-gcc-nm" CC="mips64-openwrt-linux-musl-gcc" GCC="mips64-openwrt-linux-musl-gcc" CXX="mips64-openwrt-linux-musl-g++" RANLIB="mips64-openwrt-linux-musl-gcc-ranlib" STRIP=mips64-openwrt-linux-musl-strip OBJCOPY=mips64-openwrt-linux-musl-objcopy OBJDUMP=mips64-openwrt-linux-musl-objdump SIZE=mips64-openwrt-linux-musl-size CROSS="mips64-openwrt-linux-musl-" ARCH="mips64" CROSS=1 HOSTCC=gcc CROSS_SUFFIX=mips64-openwrt-linux-musl- BINARY=64 NUM_THREADS=2 PREFIX=/usr COMMON_OPT="" TARGET=GENERIC NOFORTRAN=1 ;
```

this command seems to work; it's with the `GENERIC` target; previously, for mips64 I was using the `I6400` target

PR #16735 (2021-09-27): libs: openblas: fix build failures on MIPSEL and MIPS64 archs
Maintainer: me
Compile tested: mips64_octeonplus, mipsel  https://github.com/openwrt/openwrt/commit/2c9a07ed28e87bbe2d221d48aaa19b04672646a5
Run tested: n/a

Fixes #16649
closed 2021-09-27 by neheb