openwrt/packages #16650: qemu failure on x86

neheb, 2021-09-19:
Maintainer: @yousong 
Environment: x86

ping @vooon 

https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/qemu/compile.txt

```
Package qemu-nbd is missing dependencies for the following libraries:
libfuse3.so.3
```

vooon, 2021-09-20:
@neheb i suppose we need to add `--disable-fuse` or `+libfuse3`
```
                   FUSE exports: YES
```

yousong, 2021-09-22:
> @neheb i suppose we need to add `--disable-fuse` or `+libfuse3`
> 
> ```
>                    FUSE exports: YES
> ```

The current Makefile `--disable-xx` such features initially and only make them available until people came and requested otherwise.

1715173329, 2021-10-30:
The error still remains... Any plan to fix it?

closed 2021-11-02 by yousong