openwrt/packages #16659: samba4 Compile error

causlayer, 2021-09-19:
Maintainer: @Andy2244 
Environment: bcm53xx  v21.02
test machine:netgear R8000
log:
`[1600/3771] Compiling libcli/auth/smbencrypt.c
[1601/3771] Compiling libcli/auth/session.c
[1602/3771] Compiling libcli/auth/credentials.c
[1603/3771] Compiling libcli/auth/schannel_state_tdb.c
[1604/3771] Compiling libcli/auth/pam_errors.c
[1605/3771] Compiling libcli/auth/spnego_parse.c
[1606/3771] Compiling libcli/lsarpc/util_lsarpc.c
[1607/3771] Compiling source4/lib/messaging/messaging_send.c
[1608/3771] Compiling lib/util/unix_privs.c
[1609/3771] Compiling libcli/ldap/ldap_ndr.c
[1610/3771] Compiling libcli/ldap/ldap_message.c
[1611/3771] Compiling libds/common/flag_mapping.c
[1612/3771] Compiling lib/audit_logging/audit_logging.c
[1613/3771] Linking bin/default/source4/lib/messaging/libMESSAGING-SEND-samba4.so
[1614/3771] Compiling source4/dsdb/common/util_links.c
[1615/3771] Linking bin/default/libds/common/libflag-mapping-samba4.so
[1616/3771] Compiling source4/dsdb/common/dsdb_access.c
[1617/3771] Compiling source4/dsdb/common/dsdb_dn.c
[1618/3771] Compiling source4/dsdb/common/util_samr.c
[1619/3771] Linking bin/default/librpc/libndr-samba-samba4.so
[1620/3771] Compiling source4/dsdb/common/util_groups.c
[1621/3771] Compiling source4/dsdb/common/util_trusts.c
[1622/3771] Linking bin/default/libcli/ldap/libcli-ldap-common-samba4.so
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: librpc/ndr/ndr_compression.c.68.o: in function `ndr_pull_compression_start':
ndr_compression.c:(.text.ndr_pull_compression_start+0x2c0): undefined reference to `inflateSetDictionary'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x2cc): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x310): undefined reference to `inflate'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x350): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x3bc): undefined reference to `inflateReset'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x3c8): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x618): undefined reference to `inflateInit2_'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x624): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x654): undefined reference to `inflate'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x668): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x718): undefined reference to `inflateReset'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x724): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x778): undefined reference to `inflateSetDictionary'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_pull_compression_start+0x784): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: librpc/ndr/ndr_compression.c.68.o: in function `ndr_push_compression_end':
ndr_compression.c:(.text.ndr_push_compression_end+0x22c): undefined reference to `deflateSetDictionary'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x238): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x27c): undefined reference to `deflate'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x28c): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x32c): undefined reference to `deflateReset'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x338): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x5b8): undefined reference to `deflateInit2_'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x5c4): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x614): undefined reference to `deflate'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x628): undefined reference to `zError'
/home/slayer/Desktop/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ndr_compression.c:(.text.ndr_push_compression_end+0x694): undefined reference to `deflateReset'
[... cut ...]
make[1]: Leaving directory '/home/slayer/Desktop/openwrt'
make: *** [/home/slayer/Desktop/openwrt/include/toplevel.mk:230：world] 错误 2`

`cat /lib64/pkgconfig/zlib.pc
prefix=/usr/local/zlib
exec_prefix=${prefix}
libdir=/lib64/
sharedlibdir=${libdir}
includedir=${prefix}/include

Name: zlib
Description: zlib compression library
Version: 1.2.11

Requires:
Libs: -L${libdir} -L${sharedlibdir} -lz
Cflags: -I${includedir}
`

I have updated zlib and I don’t know why I still can’t find it，When I compiled ramips-mt7621 and did not find this problem。

Andy2244, 2021-09-20:
Maybe check the configure stage and where samba WAF is finding zlib?

causlayer, 2021-09-24:
All the problems with using ubuntu20.04  are gone

closed 2021-09-24 by causlayer