openwrt/packages #16674: curl: target mbedtls/openssl instead of wolfssl by default

stangri, 2021-09-21:
Maintainer: me

Description: There have been multiple issues with curl relying on WolfSSL recently, for example curl/WolfSSL failing to connect to some HTTPS web-sites, while curl/OpenSSL working fine (https://github.com/openwrt/packages/issues/16652), there are memory leaks in `https-dns-proxy` which have been attributed to WolfSLL (https://github.com/openwrt/packages/issues/15513) and @neheb reference to Transmission not working properly with WolfSSL.

I'd like to start conversation about options for switching curl away from WolfSSL by default. @nbd168 @hauke @dangowrt  Can `wpad` only support WPA3 with WolfSSL exclusively? Are there reasons to not go back to using mbedTLS as a default SSL lib for both WPAD and curl? If we can't go back to mbedTLS for both `wpad` and `curl`, what would you recommend we do with curl?

Thanks!

neheb, 2021-09-21:
wpad and libcurl are unrelated btw. The reason for the switch at the time was because both packages lived in base.

stangri, 2021-09-21:
> wpad and libcurl are unrelated btw. The reason for the switch at the time was because both packages lived in base.

Of course, but since wpad is probably installed on most routers with OpenWrt, would be wasteful to use a different SSL lib for curl, wouldn't it?

neheb, 2021-09-21:
Not really. It depends on what set of packages are used.

mbedTLS and OpenSSL is widely used. wolfSSL is not.

dhewg, 2021-09-21:
But wolfssl is still the standard for wpad/hostapd/ustream-ssl/px5g. With curl using a different lib it will make it non-installable on space restricted routers.
Is upstream at least aware of the issues? On curl.se there's a big `top sponsor: wolfssl` logo, which sounds like there should be enough momentum to get bugs fixed in a timely fashion?

stangri, 2021-10-01:
> Is upstream at least aware of the issues? On curl.se there's a big `top sponsor: wolfssl` logo, which sounds like there should be enough momentum to get bugs fixed in a timely fashion?

Well, at least one of the issues is definitely dependent on the compile option used by OpenWrt, so we could tweak it, and it will address https://github.com/openwrt/packages/issues/16652. Although if it's widely known that this feature is by default disabled in wolfssl and enabled on other crypto libs, it may give false sense of security to people using wolfssl on OpenWrt and expecting it to behave similarly to wolfssl on other distros.

However some other issues, like memory leaks and/or [letsencrypt global root cert expiry](https://forum.openwrt.org/t/21-02-0-cannot-connect-to-any-letsencrypt-site-using-ssl-letsencrypt-global-root-cert-expiration/107925/1) might require someone more qualified than me to reproduce and see how to report them along with some actionable data for upstream.

stangri, 2021-10-01:
> But wolfssl is still the standard for wpad/hostapd/ustream-ssl/px5g. With curl using a different lib it will make it non-installable on space restricted routers. 

@ynezz @adschm since you have authored [switch to wolfssl for wpad](https://github.com/openwrt/openwrt/commit/a14f5bb4bd263c21e103f13279d0c2ff03e48fe5) -- can you please provide your feedback? Is it possible to create a WPA3/SAE supporting variant of wpad based on mbedtls now (don't know how to check if it's supported in hostapd)?

There are issues (some linked above and a recently surfaced issue [on the forum](https://forum.openwrt.org/t/21-02-0-cannot-connect-to-any-letsencrypt-site-using-ssl-letsencrypt-global-root-cert-expiration/107925/1) with curl built with wolfssl, and I'd like to explore options to switch curl away from wolfssl by default, but as @dhewg pointed out, it would be wasteful to have wpad depend on wolfssl and curl on mbedtls if installed on the same device. 

If it is possible to create an mbedtls-dependent variant of wpad (and other default/popular packages), could we return to mbedtls and not wolfssl by default? Any insight/opinion you could provide would be hugely appreciated.

dhewg, 2021-10-02:
The culprint for #16652 is -DWOLFSSL_STATIC_RSA. If that works with other ssl libs, but not wolfssl, enabling it for wolfssl is just elevating its feature set to those of the other ssl libs, no? IMHO that should be preferred to switching the ssl lib since the other libs do it anyway. Switching would have the same implications in the sense of security.

About the forum post:
```
$ nmap --script ssl-enum-ciphers -p 443 shop.bbc.com      
Starting Nmap 7.80 ( https://nmap.org ) at 2021-10-02 09:55 CEST
Nmap scan report for shop.bbc.com (23.227.38.74)
Host is up (0.022s latency).

PORT    STATE SERVICE
443/tcp open  https
| ssl-enum-ciphers: 
|   TLSv1.2: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256-draft (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_128_GCM_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_GCM_SHA384 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA256 (rsa 2048) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA256 (rsa 2048) - A
|     compressors: 
|       NULL
|     cipher preference: server
|_  least strength: A
```

```
$ nmap --script ssl-enum-ciphers -p 443 help.behance.net   
Starting Nmap 7.80 ( https://nmap.org ) at 2021-10-02 09:56 CEST
Nmap scan report for help.behance.net (104.16.51.111)
Host is up (0.023s latency).
Other addresses for help.behance.net (not scanned): 104.16.53.111

PORT    STATE SERVICE
443/tcp open  https
| ssl-enum-ciphers: 
|   TLSv1.2: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256-draft (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_128_GCM_SHA256 (rsa 4096) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 4096) - A
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_GCM_SHA384 (rsa 4096) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 4096) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA256 (rsa 4096) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA256 (rsa 4096) - A
|     compressors: 
|       NULL
|     cipher preference: server
|     warnings: 
|       Key exchange (ecdh_x25519) of lower strength than certificate key
|_  least strength: A
```

That looks like it may be fixed by either CONFIG_WOLFSSL_HAS_ECC25519 or DWOLFSSL_STATIC_RSA

dhewg, 2021-10-02:
Completely untested, but does this fix it?
[0001-wolfssl-enable-ecc-curve-25519-support-by-default.patch.txt](https://github.com/openwrt/packages/files/7271181/0001-wolfssl-enable-ecc-curve-25519-support-by-default.patch.txt)
[0002-wolfssl-add-and-enable-RSA-static-key-cipher-suites-.patch.txt](https://github.com/openwrt/packages/files/7271182/0002-wolfssl-add-and-enable-RSA-static-key-cipher-suites-.patch.txt)

dhewg, 2021-10-03:
Nope, the first patch doesn't fix it. It looks like it's this bug:
https://github.com/wolfSSL/wolfssl/issues/4443

dhewg, 2021-10-06:
Thanks for taking the time to respond here too, this is what open source interoperability should look like ;)

I linked it already from the WolfSSL issue, but I raised this issue on the LE forums too:
https://community.letsencrypt.org/t/wolfssl-cert-issue/162212

dhewg, 2021-10-06:
See https://github.com/openwrt/openwrt/pull/4644

stangri, 2021-10-21:
> > Is it possible to create a WPA3/SAE supporting variant of wpad based on mbedtls now
> 
> Nope.

Closing this then as it doesn't make sense to target different SSL libs between wpad and curl.

closed 2021-10-21 by stangri

gstrauss, 2022-10-06:
Constructive competition between open source projects can lead to improvements in both projects.

In that vein: request for testers (RFT) for hostapd and wpa_supplicant built with mbedtls option.
[RFT]hostapd: add mbedtls variant https://github.com/openwrt/openwrt/pull/10727