openwrt/packages #16684: lxc-create: requires getopts as a dep

graysky2, 2021-09-22:
Maintainer: @neheb 
It seems that `lxc-create` requires `getopt` as a dependency but it is not pulled down when the package is installed.

```
# lxc-create --name myLMS --template download
/usr/share/lxc/templates/lxc-download: line 241: getopt: not found
LXC container image downloader
...
lxc-create: myLMS: lxccontainer.c: create_run_template: 1616 Failed to create container from template
lxc-create: myLMS: tools/lxc_create.c: main: 319 Failed to create container myLMS
```

neheb, 2021-09-22:
I don't actually use or maintain LXC. Want to be a maintainer?

graysky2, 2021-09-22:
@neheb - My bad.  I saw you several times in the history of [Makefile](https://github.com/openwrt/packages/commits/master/utils/lxc/Makefile) plus your activity with it so assumed you were.  I see [Marko](https://github.com/openwrt/packages/blob/master/utils/lxc/Makefile#L19) hard-coded in the `Makefile` ... how can I find Marko's github nick?  Guess it is @robimarko

rmilecki, 2023-05-19:
I noticed this change while updating LXC in stable releases.

This change adds `getopt` as dependency of `lxc-create`. I see few problems here:
1. `lxc-create` requires more than just `getopt` to work
2. We already have `LXC_BUSYBOX_OPTIONS` which takes care of requirements (including `getopt`)
3. We ended up with
   * `lxc-create` selecting `getopt`
   * `LXC_BUSYBOX_OPTIONS` selecting `BUSYBOX_CONFIG_GETOPT`