openwrt/packages #16698: adguardhome: How to install and run?

pkejval, 2021-09-24:
Installed package ` adguardhome (0.106.3-1)` from Luci and what next? No new option added in Luci. So I tried SSH and see what is going on:

`root@Router:~# /etc/init.d/adguardhome status` returns `running`
but I can't see any process running in top nor ps. There is no config file `/etc/adguardhome.yml` nor in `/etc/config`. And there is no app listening at http://routerip:3000

Seems like package installation isn't working or am I doing something wrong?

My OpenWRT is running in KVM `OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175`

pkejval, 2021-09-24:
I was playing around with it and I found that:

```
root@Router:~# /usr/bin/AdGuardHome
panic: qtls.CertificateRequestInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-16.init.0()
        github.com/marten-seemann/qtls-go1-16@v0.1.3/unsafe.go:17 +0x198
```

BUT then I downloaded binary from AdguardHome Github and replaced `/usr/bin/AdGuardHome` with downloaded binary version `v0.107.0-b.11`, created config `/etc/adguardhome.yaml` and now AdGuardHome is running!

Issues with package are:

- no config file is saved in `/etc/adguardhome.yaml` after installing package
- `/usr/bin/AdGuardHome` can't be run (error above) and `/etc/init.d/adguardhome status` shows running even service isn't running

alaviss, 2021-12-20:
From a cursory web search this seems to be the issue with a dependency of AdGuard Home having hard dependency on a given version of Go. See this similar issue: https://github.com/caddyserver/caddy/issues/4118

The release version of AdGuard Home is only compatible with Go 1.16 from the error message, and OpenWrt uses Go 1.17, hence why it doesn't work.

closed 2021-12-25 by BKPepe