openwrt/packages #16729: node: build failure with binutils 2.37

LGA1150, 2021-09-25:
Maintainer: @nxhack
Environment: arm

Description:
Since commit openwrt/openwrt@f48ced582dd33b6d0921f5851c1ea2cef2ff730c, node failed to build.
See https://www.mail-archive.com/bug-binutils@gnu.org/msg38662.html

nxhack, 2021-09-26:
Hi @LGA1150 

This can be avoided by increasing the open file descriptors.

If your "shell" is bash, run the following command and try to build.
```
ulimit -n 2048
```
 
ref:
https://github.com/nodejs/node/issues/39452
https://sourceware.org/bugzilla/show_bug.cgi?id=28138

svlobanov, 2022-01-05:
@nxhack could you please add 'ulimit -n 2048' solution to Makefile? otherwise, many other will spend their time like me (create a ticket, wait for your help...)

nxhack, 2022-01-08:
@LGA1150 @svlobanov
A patch has been applied to binutils 2.37. Please check it.
https://github.com/openwrt/openwrt/commit/8c501bf9feeaadfb9d950806a312b328baabddcb

svlobanov, 2022-01-09:
@nxhack it works for me, thank you

I've checked on Ubuntu 20.04.3 LTS x86_64 with default ulimit -n (1024)

closed 2022-04-05 by LGA1150