openwrt/packages #16736: cryptsetup/kmod-crypto-iv - ca

mwoz123, 2021-09-26:
Maintainer: @jmarcet 
Environment:
```bash
# cat /etc/os-release 
NAME="OpenWrt"
VERSION="21.02.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.0"
VERSION_ID="21.02.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16279-5cc0535800"
OPENWRT_BOARD="ath79/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.0 r16279-5cc0535800"
```

# Description:

Neither can open `cryptsetup luksOpen` under Ubuntu 20.04 disc nor can `cryptsetup LuksFormat` in under OpenWrt.

Doing setup as : https://openwrt.org/docs/guide-user/storage/disk.encryption **I found missing package kmod-crypto-iv** that doesn't exist in repo.

```bash
# opkg install  kmod-crypto-iv
Unknown package 'kmod-crypto-iv'.
Collected errors:
 * opkg_install_cmd: Cannot install package kmod-crypto-iv.
 ```

Trying to open luks device:

```bash
#  cryptsetup luksOpen /dev/sdb4 sdb4
Enter passphrase for /dev/sdb4: 
Killed
root@OpenWrt:/# 
```

Trying to format as luks:
```bash
# cryptsetup -v luksFormat /dev/sdb4
WARNING: Device /dev/sdb4 already contains a 'crypto_LUKS' superblock signature.

WARNING!
========
This will overwrite data on /dev/sdb4 irrevocably.

Are you sure? (Type 'yes' in capital letters): YES
Enter passphrase for /dev/sdb4: 
Verify passphrase: 
Existing 'crypto_LUKS' superblock signature (offset: 0 bytes) on device /dev/sdb4 will be wiped.
Existing 'crypto_LUKS' superblock signature (offset: 16384 bytes) on device /dev/sdb4 will be wiped.
Not compatible PBKDF2 options (using hash algorithm sha256).
Command failed with code -1 (wrong or missing parameters).
root@OpenWrt:/#
```

Benchmark details: 
```bash
#  cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1          N/A
PBKDF2-sha256        N/A
PBKDF2-sha512        N/A
PBKDF2-ripemd160     N/A
PBKDF2-whirlpool     N/A
argon2i       4 iterations,  7734 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
argon2id      4 iterations,  7833 memory, 4 parallel threads (CPUs) for 256-bit key (requested 2000 ms time)
#     Algorithm |       Key |      Encryption |      Decryption
        aes-cbc        128b         7.8 MiB/s         8.5 MiB/s
    serpent-cbc        128b         6.0 MiB/s         6.6 MiB/s
    twofish-cbc        128b         8.4 MiB/s         9.3 MiB/s
        aes-cbc        256b         6.7 MiB/s         6.8 MiB/s
    serpent-cbc        256b         6.5 MiB/s         6.7 MiB/s
    twofish-cbc        256b         9.1 MiB/s         9.3 MiB/s
        aes-xts        256b         7.9 MiB/s         7.8 MiB/s
    serpent-xts        256b         6.3 MiB/s         6.3 MiB/s
    twofish-xts        256b         8.7 MiB/s         8.7 MiB/s
        aes-xts        512b         6.4 MiB/s         6.4 MiB/s
    serpent-xts        512b         6.3 MiB/s         6.3 MiB/s
    twofish-xts        512b         8.7 MiB/s         8.6 MiB/s
root@OpenWrt:/#
```

mwoz123, 2021-09-26:
Pinging other crypt setup committers as I'm not sure who's maintener : 
@dangowrt , @neheb

micmac1, 2021-09-26:
What happens when you install kmod-crypto-hmac and kmod-crypto-sha256?

mwoz123, 2021-09-27:
Thanks:) Works:) now I've `PBKDF2` :)

Minor issue's that I still wasn't able to open when formated partition under Ubuntu 20.04 but it must be using a different crypt method. Workaround it by doing `luksFormat` under OpenWrt :)

@micmac1, @dangowrt , @neheb
could you update in https://openwrt.org/docs/guide-user/storage/disk.encryption:
`opkg install kmod-crypto-ecb kmod-crypto-xts kmod-crypto-iv kmod-crypto-misc kmod-crypto-user cryptsetup`
to 
`opkg install kmod-crypto-ecb kmod-crypto-xts kmod-crypto-hmac kmod-crypto-sha256 kmod-crypto-misc kmod-crypto-user cryptsetup`
?
and let's close this issue report then;)

micmac1, 2021-09-27:
I looked at the wiki article. But the part where it tells you which packages to install I'm not really on board with. This is a guide for a specific setup. Maybe it doesn't fit the user's situation (it didn't fit your situation initially, you said you couldn't get it to mount a Luks volume created on Fedora or whatnot). I think it should go more along the lines of "Watch out, the package relies on the kernel's crypto API, not openssl (nor gnutls nor whatever else cryptsetup can use). So depending on what you want to do you need to install some extra kernel modules, for example {list}".

Hence  I'm not going to change this as suggested. Maybe somebody else will. Rewriting the article a bit would be another story altogether (not my favorite pastime).

Kind regards,
Seb

mwoz123, 2021-09-27:
Ok. I'm closing bug then.

Not sure if we're on the same page:
FYI in v21.02 package:
- https://openwrt.org/packages/pkgdata/kmod-crypto-iv seems to not exist
- without `kmod-crypto-hmac` and `kmod-crypto-sha256` I wasn't able to even format it under openwrt 
(see https://github.com/openwrt/packages/issues/16736#issue-1007364064 `Trying to format as luks:` part)

closed 2021-09-27 by mwoz123