openwrt/packages #16751: mwan3: iptables rule fail if ipset is not created.

castorinop, 2021-09-27:
Maintainer: Florian Eckert <fe@dev.tdt.de>,  Aaron Goodman <aaronjg@alumni.stanford.edu>
Environment: MIPS?, TP-Link Archer C20 v4, OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-21.252.54297-fc2ff4d

Description:

Iptables fail if rule use an ipset and this is not create. 
i add this line on /lib/mwan3/mwan3.sh to create the ipset before add iptables rule.
```
  if [ -n "$ipset" ]; then                  
+               $IPS create $ipset hash:ip          
                ipset="-m set --match-set $ipset dst"                            
  fi            
```

feckert, 2021-09-27:
The question here is whether mwan3 should create **ipsets**.
I think it would be better in the firewall configuration?
https://openwrt.org/docs/guide-user/firewall/firewall_configuration#ip_sets

pprindeville, 2023-09-25:
Not obvious that it shouldn't create `ipset`'s.  I think they're useful for policy-based routing, for instance my mail server can only do SMTP outbound on the primary (`wan`) interface and not the alternate (`wanb`).  It's a good application of `ipset`'s.

feckert, 2023-09-26:
I think I have not expressed my self incorrectly. Mwan3 can already use Ipsets, but it should not create them. This is the task if firewall3/firewall4. We already have a uci config for this in firewall3/firewall4. The problem is now only that mwan3 could not use them in firewall4 because of nftables.