openwrt/packages #16773: nmap: is not happy

wulfy23, 2021-09-30:
Maintainer: @nunojpg 
Environment: aarch64, r17631

Description:

```
[ /usbstick 55°] nmap -v
Error relocating /usr/bin/nmap: _ZSt28__throw_bad_array_new_lengthv: symbol not found
```

nunojpg, 2021-09-30:
Maybe related to 64 bits? Do you have any more info? I don't have 64 bits test hardware.

wulfy23, 2021-09-30:
thanks for the fast reply... no more info at this stage i'm afraid, but likely related to all the recent toolchain bumps etc.

i have a feeling this may become clearer over the next week or two... so maybe hold off on looking at it deeply for now...

wulfy23, 2021-11-03:
```
[root@dca632 /usbstick 56°]# strace nmap -v

execve("/usr/bin/nmap", ["nmap", "-v"], 0x7fd6bf6e68 /* 21 vars */) = 0
set_tid_address(0x7fbe48f248)           = 7455
brk(NULL)                               = 0x3f9f6000
brk(0x3f9f8000)                         = 0x3f9f8000
mmap(0x3f9f6000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3f9f6000
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libpcre.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libpcre.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libpcre.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=221187, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0P\25\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 290816, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbe3b5000
mmap(0x7fbe3fa000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x35000) = 0x7fbe3fa000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=270659, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0@l\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 339968, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbe362000
mmap(0x7fbe3b1000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x3f000) = 0x7fbe3b1000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=91888, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\360#\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 159744, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbe33b000
mmap(0x7fbe360000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x7fbe360000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/local/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=1706320, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\220\365\v\0\0\0\0\0"..., 960) = 960
mmap(NULL, 1785856, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbe187000
mmap(0x7fbe328000, 77824, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x191000) = 0x7fbe328000
mmap(0x7fbe338000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fbe338000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=73744, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0-\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 143360, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbe164000
mmap(0x7fbe185000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x11000) = 0x7fbe185000
close(3)                                = 0
mprotect(0x7fbe3fa000, 4096, PROT_READ) = 0
mprotect(0x7fbe3b1000, 12288, PROT_READ) = 0
mprotect(0x7fbe360000, 4096, PROT_READ) = 0
mprotect(0x7fbe328000, 61440, PROT_READ) = 0
mprotect(0x7fbe185000, 4096, PROT_READ) = 0
writev(2, [{iov_base="Error relocating /usr/bin/nmap: "..., iov_len=67}, {iov_base=": symbol not found", iov_len=18}], 2Error relocating /usr/bin/nmap: _ZSt28__throw_bad_array_new_lengthv: symbol not found) = 85
writev(2, [{iov_base="", iov_len=0}, {iov_base=NULL, iov_len=0}], 2) = 0
writev(2, [{iov_base="\n", iov_len=1}, {iov_base=NULL, iov_len=0}], 2
) = 1
mprotect(0x4ea000, 4096, PROT_READ)     = 0
exit_group(127)                         = ?
+++ exited with 127 +++

[root@dca632 /usbstick 55°]# strace zerotier-cli -v

execve("/usr/bin/zerotier-cli", ["zerotier-cli", "-v"], 0x7fd745b048 /* 21 vars */) = 0
set_tid_address(0x7faca8c248)           = 7500
brk(NULL)                               = 0x3360000
brk(0x3362000)                          = 0x3362000
[... 24 lines cut ...]
fstat(3, {st_mode=S_IFREG|0755, st_size=1706320, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\220\365\v\0\0\0\0\0"..., 960) = 960
mmap(NULL, 1785856, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fac814000
mmap(0x7fac9b5000, 77824, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x191000) = 0x7fac9b5000
mmap(0x7fac9c5000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fac9c5000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=73744, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0-\0\0\0\0\0\0"..., 960) = 960
[... cut ...]
mprotect(0x7fac812000, 4096, PROT_READ) = 0
writev(2, [{iov_base="Error relocating /usr/bin/zeroti"..., iov_len=75}, {iov_base=": symbol not found", iov_len=18}], 2Error relocating /usr/bin/zerotier-cli: _ZSt28__throw_bad_array_new_lengthv: symbol not found) = 93
writev(2, [{iov_base="", iov_len=0}, {iov_base=NULL, iov_len=0}], 2) = 0
writev(2, [{iov_base="\n", iov_len=1}, {iov_base=NULL, iov_len=0}], 2
) = 1
mprotect(0x4f7000, 4096, PROT_READ)     = 0
exit_group(127)                         = ?
+++ exited with 127 +++
```

wulfy23, 2021-11-03:
rtorrent is also effected
@neheb

wulfy23, 2021-11-03:
openzwave is also effected
@dwmw2

wulfy23, 2021-11-03:
squid is also effected
@marko.ratkaj@sartura.hr

wulfy23, 2021-11-03:
tesseract is also effected
@vk496

neheb, 2021-11-03:
did you rebuild your toolchain completely?

wulfy23, 2021-11-03:
Im using imagebuilder (official sources)...

snort3 is also effected
W. Michael Petullo <mike@flyn.org>

neheb, 2021-11-03:
belongs in https://bugs.openwrt.org then.

closed 2021-11-03 by wulfy23