openwrt/packages #16777: prometheus-node-exporter-lua: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)

stintel, 2021-09-30:
Maintainer: @champtar 
Environment: OpenWrt master r17637-fa13dd658f on ath79

Description: since updating my TP-Link EAP245 v3 to a recent master build my logs are spammed with the following error:

```
Thu Sep 30 23:33:43 2021 daemon.info prometheus-node-exporter-lua[19: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)
```

```
root@ap1:~# opkg info dawn
Package: dawn
Version: 2021-07-27-276ca169-1
Depends: libc, libubus20210630, libubox20210819, libblobmsg-json20210819, libuci20130104, libgcrypt, libiwinfo20210430, umdns
Status: install user installed
Architecture: mips_24kc
Conffiles:
 /etc/config/dawn d1bdd8723b7fb624d29b98d1d2388186d35495cc1be6ddd38054c38a4cfc4e80
Installed-Time: 1633026970

root@ap1:~# opkg info prometheus-node-exporter-lua
Package: prometheus-node-exporter-lua
Version: 2021.09.24-1
Depends: libc, luasocket, lua
Status: install user installed
Architecture: all
Conffiles:
 /etc/config/prometheus-node-exporter-lua c21bb5cd6e12fa88e0f040083516baed3da6354643f0833d30ee6de004e0c711
Installed-Time: 1633026970
```

stintel, 2021-09-30:
Updated the ticket as this is more likely a problem in dawn than in prometheus-node-exporter-lua.

PolynomialDivision, 2021-09-30:
Did dawn maybe crashed?

stintel, 2021-09-30:
> Did dawn maybe crashed?

Interesting, I thought I checked and that dawn was running, but it seems not. Restarting it makes the problems go away. Looks like it crashed at some point, on two different devices, but both ath79:

```
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: Starting Service...
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: UMDNS with port 1026
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: Dawn instance started!
Oct  1 00:36:43 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:20:57 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:07 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:17 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:26 10.50.30.221 kernel: [  102.818981] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77e21048
Oct  1 02:21:26 10.50.30.221 kernel: [  102.827586] epc = 555d0a9f in dawn[555ca000+12000]
Oct  1 02:21:26 10.50.30.221 kernel: [  102.832543] ra  = 555d227b in dawn[555ca000+12000]
Oct  1 02:21:28 10.50.30.221 prometheus-node-exporter-lua[19: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)
Oct  1 02:21:33 10.50.30.221 prometheus-node-exporter-lua[19: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)
```

```
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: Starting Service...
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: UMDNS with port 1026
Oct  1 00:36:40 10.50.30.221 procd: /etc/rc.d/S81dawn: Dawn instance started!
Oct  1 00:36:43 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:20:57 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:07 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:17 10.50.30.221 dawn[2546]: Connection failed (ERROR)
Oct  1 02:21:26 10.50.30.221 kernel: [  102.818981] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77e21048
Oct  1 02:21:26 10.50.30.221 kernel: [  102.827586] epc = 555d0a9f in dawn[555ca000+12000]
Oct  1 02:21:26 10.50.30.221 kernel: [  102.832543] ra  = 555d227b in dawn[555ca000+12000]
Oct  1 02:21:28 10.50.30.221 prometheus-node-exporter-lua[19: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)
Oct  1 02:21:33 10.50.30.221 prometheus-node-exporter-lua[19: /usr/lib/lua/prometheus-collectors/dawn.lua:11: bad argument #1 to 'pairs' (table expected, got nil)
```

stintel, 2021-09-30:
Just crashed again, so something is definitely wrong:

```

```

stintel, 2021-09-30:
FYI, running in gdbserver now on one of the APs, let's hope it crashes quickly so I can give you a backtrace.

stintel, 2021-10-01:
Not very useful :-(

```
Program received signal SIGSEGV, Segmentation fault.
0x5556325f in get_channel_utilization ()
(gdb) thread apply all bt

Thread 1 (Thread 10213.10213 "dawn"):
#0  0x5556325f in get_channel_utilization ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

stintel, 2021-10-01:
And another not very useful one :-(

```
0x5555ba9f in decide_function ()
(gdb) bt
#0  0x5555ba9f in decide_function ()
#1  0x5555d27b in hostapd_notify ()
Backtrace stopped: frame did not save the PC
```

PolynomialDivision, 2021-10-01:
I have similar issues on the EAP 225 Oudoor. This is the only AP there DAWN is having issues. I run that on severl ipq40xx there nothing is crashing. Any idea @svanheule ?

PolynomialDivision, 2021-10-01:
@stintel Did you try to run DAWN on another device?

stintel, 2021-10-01:
@PolynomialDivision D-Link DAP-2695 and TP-Link EAP245 v3.

XiaoliChan, 2021-10-03:
I also get the same problem.

PolynomialDivision, 2021-10-03:
> I also get the same problem.

Which device?

XiaoliChan, 2021-10-03:
> Which device?

Newifi-D1

[Portal](https://github.com/berlin-open-wireless-lab/DAWN/issues/149)