openwrt/packages #16779: netifd proto bonding: cannot create bond without IP/netmask

bobafetthotmail, 2021-09-30:
Hi, I've tried to create a bond without an IP/netmask to add it as a slave of a bridge. 
For example I want to bond port1 and port2 of my router and make a bridge between the bond, port3 and port4.
The bond is set as "active-backup" to provide redundancy.
This allows me to connect non-bonding-aware devices to an existing redundant/high availability network setup with two switches where all other nodes are bonding-aware.

But from what I have seen, this protocol (and the luci-proto) requires an Ip and netmask. I also tried deleting the ip and netmask from config file, restarted network service and no bond is created.

I know it is possible to make bonds without an IP/netmask and use them as slaves in a bridge in Linux, I've been doing this on OpenSUSE (also works on Debian) for a while. A IP-less bond can also be used as a target to create VLANs over the bond.

Is there something I can change in the protocol script to just make it ignore the need for IP/netmask and build the bond anwyay?

maintainer (of the bonding proto): @he-ma

he-ma, 2021-10-01:
@bobafetthotmail 
Hi, I'm not sure if this is a solution to your problem (as I have no equipment to test this), but could you try assigning IP 0.0.0.0 with netmask 0.0.0.0

bobafetthotmail, 2021-10-01:
Tried and it does not work (it creates a bond with IP 0.0.0.0 and if I add that to a bridge it does not work.

I have commented out all parts about setting an IP in the script in **/lib/netifd/proto/bonding.sh**, and the bond1 I added comes up but it's not configured correctly if I look from **cat /proc/net/bonding/bond1** (it shows no interfaces bound and it's not set as active-backup).

I've added **ip-full** package and then tried to do the commands manually like 

**ip link add bond1 type bond miimon 100 mode active-backup**

and now it creates the bond correctly and it shows the right configuration, but then I can't add any slave to it with 

**ip link set dev ethernet3@eth0 master bond1**

as it fails with 

**Cannot find device "ethernet3@eth0"**

but that's how the "ethernet ports" are called in this device's onboard switch.

I'm suspecting the fact that this device is using DSA for switch configuration (not the old swconfig) is screwing up this netifd protocol.

So, I created a single-port bridge on top of port 3 and another on top of port 4, and created a single unmanaged interface for each so these bridges are created on boot:

Then I went in /etc/rc.local and added 

**ip link add dev bond1 type bond miimon 100 mode active-backup
ip link set dev brport3 master bond1
ip link set dev brport4 master bond1**

now it's all good in theory but neither the single-port bridges nor the bond1 never comes up even if I do 
**ip link set dev brport3 up**
**ip link set dev brport4 up**
**ip link set dev bond1 up**

So there is something weird going on with the DSA switch infrastructure that breaks bonding or I'm a noob with it.

he-ma, 2021-10-04:
Thank you very much for your detailed explanations.
So, I assume you solved your problem by your way.
I'm also afraid I coudn't help you anymore as I have no experiences with DSA.

Kritzefitz, 2022-10-01:
@bobafetthotmail `ethernet3@eth0` is, as you guessed, a DSA port. It means there's a port `ethernet3` that part of the physical switch `eth0`. It's a bit confusing, because when you refer to that port you just call it `ethernet3`, without the `@eth0`. So your `ip link add` should be:

```
ip link set dev ethernet3 master bond1
```

On my device I can use manual `ip` command to configure bonding over bridge ports this way and it works fine. However, I think it would be better if I could configure this decoratively in `/etc/config/network` and didn't have to rely on imperative setup in `/etc/rc.local`.

justarandomgeek, 2023-12-21:
I realize i'm a bit late to this thread, but I wanted this also and found that commenting out
```
	# For static configuration we _MUST_ have an IP address
	#[ -z "$ipaddr" ] && {
	#	echo "$cfg" "No local IP address defined"
	#	proto_notify_error "$cfg" INVALID_LOCAL_ADDRESS
	#	proto_block_restart "$cfg"
	#	return
	#}

	#proto_add_ipv4_address "$ipaddr" "$netmask"
``` in /lib/netifd/proto/bonding.sh brings the bond up just fine, and then i can include it in a bridge and configure everything above that as normal - but in doing all this it strikes me, that perhaps a bond would be better treated as a "device" rather than an "interface"?

webknjaz, 2024-02-25:
Subscribing for updates. I'm also puzzled as to why it isn't treated as a device.

Dacesilian, 2024-03-27:
I'm also wondering, why bonding requires to set IP. I want to create bond (device), add it to a bridge and set IP to bridge, not bond.

trenchcoatjedi, 2024-10-08:
Agree with other commenters. Bonds should be treated as devices not interfaces. [The workaround above ](https://github.com/openwrt/packages/issues/16779#issuecomment-1865344679) allows the bond to be treated as a device. I can even add it to status pages by editing board.json
[image]

justarandomgeek, 2024-11-17:
I noticed when updating my router yesterday that someone seems to have integrated changes similar to those from my comment above into ` /lib/netifd/proto/bonding.sh` and this now seems to work correctly right out of the box! Thanks to whoever did that! :)

trenchcoatjedi, 2024-11-17:
Here's who to thank:

https://github.com/openwrt/packages/pull/23723#issuecomment-2027244518

> I noticed when updating my router yesterday that someone seems to have integrated changes similar to those from my comment above into ` /lib/netifd/proto/bonding.sh` and this now seems to work correctly right out of the box! Thanks to whoever did that! :)

regunakyle, 2024-12-07:
> I noticed when updating my router yesterday that someone seems to have integrated changes similar to those from my comment above into ` /lib/netifd/proto/bonding.sh` and this now seems to work correctly right out of the box! Thanks to whoever did that! :)

Hi, how did you set up your bond interface? 

My (failed) attempt:
I used the `luci-proto-bonding` package to set up LACP. This is my config:
```bash
root@OpenWrt:~# uci show | grep lacp
firewall.@zone[0].network='lan' 'lacp'
network.@device[0].ports='bond-lacp'
network.lacp=interface
network.lacp.proto='bonding'
network.lacp.slaves='eth2' 'eth3'
network.lacp.bonding_policy='802.3ad'
network.lacp.min_links='0'
network.lacp.ad_actor_sys_prio='65535'
network.lacp.ad_select='stable'
network.lacp.lacp_rate='slow'
network.lacp.xmit_hash_policy='layer2'
network.lacp.all_slaves_active='0'
network.lacp.link_monitoring='off'
network.lacp.ipaddr='127.0.0.1'
network.lacp.netmask='255.0.0.0'
network.@bridge-vlan[0].ports='bond-lacp:t'
network.@bridge-vlan[1].ports='bond-lacp:t'
```

I replaced `/lib/netifd/proto/bonding.sh` with the latest version in Github, then ran `uci delete network.lacp.ipaddr` and `uci delete network.lacp.netmask`. 
The bonding interface was not created after a reboot and I have to revert the changes. Not sure what went wrong in my case.

trenchcoatjedi, 2025-01-17:
Luci-proto-bonding is depreciated in 24.10

GermanAizek, 2025-01-27:
> > I noticed when updating my router yesterday that someone seems to have integrated changes similar to those from my comment above into ` /lib/netifd/proto/bonding.sh` and this now seems to work correctly right out of the box! Thanks to whoever did that! :)
> 
> Hi, could you show me how did you set up your bond interface?
> 
> My (failed) attempt: I used the `luci-proto-bonding` package to set up LACP. This is my config:
> 
> root@OpenWrt:~# uci show | grep lacp
> firewall.@zone[0].network='lan' 'lacp'
> network.@device[0].ports='bond-lacp'
> network.lacp=interface
> network.lacp.proto='bonding'
> network.lacp.slaves='eth2' 'eth3'
> network.lacp.bonding_policy='802.3ad'
> network.lacp.min_links='0'
> network.lacp.ad_actor_sys_prio='65535'
> network.lacp.ad_select='stable'
> network.lacp.lacp_rate='slow'
> network.lacp.xmit_hash_policy='layer2'
> network.lacp.all_slaves_active='0'
> network.lacp.link_monitoring='off'
> network.lacp.ipaddr='127.0.0.1'
> network.lacp.netmask='255.0.0.0'
> network.@bridge-vlan[0].ports='bond-lacp:t'
> network.@bridge-vlan[1].ports='bond-lacp:t'
> 
> I replaced `/lib/netifd/proto/bonding.sh` with the latest version in Github, then ran `uci delete network.lacp.ipaddr` and `uci delete network.lacp.netmask`. The bonding interface was not created after a reboot and I have to revert the changes. Not sure what went wrong in my case.
> 
> (The router is running 23.05.5)

Same trouble issue

GermanAizek, 2025-01-27:
```
root@checkpoint-l50:~# uci delete network.lacp.ipaddr
uci: Entry not found
root@checkpoint-l50:~# uci delete network.lacp.netmask
uci: Entry not found
```