openwrt/packages #16794: https-dns-proxy error using DoH/adblock (https_client.c:297)

freak357, 2021-10-03:
Hello, i just registered so please forgive me any mistakes reporting this bug.

I'm using OpenWrt 21.02.0 on a TP-Link Archer C6 2.0 with packages https-dns-proxy and adblock.
After 24h disconnect and autoredial via PPPoE i get the following errors in my system log:

daemon.info https-dns-proxy[]: [E] https_client.c:297 Connecting and sending request to resolver was successful, but no response was sent back

Resulting in DNS not working. Problem can be fixed stopping/restarting https-dns-proxy service.

stangri, 2021-10-03:
The `https-dns-proxy` should be restarted on WAN update, if you could capture the log and post it here it may be helpful in diagnosing the problem.

freak357, 2021-10-03:
Do you want me to copy and post the system log in the GUI and post it here or should i use the CLI? If CLI is needed please tell me the log file path/location.

freak357, 2021-10-03:
@stangri 
I hope this is enough for diagnosing. I removed the MAC adresses for privacy reasons and shortened the log:

Sun Oct  3 04:03:28 2021 daemon.info https-dns-proxy[4840]: [E] 1633233808.361058 https_client.c:297 01A0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:28 2021 daemon.info https-dns-proxy[4839]: [E] 1633233808.362927 https_client.c:297 01A0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:28 2021 daemon.info https-dns-proxy[4840]: [E] 1633233808.503560 https_client.c:297 EAE0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:28 2021 daemon.info https-dns-proxy[4839]: [E] 1633233808.504631 https_client.c:297 EAE0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:30 2021 daemon.info https-dns-proxy[4840]: [E] 1633233810.846022 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:30 2021 daemon.info https-dns-proxy[4839]: [E] 1633233810.847291 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:30 2021 daemon.info https-dns-proxy[4840]: [E] 1633233810.862642 https_client.c:297 01A0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:30 2021 daemon.info https-dns-proxy[4839]: [E] 1633233810.863783 https_client.c:297 01A0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:32 2021 daemon.info https-dns-proxy[4840]: [E] 1633233812.553654 https_client.c:297 C115: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:32 2021 daemon.info https-dns-proxy[4839]: [E] 1633233812.554883 https_client.c:297 C115: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4840]: [E] 1633233813.348430 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4839]: [E] 1633233813.350127 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4840]: [E] 1633233813.388999 https_client.c:297 B269: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4839]: [E] 1633233813.389954 https_client.c:297 B269: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4840]: [E] 1633233813.507606 https_client.c:297 EAE0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:33 2021 daemon.info https-dns-proxy[4839]: [E] 1633233813.509666 https_client.c:297 EAE0: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:35 2021 daemon.info https-dns-proxy[4840]: [E] 1633233815.850918 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:35 2021 daemon.info https-dns-proxy[4839]: [E] 1633233815.852031 https_client.c:297 CF64: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:35 2021 daemon.info https-dns-proxy[4840]: [E] 1633233815.866553 https_client.c:297 B269: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:03:35 2021 daemon.info https-dns-proxy[4839]: [E] 1633233815.868668 https_client.c:297 B269: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 04:04:24 2021 user.err adblock-4.1.3[6582]: dns backend restart with adblock blocklist failed
Sun Oct  3 04:04:25 2021 daemon.info dnsmasq[7291]: exiting on receipt of SIGTERM
Sun Oct  3 04:04:25 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct  3 04:04:25 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: Connected to system UBus
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: started, version 2.85 cachesize 150
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: DNS service limited to local subnets
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: UBus support enabled: connected to system bus
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq-dhcp[7730]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: using only locally-known addresses for domain test
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: using only locally-known addresses for domain onion
Sun Oct  3 04:04:29 2021 daemon.info dnsmasq[7730]: using only locally-known addresses for domain localhost
[... cut ...]
Sun Oct  3 07:24:07 2021 daemon.info dnsmasq-dhcp[7730]: DHCPOFFER(br-lan) 192.168.1.183 
Sun Oct  3 07:24:07 2021 daemon.info dnsmasq-dhcp[7730]: DHCPREQUEST(br-lan) 192.168.1.183 
Sun Oct  3 07:24:07 2021 daemon.info dnsmasq-dhcp[7730]: DHCPACK(br-lan) 192.168.1.183
Sun Oct  3 07:27:53 2021 daemon.info https-dns-proxy[4839]: [E] 1633246073.544876 https_client.c:297 2D6F: Connecting and sending request to resolver was successful, but no response was sent back
Sun Oct  3 07:36:58 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED

stangri, 2021-10-05:
I don't see any PPPoE/WAN up/down/update in the log, but I do see that STA AP was used. What's your primary internet connection?

What are the resolvers you're using?

freak357, 2021-10-05:
My internet connection is VDSL via PPPoE. I tried the following resolvers: libredns, cloudflare, quad9, nextdns.

I just restarted the WAN interface manually resulting in instant loss of DNS and system log getting spammed with hundreds of lines of: 

daemon.info https-dns-proxy[]: [E] https_client.c:297 Connecting and sending request to resolver was successful, but no response was sent back

Stopping/restarting https-dns-proxy service fixed it.

Here is the log:

Tue Oct  5 05:56:28 2021 daemon.info pppd[12670]: Terminating on signal 15
Tue Oct  5 05:56:28 2021 daemon.info pppd[12670]: Connect time minutes.
Tue Oct  5 05:56:28 2021 daemon.info pppd[12670]: Sent 9818049 bytes, received 51580723 bytes.
Tue Oct  5 05:56:28 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Tue Oct  5 05:56:28 2021 daemon.notice pppd[12670]: Connection terminated.
Tue Oct  5 05:56:28 2021 daemon.info pppd[12670]: Sent PADT
Tue Oct  5 05:56:28 2021 daemon.info pppd[12670]: Exit.
Tue Oct  5 05:56:28 2021 daemon.notice netifd: Interface 'wan' is now down
Tue Oct  5 05:56:28 2021 daemon.notice netifd: Interface 'wan' is setting up now
Tue Oct  5 05:56:28 2021 daemon.err insmod: module is already loaded - slhc
Tue Oct  5 05:56:28 2021 daemon.err insmod: module is already loaded - ppp_generic
Tue Oct  5 05:56:28 2021 daemon.err insmod: module is already loaded - pppox
Tue Oct  5 05:56:28 2021 daemon.err insmod: module is already loaded - pppoe
Tue Oct  5 05:56:29 2021 daemon.info pppd[14162]: Plugin rp-pppoe.so loaded.
Tue Oct  5 05:56:29 2021 daemon.info pppd[14162]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Tue Oct  5 05:56:29 2021 daemon.notice pppd[14162]: pppd 2.4.8 started by root, uid 0
Tue Oct  5 05:56:29 2021 daemon.err pppd[14162]: PADS: System-Error: No resources
Tue Oct  5 05:56:34 2021 user.info adblock-4.1.3[14322]: adblock instance started ::: action: start, priority: 0, pid: 14322
Tue Oct  5 05:56:39 2021 daemon.warn pppd[14162]: Timeout waiting for PADS packets
Tue Oct  5 05:56:39 2021 daemon.err pppd[14162]: Unable to complete PPPoE Discovery
Tue Oct  5 05:56:39 2021 daemon.info pppd[14162]: Exit.
Tue Oct  5 05:56:39 2021 daemon.notice netifd: Interface 'wan' is now down
Tue Oct  5 05:56:39 2021 daemon.notice netifd: Interface 'wan' is setting up now
Tue Oct  5 05:56:39 2021 daemon.err insmod: module is already loaded - slhc
Tue Oct  5 05:56:39 2021 daemon.err insmod: module is already loaded - ppp_generic
Tue Oct  5 05:56:39 2021 daemon.err insmod: module is already loaded - pppox
Tue Oct  5 05:56:39 2021 daemon.err insmod: module is already loaded - pppoe
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: Plugin rp-pppoe.so loaded.
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: pppd 2.4.8 started by root, uid 0
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: PPP session is 1438
Tue Oct  5 05:56:39 2021 daemon.warn pppd[14492]: Connected to via interface eth0.2
Tue Oct  5 05:56:39 2021 kern.info kernel: [266215.061752] pppoe-wan: renamed from ppp0
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: Renamed interface ppp0 to pppoe-wan
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: Using interface pppoe-wan
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: Connect: pppoe-wan <--> eth0.2
Tue Oct  5 05:56:39 2021 daemon.info pppd[14492]: Remote message: Accepted.
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: PAP authentication succeeded
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: peer from calling number authorized
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: local  IP address 
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: remote IP address 
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: primary   DNS address 
Tue Oct  5 05:56:39 2021 daemon.notice pppd[14492]: secondary DNS address 
Tue Oct  5 05:56:39 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
Tue Oct  5 05:56:39 2021 daemon.notice netifd: Interface 'wan' is now up
Tue Oct  5 05:56:39 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Tue Oct  5 05:56:40 2021 user.notice https-dns-proxy: Reloading https-dns-proxy due to ifup of wan
Tue Oct  5 05:56:40 2021 user.info adblock-4.1.3[14322]: download of 'adaway' failed, url: https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt, rule: /^127\.0\.0\.1[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories: -, rc: 4, log: Downloading 'https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt' Failed to send request: Operation not permitted
Tue Oct  5 05:56:40 2021 user.info adblock-4.1.3[14322]: download of 'adguard' failed, url: https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt, rule: BEGIN{FS="[/|^|\r]"}/^\|\|([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+[\/\^\r]+$/{print tolower($3)}, categories: -, rc: 4, log: Downloading 'https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt' Failed to send request: Operation not permitted
Tue Oct  5 05:56:40 2021 user.info adblock-4.1.3[14322]: download of 'disconnect' failed, url: https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt, rule: /^([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($1)}, categories: -, rc: 4, log: Downloading 'https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt' Failed to send req [... cut ...]
Tue Oct  5 05:56:48 2021 daemon.info https-dns-proxy[4839]: [E] 1633413408.163472 https_client.c:297 30D7: Connecting and sending request to resolver was successful, but no response was sent back
Tue Oct  5 05:56:48 2021 daemon.info https-dns-proxy[4839]: [E] 1633413408.164970 https_client.c:297 1112: Connecting and sending request to resolver was successful, but no response was sent back
Tue Oct  5 05:56:48 2021 daemon.info https-dns-proxy[4839]: [E] 1633413408.166455 https_client.c:297 1112: Connecting and sending request to resolver was successful, but no response was sent back

closed 2021-10-12 by freak357