openwrt/packages #16803: TVHeadend failing due to gettext dependency

rudihorn, 2021-10-05:
Maintainer: Marius Dinu  (could not find github user), 
Environment: ARM, MT7622, master and v21.02.0

Description:

Compiling the `tvheadend` package fails because of missing gettext binaries. As far as I can tell the issue is that it not only requires the `gettext` package, but also the `gettext-full` package to compile. Neither of these are listed as dependencies.

I didn't find an option in the openwrt `make menuconfig` to install the `gettext-full` package either, and had to do it manually. I'm not entirely sure if it should be somewhere in the menu, or if the correct solution would be to simply add the `gettext-full` package as a dependency to the `tvheadend` package.

The `gettext-full` package is maintained by @jow-, but is in the main openwrt repository.

rudihorn, 2021-10-19:
Currently doing the following:

```
git clone https://github.com/openwrt/openwrt.git
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a
```

Then either:
* `make menuconfig`. Select Target System -> MediaTek Ralink ARM, Multimedia -> tvheadend [*].
* use attached [config](https://github.com/openwrt/packages/files/7376244/openwrt.txt)

```
make -j $(nproc) download world
```

=> Compilation fails. Run with debugging:

```
make -j1 V=sc
```

Compilation fails with output:
```
Checking support/features
  checking for cc execinfo.h ...                    fail
  checking for cc -mmmx ...                         fail
  checking for cc -msse2 ...                        fail
  checking for cc -Wunused-result ...               ok
  checking for cc strlcat ...                       ok
  checking for cc strlcpy ...                       ok
  checking for cc fdatasync ...                     ok
  checking for cc getloadavg ...                    ok
  checking for cc atomic32 ...                      ok
  checking for cc atomic64 ...                      ok
  checking for cc atomic_time_t ...                 ok
  checking for cc bitops64 ...                      ok
  checking for cc lockowner ...                     fail
  checking for cc qsort_r ...                       fail
  checking for cc stime ...                         ok
  checking for cc gmtoff ...                        ok
  checking for cc recvmmsg ...                      ok
  checking for cc sendmmsg ...                      ok
  checking for cc libiconv ...                      ok
    ^ using external iconv library
  checking for cc libdvben50221 ...                 fail
  checking for cc ifnames ...                       ok
  checking for py module gzip ...                   ok
  checking for /home/administrator/github/openwrt/staging_dir/host/bin/pkg-config ...ok
  checking for xgettext ...                         fail
ERROR: no gettext binaries found
```

Compile `gettext-full` first with `make package/gettext-full/compile`. Then compile with `make -j $(nproc)`. 

=> Compilation succeeds

M95D, 2021-10-21:
I used your config and latest git (yesterday). It works. Image built, packages built, everything.
Are you sure it's not caused by missing gettext on your host?

libhdhomerun was patched: https://github.com/openwrt/packages/pull/16932