openwrt/packages #16817: mwan3: LTE interface always disabled after reboot

raidenii, 2021-10-06:
Maintainer: @feckert 
Environment: 21.02, x86-64, mwan3 2.10.12-1

Description:

I am using USB LTE modem as the 2nd uplink for my router. I discovered that upon reboot, the LTE wwan interface is always marked as "disabled", and detail status shows things like:

```
interface wwan is online 00h:00m:00s, uptime 00h:00m:19s and tracking is paused
```

However, if rebooting the mwan3 service, everything works as expected.

In syslog:

```
Tue Oct  5 21:59:19 2021 user.warn mwan3-hotplug[5386]: failed to add wan.ip.address/24 dev eth1 proto static scope link metric 10 to table 1
Tue Oct  5 21:59:19 2021 user.info mwan3track[2986]: Detect ifup event on interface wan ()
Tue Oct  5 21:59:19 2021 user.notice mwan3track[2986]: Interface wan (eth1) is online
Tue Oct  5 21:59:27 2021 user.notice mwan3-hotplug[6787]: Execute ifup event on interface wwan (wwan0)
Tue Oct  5 21:59:27 2021 user.info mwan3track[2987]: Detect ifup event on interface wwan ()
Tue Oct  5 21:59:33 2021 user.notice mwan3-hotplug[7214]: mwan3 hotplug on wwan_4 not called because interface disabled
```

yet in route table 1, wan default route (the working/cable connecton) exists.

```
root@openwrt:~# ip r s t 1
default via wan.ip.gateway dev eth1 proto static src wan.ip.address metric 10
...
```

and in route table 2 wwan default route exists as well:
```
root@openwrt:~# ip r s t 2
default via wwan.ip.gateway dev wwan0 proto static src wwan.ip.address metric 20
...
```

/etc/config/network:
```
...
config interface 'wan'
        option proto 'dhcp'
        option peerdns '0'
        list dns '127.0.0.1'
        option hostname 'router'
        option device 'eth1'
        option metric '10'

config interface 'wwan'
        option proto 'ncm'
        option delegate '0'
        option apn 'broadband'
        option metric '20'
        option peerdns '0'
        option pdptype 'IP'
        option ipv6 '0'
        option device '/dev/ttyUSB0'
        list dns '127.0.0.1'
...
```

/etc/config/mwan3:
```
config rule 'ipv4_failover'
        option family 'ipv4'
        option dest_ip '0.0.0.0/0'
        option proto 'all'
        option sticky '0'
        option use_policy 'failover'
        option src_ip '0.0.0.0/0'

config globals 'globals'
        option mmx_mask '0x3F00'

config interface 'wan'
        option enabled '1'
        option family 'ipv4'
        option initial_state 'online'
        list track_ip '8.8.8.8'
        list track_ip '1.1.1.1'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '2'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'

config interface 'wwan'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv4'
        list track_ip '8.8.4.4'
        list track_ip '1.0.0.1'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'

config member 'fios'
        option interface 'wan'
        option metric '1'

config member 'lte'
        option interface 'wwan'
        option metric '2'

config policy 'failover'
        list use_member 'fios'
        list use_member 'lte'
        option last_resort 'unreachable'

config policy 'fios_only'
        list use_member 'fios'
        option last_resort 'unreachable'
```

raidenii, 2021-10-06:
One thing I noticed that is for wwan interface, ping always start to get replies after 4th seq:

```
root@openwrt:~# ping -4 -I wwan0 -c20 www.google.com
PING www.google.com (172.217.168.196): 56 data bytes
64 bytes from 172.217.168.196: seq=4 ttl=112 time=2166.851 ms
64 bytes from 172.217.168.196: seq=5 ttl=112 time=1176.633 ms
64 bytes from 172.217.168.196: seq=6 ttl=112 time=176.547 ms
64 bytes from 172.217.168.196: seq=7 ttl=112 time=136.422 ms
64 bytes from 172.217.168.196: seq=8 ttl=112 time=135.525 ms
64 bytes from 172.217.168.196: seq=9 ttl=112 time=135.516 ms
64 bytes from 172.217.168.196: seq=10 ttl=112 time=135.266 ms
64 bytes from 172.217.168.196: seq=11 ttl=112 time=175.213 ms
64 bytes from 172.217.168.196: seq=12 ttl=112 time=135.075 ms
64 bytes from 172.217.168.196: seq=13 ttl=112 time=134.945 ms
64 bytes from 172.217.168.196: seq=14 ttl=112 time=134.858 ms
64 bytes from 172.217.168.196: seq=15 ttl=112 time=165.155 ms
64 bytes from 172.217.168.196: seq=16 ttl=112 time=134.648 ms
64 bytes from 172.217.168.196: seq=17 ttl=112 time=134.571 ms
64 bytes from 172.217.168.196: seq=18 ttl=112 time=164.534 ms
64 bytes from 172.217.168.196: seq=19 ttl=112 time=134.439 ms

--- www.google.com ping statistics ---
20 packets transmitted, 16 packets received, 20% packet loss
round-trip min/avg/max = 134.439/336.012/2166.851 ms
```

Not sure if that is something matters.

feckert, 2021-10-06:
I think this is a problem with your mobile connection!
Could you please change the ping timeout, so the pings do have more time to reach the tracking server?

I think you have to play with your tracking [configuration](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration) options to find the correct settings.
```
 option reliability '1'
 option count '1'
 option size '56'
 option max_ttl '60'
 option timeout '4'
 option interval '10'
```

A `interval` of 10 seconds and `count` of 1 is very aggressive!

raidenii, 2021-10-06:
I tested changing `interval` to 30 and timeout to `10` but the interface is still disabed upon router reboot. Also, restarting mwan3 service afterwards and everything works perfectly fine. Sorry but I failed to see this is a connection issue.

I wonder why the wwan interface was not set to `online` after mwan3track detected the `ifup` event, unlike the wan counterpart. is it because the IP address was actually set on `wwan_4` instead of `wwan`?

The mwan settings page also says "Warning: Interface wwan has no default route in the main routing table", yet `ip r` does show it:

```
root@openwrt:~# ip r
default via wan.gateway dev eth1 proto static src wan.ip metric 10 
default via wwan.gateway dev wwan0 proto static src wwan.ip metric 20
...
```

raidenii, 2021-10-06:
After some testing I found this might be responsible:
https://github.com/openwrt/packages/blob/9c4d79519ced3d0ccf6917e24b32b747e1672c62/net/mwan3/files/usr/sbin/mwan3track#L166
specifically, for NCM devices, it takes some time for it to create the wwan_4 interface after boot, which it might not get IP yet.

changing that line to
```
if [ -z "DEVICE" ]; then
```

and after reboot, LTE interface is marked as online immediately. The log says:

```
Wed Oct  6 19:18:49 2021 user.info mwan3track[3226]: Detect ifup event on interface wwan ()
Wed Oct  6 19:18:49 2021 user.warn mwan3track[3226]: no src ipv4 address found for interface 'wwan_4' dev ''
Wed Oct  6 19:18:49 2021 user.notice mwan3track[3226]: Interface wwan () is online
```

which might be the reason that `network_is_up` thinks the interface wasn't up immediately but shortly after.

For a proper fix, maybe add a parameter to allow mwan3 retry on the interface for some value of seconds, and if still fail, mark it as "offline" rather than "disabled"?

feckert, 2021-10-13:
I have now set this up.
As I see it, we also get an ifup event from netifd for the `<iface>_x` interface.
`Wed Oct 13 13:45:27 2021 user.notice test/iface[673]: ====== INTERFACE=wwan_4 ACTION=ifup ======`
But this will not be proceed further, as the `<iface>_x` does not exist in the mwan3 configuration.
https://github.com/openwrt/packages/blob/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L42-L46

We could therefore send an event to the interface's mwan3track that the interface is now setup correctly and we wouldn't need to poll it.
https://github.com/openwrt/packages/blob/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/net/mwan3/files/usr/sbin/mwan3track#L35-L41

This would then do another firstconnect try in this line.
https://github.com/openwrt/packages/blob/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/net/mwan3/files/usr/sbin/mwan3track#L398-L402

pgaufillet, 2023-01-01:
Still seen on 22.03.2 r19803-9a599fee93 on an openvpn tunnel:
The issue is that the state of the tunnel _tun0_ (which is unmanaged) is _UNKNOWN_ and not _UP_:

```
~#  ip ad sh
1: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 500
```

ValerioBob, 2025-02-20:
Still seen in  23.05.5 (r24106-10cc5fcd00). 

The simplest solution:

in /etc/rc.local:

```
/etc/init.d/mwan3 restart
exit 0
```

stefanrybacki, 2025-06-04:
> Still seen in 23.05.5 (r24106-10cc5fcd00).
> 
> The simplest solution:
> 
> in /etc/rc.local:
> 
> ```
> /etc/init.d/mwan3 restart
> exit 0
> ```

`/etc/init.d/mwan3 restart` will cause my openwrt machine to lose all network connections even local (so I cannot reach it locally anymore)