openwrt/packages #16818: Possible upstream switch to `firewall4` aka `nftables` instead of `iptables`

aparcar, 2021-10-06:
Hi all, especially @openwrt/packages-write,

for the next OpenWrt release `firewall4` is [considered](https://github.com/openwrt/openwrt/pull/4642) as a replacement of the current `iptables` based `firewall` package. While the configuration stays within `/etc/config/firewall`, packages using `iptables` directly may see trouble.

This is a heads up for everyone maintaining such packages but also please post packages here that would be affected so a smother migration is possible.

aparcar, 2021-10-06:
It's **possible**, an idea to be discussed. There is no definite decision yet, however ideally maintainer start looking at `firewall4` to have an idea what could change. Long term there might be a `firewall5` package using eBPF things are moving :)

ldir-EDB0, 2021-10-06:
My concern is over 'ipset' equivalent functionality support.  dnsmasq v2.87 (not yet released) has immature support.   adblock & banip rely heavily on ipsets and will need adjusting.
miniupnpd has nftables support, the integration into fw4 will need looking it.

feckert, 2021-10-06:
Also acme does some iptables command 
https://github.com/openwrt/packages/blob/6c73457c09f838279b240bef59730cbff60ae799/net/acme/files/run.sh#L127-L133

tohojo, 2021-10-06:
On 6 October 2021 12:31:12 CEST, Florian Eckert ***@***.***> wrote:
>Also acme does some iptables command 
>https://github.com/openwrt/packages/blob/6c73457c09f838279b240bef59730cbff60ae799/net/acme/files/run.sh#L127-L133

Ah yes, so it does - totally forgot about that. Thanks for the reminder!

dibdot, 2021-10-06:
adblock is purely DNS and doesn't use any direct iptables calls. Maybe simple-adblock is affected here - I don't know.

stangri, 2021-10-06:
> @stangri please track those packages here and not at luci.git
> 
> > https://github.com/openwrt/packages/tree/master/net/vpn-policy-routing
> > https://github.com/openwrt/packages/tree/master/net/vpnbypass
> > I've also started working on a fork of the former called pbr and it's also iptables-dependent.

Sorry, didn't see notification about this issue until after this quote.

If there's no compatibility binary to allow iptables calls while using nftables, both `vpn-policy-routing` and `vpnbypass` are in trouble. They are heavily iptables dependent and it may take me a while to find the time to figure out the switch to nftables.

Both `https-dns-proxy` and `simple-adblock` are good to go.

brada4, 2021-10-10:
iptables-nft can use ipset, esp for places where only access list is ipset. nft has own sets, but traditional ipset command does not bridge the gap

stintel, 2021-11-08:
>  miniupnpd has nftables support, the integration into fw4 will need looking it.

As ex maintainer, would you mind having a look at #17094?

stintel, 2021-11-08:
>  miniupnpd has nftables support, the integration into fw4 will need looking it.

@ldir-EDB0 As ex maintainer, would you mind having a look at #17094?

pprindeville, 2021-11-13:
> I would expand the list a little further:
> 
> ```
> feeds/packages $ grep -E "(ip6?tables(-save|-restore)?( |$|\"|'|\))|lib/iptables|\+iptables)" -R */ | cut -d/ -f2 | sort -u
> acme
> adblock
> apfree-wifidog
> banip
> bcp38
> collectd
> coova-chilli
> dockerd
> etherwake-nfqueue
> fail2ban
> frr
> fwknop
> gnunet
> https-dns-proxy
> jool
> keepalived
> libreswan
> miniupnpd
> mwan3
> podman
> pppossh
> redsocks
> shadowsocks-libev
> shorewall
> shorewall6
> shorewall6-lite
> shorewall-lite
> simple-adblock
> sqm-scripts
> strongswan
> trafficshaper
> uacme
> v2raya
> vpnbypass
> vpnc-scripts
> vpn-policy-routing
> wifidog
> xtables-addons
> ```
> 
> Except from shorewall and xtables-addons, which are clearly not compatible with firewall4/nftable, the rest is still open. Anything that depends on iptables or calls iptables(-save/-restore) needs some testing, specially if iptables-nft is in use.
> 
> Maybe we could edit this issue description mentioning maintainers after the package to ping them all?
> 
> I doubt that iptables-nft will be installed by default and I think fw3 will still be available as a fallback. It would be interesting to see how we deal with dependencies when both standard iptables and iptables-nft are available and the user could either use fw3 or fw4. Some packages might require iptables/nftables flavors.

Not aware of support for `nftables` in `strongswan`.  @Thermi?

Also, locally we use `ipset` for blocking traffic from non-essential devices after hours so they don't interfere with remote backups... as well as `xtgeoip` from `xtables-addons` to block traffic from hostile countries.

pprindeville, 2021-11-13:
> aside from the command-line iptables/ip6tables, do we also need to keep an eye open for all packages which depend on kmod-ipt.*?

This list can be pared down to ignore `kmod-ipt-*` and `iptables-mod-*` entries:

```
$ grep -e 'Package:' -e 'Depends:.*kmod-ipt' tmp/.packageinfo | grep -B1 'Depends:'
Package: kmod-ipt-conntrack
Depends:  +kmod-ipt-core +kmod-nf-conntrack
Package: kmod-ipt-conntrack-extra
Depends:  +kmod-ipt-core +kmod-ipt-conntrack
Package: kmod-ipt-conntrack-label
Depends:  +kmod-ipt-core +kmod-ipt-conntrack
Package: kmod-ipt-filter
Depends:  +kmod-ipt-core +kmod-lib-textsearch +kmod-ipt-conntrack
Package: kmod-ipt-offload
Depends:  +kmod-ipt-core +kmod-nf-flow
Package: kmod-ipt-ipopt
Depends:  +kmod-ipt-core 
Package: kmod-ipt-ipsec
Depends:  +kmod-ipt-core 
Package: kmod-ipt-ipset
Depends:  +kmod-ipt-core +kmod-nfnetlink
Package: kmod-nf-ipvs
Depends: @IPV6 +kmod-lib-crc32c +kmod-ipt-conntrack +kmod-nf-conntrack +kmod-ipt-core +kmod-ipt-conntrack
--
Package: kmod-ipt-nat
Depends:  +kmod-ipt-core +kmod-nf-nat
Package: kmod-ipt-raw
Depends:  +kmod-ipt-core 
Package: kmod-ipt-raw6
Depends: @IPV6 +kmod-ipt-core +kmod-ip6tables
Package: kmod-ipt-nat6
Depends: @IPV6 +kmod-ipt-core +kmod-nf-nat6 +kmod-ipt-core +kmod-ipt-conntrack +kmod-ipt-core +kmod-ipt-nat +kmod-ipt-core +kmod-ip6tables
Package: kmod-ipt-nat-extra
Depends:  +kmod-ipt-core +kmod-ipt-nat
--
Package: kmod-nf-nathelper-extra
Depends: +kmod-nf-nat +kmod-lib-textsearch +kmod-ipt-raw +kmod-asn1-decoder
Package: kmod-ipt-ulog
Depends:  +kmod-ipt-core 
Package: kmod-ipt-nflog
Depends:  +kmod-ipt-core +kmod-nfnetlink-log
Package: kmod-ipt-nfqueue
Depends:  +kmod-ipt-core +kmod-nfnetlink-queue
Package: kmod-ipt-debug
Depends:  +kmod-ipt-core +kmod-ipt-raw +IPV6:kmod-ipt-raw6
Package: kmod-ipt-led
Depends:  +kmod-ipt-core 
Package: kmod-ipt-tproxy
Depends:  +kmod-ipt-conntrack +IPV6:kmod-nf-conntrack6 +IPV6:kmod-ip6tables +kmod-ipt-core 
Package: kmod-ipt-tee
Depends: +kmod-ipt-conntrack +kmod-ipt-core 
Package: kmod-ipt-u32
Depends:  +kmod-ipt-core 
Package: kmod-ipt-checksum
Depends:  +kmod-ipt-core 
Package: kmod-ipt-iprange
Depends:  +kmod-ipt-core 
Package: kmod-ipt-cluster
Depends:  +kmod-ipt-core +kmod-nf-conntrack
Package: kmod-ipt-clusterip
Depends:  +kmod-ipt-core +kmod-nf-conntrack
Package: kmod-ipt-extra
Depends:  +kmod-ipt-core 
Package: kmod-ipt-physdev
Depends:  +kmod-ipt-core +kmod-br-netfilter
Package: kmod-ip6tables
Depends: @IPV6 +kmod-nf-reject6 +kmod-nf-ipt6 +kmod-ipt-core
--
Package: kmod-arptables
[... 225 lines cut ...]
Package: kmod-ipt-psd
Depends: +kmod-ipt-core 
Package: iptables-mod-quota2
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread iptables +kmod-ipt-quota2
Package: kmod-ipt-quota2
Depends: +kmod-ipt-core 
Package: iptables-mod-sysrq
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread iptables +kmod-ipt-sysrq
Package: kmod-ipt-sysrq
Depends: +kmod-ipt-core +kmod-ipt-compat-xtables +kmod-crypto-hash
Package: iptables-mod-tarpit
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread iptables +kmod-ipt-tarpit
Package: kmod-ipt-tarpit
Depends: +kmod-ipt-core +kmod-ipt-compat-xtables
--
Package: bmx6
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread +kmod-ip6-tunnel +kmod-iptunnel6 +kmod-tun
--
Package: bmx7-tun
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread bmx7 +kmod-ip6-tunnel +kmod-iptunnel6 +kmod-tun
--
Package: kmod-ipt-rtpengine
Depends: +PACKAGE_kmod-ipt-rtpengine:kmod-crypto-hash +PACKAGE_kmod-ipt-rtpengine:kmod-ipt-core
$ 
```

Thermi, 2021-11-13:
Hi, none natively, just via translation layer.

Thermi, 2021-11-14:
Also, as far as I remember some plugins (connmark, forecast) link libiptc and libip4tc, so that whould need to be changed for proper nftables support.

oskarirauta, 2021-11-25:
I have added cni-plugins-nft for podman, so wouldn't that then check podman out?

aparcar, 2021-12-29:
@oskarirauta added, thank you.

aparcar, 2022-01-07:
@openwrt/packages-write heads up, firewall4 gets into a fine state (thanks to @stintel and @jow- )and I'm planing to make it the default within the next week. For all scripts that require _legacy_ `iptables` please test your apps with `iptables-nft` and report back!

AQPSal, 2022-01-08:
Getting this message in the latest snapshot - noticed that the router had connectivity, clients could connect to internet using IPV6 but not IPV4 - seemed a NAT/masquerade issue. No rules appeared in the firewall status. Checked the firewall - it did not seem to be running. Upon trying to start it, got the messages below.

root@OpenWrt:~# fw4 start
Section @forwarding[4] option 'src' is mandatory but not set
Reference error: left-hand side expression is not an array or object
In [anonymous function](), file /usr/share/ucode/fw4.uc, line 1892, byte 21:
  called from function [arrow function] (/usr/share/ucode/fw4.uc:540:77)
  called from function foreach ([C])
  called from function [anonymous function] (/usr/share/ucode/fw4.uc:540:78)
  called from function render_ruleset (/usr/share/firewall4/main.uc:100:24)
  called from anonymous function (/usr/share/firewall4/main.uc:167:28)

 `            let f1 = fwd.src.zone ? fwd.src.zone.family : 0;`
  Near here -------------------^

(Raspberry Pi 4, a few VLANs and zones)

stintel, 2022-01-08:
> root@OpenWrt:~# fw4 start Section @forwarding[4] option 'src' is mandatory but not set Reference error: left-hand side expression is not an array or object In anonymous function, file [...]

Please attach your /etc/config/firewall/

AQPSal, 2022-01-08:
> > root@OpenWrt:~# fw4 start Section @forwarding[4] option 'src' is mandatory but not set Reference error: left-hand side expression is not an array or object In anonymous function, file /usr/share/ucode/fw4.uc, line 1892, byte 21: called from function [arrow function] (/usr/share/ucode/fw4.uc:540:77) called from function foreach ([C]) called from function [anonymous function] (/usr/share/ucode/fw4.uc:540:78) called from function render_ruleset (/usr/share/firewall4/main.uc10024) called from anonymous function (/usr/share/firewall4/main.uc:167:28)
> > ` let f1 = fwd.src.zone ? fwd.src.zone.family : 0;` Near here -------------------^
> 
> Please attach your /etc/config/firewall/config defaults

Please see below. I entered all the rules using LUCI - I did not use the command line nor edited the file. As I was sanitizing it I noticed four forwarding rules with only dest or src. Perhaps they are what is causing the issue.

        option output 'ACCEPT'
        option forward 'REJECT'
        option synflood_protect '1'
        option drop_invalid '1'
        option input 'REJECT'

config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        list network 'lan'
        list network 'WG'
        option log '1'
        option forward 'REJECT'

config zone
        option name 'wan'
        option output 'ACCEPT'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'
        option input 'DROP'
        option forward 'DROP'

config forwarding
        option src 'lan'
        option dest 'wan'

config rule
        option name 'Allow-DHCP-Renew'
        option src 'wan'
        option proto 'udp'
        option dest_port '68'
        option target 'ACCEPT'
        option family 'ipv4'

config rule
        option name 'Allow-Ping'
        option src 'wan'
        option proto 'icmp'
        option icmp_type 'echo-request'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-IGMP'
        option src 'wan'
        option proto 'igmp'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-DHCPv6'
        option src 'wan'
        option proto 'udp'
        option src_ip 'fc00::/6'
        option dest_ip 'fc00::/6'
        option dest_port '546'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
[... 300 lines cut ...]
        option target 'ACCEPT'
        list proto 'udp'

config rule
        list proto 'icmp'
        option target 'ACCEPT'
        option src 'IOT'
        option name 'Allow-ICMP-vlan2'
        option family 'ipv4'

config rule
        list proto 'icmp'
        option target 'ACCEPT'
        option name 'Allow-ICMP-vlan7'
        option src 'Guest'
        option family 'ipv4'
        list icmp_type 'echo-request'

config rule
        list proto 'icmp'
        option src 'UIOT'
        option target 'ACCEPT'
        option name 'Allow-ICMP-vlan8'
        option family 'ipv4'
        list icmp_type 'echo-request'

AQPSal, 2022-01-08:
> 

I manually deleted the odd-looking config forwarding rules - I have no idea how they got there. Firewall is working now.

However:
The firewall status screen in LUCI is blank. 
Firewall LUCI screen gives me a "TypeError Cannot convert undefined or null to object" on the zones screen upon trying to make any change.
The one odd thing is that I cannot ping FE80:: addresses from the router but the router replies if a ping is sent to it to its LL address.

KA2107, 2022-01-11:
I got the following error when I tried to start firewall4 with "option force_dns 1" inside /etc/config/https-dns-proxy :

```
# fw4 start
ubus rule (ubus:https-dns-proxy[instance1] rule 1) option 'src' specifies invalid value 'lan'
Reference error: left-hand side expression is not an array or object
In infer_family(), file /usr/share/ucode/fw4.uc, line 270, byte 20:
  called from function [anonymous function] (/usr/share/ucode/fw4.uc:2158:5)
  called from function [arrow function] (/usr/share/ucode/fw4.uc:548:85)
  called from function map ([C])
  called from function [anonymous function] (/usr/share/ucode/fw4.uc:548:86)
  called from function render_ruleset (/usr/share/firewall4/main.uc:100:24)
  called from anonymous function (/usr/share/firewall4/main.uc:167:28)

 `            if (!obj || obj.family == 0 || obj.family == res)`
  Near here ------------------^

```

stangri, 2022-01-11:
> I got the following error when I tried to start firewall4 with "option force_dns 1" inside /etc/config/https-dns-proxy :
> 
> ```
> # fw4 start
> ubus rule (ubus:https-dns-proxy[instance1] rule 1) option 'src' specifies invalid value 'lan'
> ```

@KA2107 Thank you for testimg!

@aparcar @stintel here's the PROCD data used in the init script:

```sh
		procd_open_data
		json_add_array firewall
		for p in $forceDNSPorts; do
			if netstat -tuln | grep 'LISTEN' | grep ":${p}" >/dev/null 2>&1 || [ "$p" = '53' ]; then
				json_add_object ''
				json_add_string type redirect
				json_add_string target DNAT
				json_add_string src lan
				json_add_string proto 'tcp udp'
				json_add_string src_dport "$p"
				json_add_string dest_port "$p"
				json_add_boolean reflection 0
				json_close_object
			else
				json_add_object ''
				json_add_string type rule
				json_add_string src lan
				json_add_string dest '*'
				json_add_string proto 'tcp udp'
				json_add_string dest_port "$p"
				json_add_string target REJECT
				json_close_object
			fi
		done
		json_close_array
		procd_close_data
```

Any idea why `lan` is invalid value for `src` in `fw4`?

KA2107, 2022-01-11:
@stangri @aparcar @stintel Looks the issue may be on my end. I had renamed the "lan" zone to "lan_1" zone in my /etc/config/firewall . I had been running with firewall3 (iptables) with "lan_1" zone for several months without any issue. 

However when I tried firewall4 (nftables), /etc/init.d/firewall script was not getting loaded at startup and upon investigating I found this issue with https-dns-proxy. Not sure if I can call this a bug or if this is just an issue with my firewall config.

aparcar, 2022-01-19:
As discussed at yesterdays meeting I merged the changes. The next release will use `firewall4` as default and all packages incompatible (e.g. using `ipset`) should add a negative dependency. Our considerations are that the default (WiFI home router) setup works fine wir `firewall4` and special cases can always replace `firewall4` with `firewall3`, which should work at least until the upcoming 5.15 Kernels.

stangri, 2022-01-19:
Paul, could you please clarify/elaborate on the `ipset` incompatibility with `firewall4`?

stangri, 2022-01-19:
The [dnsmasq-full](https://github.com/openwrt/openwrt/blob/d2d0044ebf01b71f63cde609e09f6ac68cdfeccb/package/network/services/dnsmasq/Makefile) still builds with `ipset` support. Is there a plan to switch to 2.87 which seems to be on the way of supporting nft sets?

This concerns me because of the policy routing packages which use dnsmasq ability to populate ipsets extensively. I don't mind looking into a rewrite to depend on nft instead, but unless there's a dnsmasq version which can populate nft sets instead, that would mean dropping one of the popular features.

I thought that I could delay the full rewrite by relying on `iptables-nft` for a while, that doesn't seem feasible anymore. :(

stintel, 2022-01-20:
> The [dnsmasq-full](https://github.com/openwrt/openwrt/blob/d2d0044ebf01b71f63cde609e09f6ac68cdfeccb/package/network/services/dnsmasq/Makefile) still builds with `ipset` support. Is there a plan to switch to 2.87 which seems to be on the way of supporting nft sets?

#4977

kamasudo, 2022-01-20:
The nat6-script needs also an update: https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6

tiagogaspar8, 2022-01-21:
Hi @aparcar 
I have a question. I'm working on porting jool to a version that isn't dependant on iptables.

Jool by default supports two modes:
- iptables mode
- netfilter mode

Netfilter mode works on both iptables and neftables via the netlink API.

The iptables mode obviously only works with iptables.

My question is, since we're inevitably dropping iptables, when porting jool do I make a package with only netlink and end the porting or do I make a package with netlink and another with both?

Thanks.

aparcar, 2022-01-21:
I'd suggest to go with `netfilter` only but if you want to support people that need `iptables`, you'd need both. I don't know how much work it is so I'd focus for now on `netfilter`.

tiagogaspar8, 2022-01-21:
That's the question, netfilter is done, iptables is the one giving me a lot of work... I'll make a pull request then 😁😁😁

hnyman, 2022-01-21:
Related to discussion at #17671

> I just assumed that the iptables-nft binary installed a compatibility symlink to the iptables binary; otherwise, what's the point of having a compatibility binary if applications have to select it explicitly? Shouldn't this be fixed in the iptables-nft package instead?

Good question. 
So far there is not much info about the usage of iptables-nft in the OpenWrt context. 

@aparcar enabled some iptables option with https://github.com/openwrt/openwrt/commit/0637093e8cb0e4cfb94dfcfe7b57afc23bf88852 in order to provide some nftables compatibility, but I don't think that the option actually installs iptables-nft. I had to install it separately.  

And so far the "iptables" and "iptables-nft" lead to different binaries. I manually installed iptables-nft package, and after that it looks like this:
```
root@router1:/usr/sbin# ls -lt /usr/sbin/*tabl*
lrwxrwxrwx    1 root     root            17 Jan 21 17:48 /usr/sbin/iptables-nft -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Jan 21 17:48 /usr/sbin/iptables-nft-restore -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Jan 21 17:48 /usr/sbin/iptables-nft-save -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Jan 21 17:48 /usr/sbin/iptables-restore-translate -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Jan 21 17:48 /usr/sbin/iptables-translate -> xtables-nft-multi
lrwxrwxrwx    1 root     root            20 Jan 21 17:19 /usr/sbin/iptables -> xtables-legacy-multi
lrwxrwxrwx    1 root     root            20 Jan 21 17:19 /usr/sbin/iptables-restore -> xtables-legacy-multi
lrwxrwxrwx    1 root     root            20 Jan 21 17:19 /usr/sbin/iptables-save -> xtables-legacy-multi
-rwxr-xr-x    1 root     root         58631 Jan 21 17:19 /usr/sbin/xtables-legacy-multi
-rwxr-xr-x    1 root     root        149855 Jan 14 01:42 /usr/sbin/xtables-nft-multi
```

Would be great if somebody with actual knowledge about the nftables firewall infra would enlighten us about the transition tools. (And maybe that symlink of iptables --> iptables-nft should be added, like some desktop distros apparently do.)

dangowrt, 2022-01-21:
We could use `ALTERNATIVES` for that, see e.g. openssh vs. dropbear packages both providing candidates for `scp` and the package manager takes care of maintaining a symlink.

tiagogaspar8, 2022-01-22:
Jool netfilter is merged, you can cross off jool @aparcar 
Thanks for the help!

hnyman, 2022-01-22:
Regarding module dependencies, some interesting stuff pops up, when packages are tried to be modified. 
E.g. SQM was tried to be straightened out by @tohojo , but SQM selecting iptables-nft instead of iptables caused a circular dependency.

```
Collecting package info: done
tmp/.config-package.in:38676:error: recursive dependency detected!
tmp/.config-package.in:38676:	symbol PACKAGE_libxtables is selected by PACKAGE_tc-mod-iptables
tmp/.config-package.in:108984:	symbol PACKAGE_tc-mod-iptables is selected by PACKAGE_tc-tiny
tmp/.config-package.in:108997:	symbol PACKAGE_tc-tiny is selected by PACKAGE_sqm-scripts
tmp/.config-package.in:1243:	symbol PACKAGE_sqm-scripts depends on PACKAGE_libxtables
```

Change in SQM Makefile causing circular:
```
...
-  DEPENDS:=+tc +kmod-sched-cake +kmod-ifb +iptables \
-       +iptables-mod-ipopt +iptables-mod-conntrack-extra
+  DEPENDS:=+tc +kmod-sched-cake +kmod-ifb +iptables-nft

```
I think that we (currently) need to also select the old iptables, as iptables-nft has for some strange reason the "at" dependency for iptables, instead of selecting it.
https://github.com/openwrt/openwrt/blob/0637093e8cb0e4cfb94dfcfe7b57afc23bf88852/package/network/utils/iptables/Makefile#L111

That easily causes a circular dependency when iptables-nft is selected along other network related packages.

(Selecting also iptables in sqm Makefile fixes the circular dependency.)

On the long the better solution would likely be to change dependency directly at the iptables-nft, but I am not even sure why there is that iptables dependency at iptables-nft, as I believe that iptables-nft is more related to nftables than to iptables. (But I have no deeper knowledge about the intricacies here)
Is the legacy iptables actually needed for iptables-nft?

To me it seems that the whole iptables/nftables/tc/whatever dependency universe is currently a bit twisted, likely because the dependencies have been drafted from the "iptables as default" perspective.  (and there is separately ip6tables-nft and all the ipv6 side...)

Hopefully people with enough knowledge about the netfilter modules can straighten the dependencies out.

aparcur, 2022-01-23:
> Long term there might be a `firewall5` package using eBPF things are moving :)

@aparcar why don't we make the jump to `firewall5` aka `eBPF` instead of `nftables`?

`nftables` 🤢🤮

tiagogaspar8, 2022-01-23:
As I've read many times, eBPF isn't ready for production and stable environments for now, maybe in a year or two we can have this headache porting everything to eBPF 😆

PolynomialDivision, 2022-01-23:
We just noticed that regex matching in firewall rules is no longer supported?

For example all wireguard interfaces with naming should be included in the zone `freifunk`:
```
config zone 'zone_freifunk'
	option name 'freifunk'
        list device 'wg_+'
```

CC: @spolack

champtar, 2022-01-23:
@PolynomialDivision try * instead of +

stangri, 2022-01-24:
> As I've read many times, eBPF isn't ready for production and stable environments for now, maybe in a year or two we can have this headache porting everything to eBPF 😆

Wouldn't the maturity of eBPF just coincide perfectly with the next stable OpenWrt release timing then?

neheb, 2022-01-24:
nope. I don't think eBPF is available for all OpenWrt supported platforms.

spolack, 2022-01-24:
> @PolynomialDivision try * instead of +

This wont work, as nftables doesnt support wildcard matching withing sets.

TheB1gG, 2022-01-25:
> nope. I don't think eBPF is available for all OpenWrt supported platforms.

Wouldn't it be usefull to do that than only for supported plattforms like already done for the kernel? Looking at the possible performance gains eBPF could be usefull for plattforms in enviroments which work with 1-25 gbps. That speeds are not so uncommon for consumers anymore.

neheb, 2022-01-25:
IIRC, x86, ARM, and MIPS currently have eBPF support. No idea about PowerPC or older ARM. ARC is non existent.

jow-, 2022-01-25:
Please move the eBPF discussion elsewhere it is offtopic here.

jow-, 2022-01-25:
> This wont work, as nftables doesnt support wildcard matching within sets. Furthermore i'd expect fw4 to handle this transparent to the user, otherwise its not suitable as drop-in replacement, breaking a standard configuration for many setups.

Not all peculiarities can be perfectly translated. The `+` notation for device names is certainly an iptables specific detail which happened to work because fw3 passed through the values as-is without validation, it doesn't mean that it was ever a supported configuration. It also far from being a standard configuration.

I'll look into handling this corner case though to avoid further surprises.

brada4, 2022-01-25:
Actually there is nft_compat module which handles ipset-s just fine, approx 20kB sized on Debian.

stangri, 2022-01-25:
> Actually there is nft_compat module which handles ipset-s just fine, approx 20kB sized on Debian.

Thank you! @jow- @aparcar can you please give me an idea on direction for the ipset and nftset support in dnsmasq in the future? Is there a chance there would be both a `dnsmasq-full-ipset` and `dnsmasq-full-nftset`?

If there would be an option in snapshots/next release to have a system with either iptables/ipset support (or emulation) or nftables/nftset support that'd be great.

jow-, 2022-01-25:
>  Any chance that you could look into the dependencies of the iptables/nftables/tc/iproute2 universe including the kmods?
@hnyman - I won't have time for that, someone else has to look into it.

jow-, 2022-01-25:
@PolynomialDivision @spolack - wildcard translation implemented with https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=0bc844ba02ae460d4a895878b9136ba5d8e09b37, please test and report back if is not working.

tiagogaspar8, 2022-01-26:
I switched yesterday to nftables completely and I'm facing an issue with the hw_offloading option. When enabled this option stops firewall4 from starting leaving nftables rules empty, software offloading isn't facing this issue. Any thoughts on this?

tiagogaspar8, 2022-01-26:
> Any chance that you could look into the dependencies of the iptables/nftables/tc/iproute2 universe including the kmods?

BTW, I have submitted a pull request that fixes n-conntrack-netlink and nf-nathelper-extra dependencies so that all nftables kmods can be enabled without selecting any iptables kmods.
About tc, I managed to build it without the iptables dependency and it is working very well, I'll package it up and submit a pull request as soon as I can.

champtar, 2022-01-26:
For reference the dependencies fix PR: #4986

champtar, 2022-01-26:
> I switched yesterday to nftables completely and I'm facing an issue with the hw_offloading option. When enabled this option stops firewall4 from starting leaving nftables rules empty, software offloading isn't facing this issue. Any thoughts on this?

Here is how I would debug it:
```
fw4 print > /tmp/nfrules
nft -f /tmp/nfrules
```
see what doesn't apply and maybe open a new report at https://bugs.openwrt.org/

tiagogaspar8, 2022-01-26:
This is what I get:
```
root@router2:~# fw4 start
Section @rule[9] (Support-UDP-Traceroute) is disabled, ignoring section
/proc/self/fd/0:9:12-13: Error: Could not process rule: Not supported

/proc/self/fd/0:51:29-44: Error: Could not process rule: No such file or directory
```
As per your recommendation here is it's output:

```
root@router2:~# nft -f /tmp/nfrules
/tmp/nfrules:9:12-13: Error: Could not process rule: Not supported
        flowtable ft {
                  ^^
/tmp/nfrules:51:29-44: Error: Could not process rule: No such file or directory
                meta l4proto { tcp, udp } flow offload @ft;
                                          ^^^^^^^^^^^^^^^^
```

champtar, 2022-01-26:
@tiagogaspar8 please open a bug report and include the list of installed packages

tiagogaspar8, 2022-01-26:
Will do

tiagogaspar8, 2022-01-27:
It seems that there is an issue open already, if anyone has the same issue please check it out:
https://bugs.openwrt.org/index.php?do=details&task_id=4239

graysky2, 2022-01-28:
@aparcar - Should you add `tc-tiny` and `tc-full` to this list?

champtar, 2022-01-28:
@graysky2 this is packages repo

[107 more replies]