openwrt/packages #16869: [19.07] Python packages (zipp and importlib-metadata) related to tomli

BKPepe, 2021-10-12:
Python maintainers: @jefferyto , @commodo 
cc: @ja-pa 
Environment: all Turris routers (1. Turris 1.x - powerpc - mpc85xx, 2. Turris Omnia - mvebu - cortexa9, 3. Turris MOX - mvebu - cortexa53), OpenWrt 19.07, reproduced locally and logs can be found in OpenWrt buildbots, too.

Description:

I noticed that these two packages python-zipp and python-importlib-metadata are not compiled:

Copy&pasted from [faillogs](https://downloads.openwrt.org/releases/faillogs-19.07/aarch64_cortex-a53/packages/python-zipp/python3/compile.txt):
```
Collecting setuptools-scm
  Using cached setuptools_scm-6.3.2.tar.gz (57 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: /builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3.7 /builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /builder/shared-workdir/build/sdk/tmp/pip-build-env-rw7udqhw/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=45' wheel 'tomli>=1.0' 'packaging>=20.0'
       cwd: None
  Complete output (57 lines):
  Collecting setuptools>=45
    Using cached setuptools-58.2.0.tar.gz (2.3 MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting wheel
    Using cached wheel-0.37.0.tar.gz (65 kB)
  Collecting tomli>=1.0
    Using cached tomli-1.2.1.tar.gz (14 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'error'
    ERROR: Command errored out with exit status 2:
     command: /builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3.7 /builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /builder/shared-workdir/build/sdk/tmp/pip-build-env-0ylhv6uz/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple -- 'flit_core>=3.2.0,<4'
         cwd: None
    Complete output (35 lines):
    Collecting flit_core<4,>=3.2.0
      Using cached flit_core-3.4.0.tar.gz (27 kB)
      Getting requirements to build wheel: started
      Getting requirements to build wheel: finished with status 'done'
        Preparing wheel metadata: started
        Preparing wheel metadata: finished with status 'done'
    Collecting tomli
      Using cached tomli-1.2.1.tar.gz (14 kB)
    ERROR: Exception:
    Traceback (most recent call last):
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 188, in _main
        status = self.run(options, args)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 185, in wrapper
        return func(self, options, args)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 333, in run
        reqs, check_supported_wheels=not options.target_dir
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 179, in resolve
        discovered_reqs.extend(self._resolve_one(requirement_set, req))
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 362, in _resolve_one
        abstract_dist = self._get_abstract_dist_for(req_to_install)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 314, in _get_abstract_dist_for
        abstract_dist = self.preparer.prepare_linked_requirement(req)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 488, in prepare_linked_requirement
        req, self.req_tracker, self.finder, self.build_isolation,
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 90, in _get_prepared_distribution
        with req_tracker.track(req):
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/contextlib.py", line 112, in __enter__
        return next(self.gen)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/req/req_tracker.py", line 149, in track
        self.add(req)
      File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/req/req_tracker.py", line 116, in add
        raise LookupError(message)
    LookupError: https://files.pythonhosted.org/packages/75/50/973397c5ba854445bcc396b593b5db1958da6ab8d665b27397daa1497018/tomli-1.2.1.tar.gz#sha256=a5b75cb6f3968abb47af1b40c1819dc519ea82bcc065776a866e8d74c5ca9442 (from https://pypi.org/simple/tomli/) (requires-python:>=3.6) is already being built: tomli>=1.0 from https://files.pythonhosted.org/packages/75/50/973397c5ba854445bcc396b593b5db1958da6ab8d665b27397daa1497018/tomli-1.2.1.tar.gz#sha256=a5b75cb6f3968abb47af1b40c1819dc519ea82bcc065776a866e8d74c5ca9442
    WARNING: You are using pip version 20.1.1; however, version 21.2.4 is available.
    You should consider upgrading via the '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3.7 -m pip install --upgrade pip' command.
    ----------------------------------------
  ERROR: Command errored out with exit status 2: /builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3.7 /builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /builder/shared-workdir/build/sdk/tmp/pip-build-env-0ylhv6uz/overlay --no-warn-script-location --no-binary :all: --only-binary :none: -i https://pypi.org/simple -- 'flit_core>=3.2.0,<4' Check the logs for full command output.
  WARNING: You are using pip version 20.1.1; however, version 21.2.4 is available.
  You should consider upgrading via the '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3.7 -m pip install --upgrade pip' command.```

I tried to use the python-zipp Makefile from `master` branch and also update it to the latest version v3.6.0, but I was still not able to compile it.

commodo, 2021-10-13:
yes; pinning is the way to go;
i think also had to pin some package in 19.07;
we may need to pin some more (or all of these `HOST_PYTHON3_PACKAGE_BUILD_DEPENDS`) packages as time goes by;

cotequeiroz, 2021-10-16:
#16879 has fixed it.  I have succesfully built all 19.07 packages for mvebu and ar71xx, and both zipp and importlib-metadata are not showing up in the faillogs after Oct 15.  Im closing this.

closed 2021-10-16 by cotequeiroz