openwrt/packages #17028: ddns-scripts: use https for spdns updates

Gransi, 2021-10-30:
Maintainer: @feckert
Environment: (put here arch, model, OpenWrt version)
Qualcomm Atheros QCA9558 ver 1 rev 0
TP-Link Archer C7 v2
OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.301.66596-f28aaa3

Description:
I get in my spdyn.de account portal a error message, that i should use https for ddns updates.

old url:
```
http://[USERNAME]:[PASSWORD]@update.spdyn.de/nic/update?hostname=[DOMAIN]&myip=[IP]
```

I changed this url in my config and now it works without error messages
```
https://update.spdyn.de/nic/update?hostname=[DOMAIN]&myip=[IP]&user=[USERNAME]&pass=[PASSWORD]
```

BKPepe, 2021-10-30:
Would it be possible to send a pull request?

rmilecki, 2022-09-27:
Solved in the c0296bf25da7a4832059d0a708431aef4c5f3238 ("ddns-scripts: use HTTPS for spdyn")

closed 2022-09-27 by rmilecki